fredlcore/BSB-LAN #26: Send alphanumeric values over MQTT

hvorragend, 2019-02-11:
Aktuell werden über MQTT nur rein numerische Werte übertragen. Im Forum hat es dazu auch schon Rückmeldungen gegeben: https://forum.fhem.de/index.php/topic,29762.msg901497.html#msg901497

Du hast es  "Abhacken bei den ENUMs" genannt.

```
6801 Historie 01 Fehlercode: 110 (MQTT)
6801 Historie 01 Fehlercode: 110 - STB Störabschaltung (HTTPMOD)
Betriebsart: 1 (MQTT)
Betriebsart: 1 - Automatik (HTTPMOD)
Brennerstufe_1: 0 (MQTT)
Brennerstufe_1: 0 - Aus (HTTPMOD)
```

Auch ich würde mich sehr freuen, wenn über MQTT die gleichen Ausgaben übertragen werden könnten, wie auf der Weboberfläche angezeigt. 

Danke.

fredlcore, 2019-05-27:
Sorry für die lange Wartezeit, aber könntest Du mal probieren, ob Ersetzen dieser Zeile:
`MQTTClient.publish(MQTTTopic.c_str(), strtok(query(log_parameters[i],log_parameters[i],1)," "));`
mit diesem Block:
```
          uint32_t c=0;
          int line=findLine(log_parameters[i],0,&c);
          uint8_t type=get_cmdtbl_type(line);
          if (type == VT_ENUM || type == VT_BIT || type == VT_ERRORCODE) {
            MQTTClient.publish(MQTTTopic.c_str(), query(log_parameters[i],log_parameters[i],1));
          } else {
            MQTTClient.publish(MQTTTopic.c_str(), strtok(query(log_parameters[i],log_parameters[i],1)," "));
          }
```
die gewünschte Änderung für die Datentypen VT_ENUM, VT_BIT und VT_ERRORCODE bringt? Bitte ggf. rückmelden, falls es weitere Datentypen gibt, wo statt der numerischen Werte die komplette Rückgabe übergeben werden soll (evtl. bei VT_DATETIME, Sommerzeit, Zeitprogrammen?).
Wenn das so klappt, übernehme ich das in die offizielle Version.

hvorragend, 2019-05-29:
Danke dir für dein Engagement.
Ich würde das wirklich gerne testen, aber ich komme vermutlich erst Mitte/Ende Juni dazu.

hvorragend, 2019-06-09:
Hallo,

ich bin nun endlich dazu gekommen und habe dabei auf festgestellt, dass du die Code-Zeilen bereits in https://github.com/fredlcore/bsb_lan/commit/be82d2da232541bb5bd6efa61a4a1df5a1b210b6 geändert hast. 

Die (tagesaktuelle) Version hat jedoch alle Readings in FHEM nun geleert.
Es wird also nun NICHTS übertragen via MQTT.

Folgende Konfiguration:

```
const int fixed_device_family = 162;
const int fixed_device_variant = 14;
```

```
int log_parameters[80] = {
  700,
  710,
  712,
  720,
  6705, // Alarm 
  6710,
  6740,
  6741,
  6743,
  6745,
  6800, // Fehler
  6801,
  6805,
  6806,
  6810,
  6811,
  6815,
  6816,
  6820,
  6821,
  6825,
  6826,
  6830,
  6831,
  6835,
  6836,
  6840,
  6841,
  6845,
  6846,
  6850,
  6851,
  6855,
  6856,
  6860,
  6861,
  6865,
  6866,
  6870,
  6871,
  6875,
  6876,
  6880,
  6881,
  6885,
  6886,
  6890,
  6891,
  6895,
  6896,
  8000,
  8003,
  8005,
  8009,
  8300,
  8301,
  8304,
  8308,
  8310,
  8312,
  8314,                   // Rücklauftemperatur
  8311,
  8315,
  8323,
  8324,
  8326,
//  8331,                   // Startzähler 1.Stufe
//  8333,                   // Startzähler 2.Stufe
  8700,                   // Außentemperatur
  8703,
  8704,
  8740,
  8741,
  8742,
  8744,
  8830,
  8831,
  8950,
  8951
//  30000,                  // Logging von "rohen" Bus-Datentelegrammen (macht nur als alleiniger Parameter Sinn)
//  20000,                  // Spezialparameter: Brenner-Laufzeit Stufe 1(/B)
//  20001,                  // Spezialparameter: Brenner-Takte Stufe 1 (/B)
//  20002,                  // Spezialparameter: Brenner-Laufzeit Stufe 2(/B)
//  20003,                  // Spezialparameter: Brenner-Takte Stufe 2 (/B)
//  20004,                  // Spezialparameter: TWW-Laufzeit (/B)
//  20005,                  // Spezialparameter: TWW-Takte (/B)
//  20006,                  // Spezialparameter: 24h-Durchschnittswerte (/A)
//  20101,                  // Spezialparameter 20100-20199: DHT22-Sensoren 1-100 (/H)
//  20200                   // Spezialparameter 20200-20299: DS18B20-Sensoren 1-100 (/T)
};
```

hvorragend, 2019-06-12:
Ich mache das Ticket noch einmal kurz auf.  
Könntest du VT_ONOFF bitte auch aufnehmen?
Danke.

fredlcore, 2019-06-12:
Was meinst Du genau?

hvorragend, 2019-06-12:
Folgende BSB Zeilen werden via MQTT nur mit 0 übermittelt. 

''
8300 Diagnose Erzeuger - 1. Brennerstufe T2: 0 - Aus		
8301 Diagnose Erzeuger - 2. Brennerstufe T8: 0 - Aus
'M

fredlcore, 2019-06-12:
Ich habe das zwar jetzt eben so committed, aber eigentlich in der Annahme, dass das "aus" weg soll und nur der numerische Wert übertragen werden soll. Bei ENUMs verstehe ich das ja, aber gerade an/aus Werte will man ja gerne mal plotten, und da macht ein rein numerischer Wert wirklich mehr Sinn. Zumal 0=aus und 1=ein ziemlich eingängig ist, anders als bei langen ENUM-Listen, wo man sich nichts draus ableiten kann...

closed 2019-06-12 by fredlcore

hvorragend, 2019-06-12:
Okay. Dann lass ich mir was in FHEM einfallen.