fredlcore/BSB-LAN #33: Possibility to publish 1wire sensors values on MQTT

seb821, 2019-05-11:
It would nice to get an option to publish 1wire sensors like DS18B20 on the MQTT server.
Same for DHT sensors may also reveal usefull.

fredlcore, 2019-05-27:
With the most recent update, this should work now. Parameter numbers are the same as for logging on SD card, i.e. 20000-20005 for burner status, 20100++ for DHT and 20200++ for 1Wire. Please check if all works as expected.

seb821, 2019-05-31:
Hello,

I found a few things to make it work.

On line 6052, it is necessary to use a value higher than 20000.

Before line 6094, it requires a #ifdef DHT_BUS or it won't compile.

I would suggest to put DHT.read22 and sensors.requestTemperatures (with appropriate #ifdef) before the loop on numLogValues as it will save some ressources avoiding asking for all sensors several times.

Most important, it looks like the sprintf function does not work properly with floats in the arduino environment. Value obtained is a question mark. I tested the following code and that worked:

```char *tmpSign = (t < 0) ? "-" : "";
float tmpVal = (t < 0) ? -t : t;
int tmpInt1 = tmpVal;
float tmpFrac = tmpVal - tmpInt1;
int tmpInt2 = trunc(tmpFrac * 100);
sprintf (buffer, "%s%d.%02d\n", tmpSign, tmpInt1, tmpInt2);
```
Regards,

fredlcore, 2019-05-31:
Thanks for these valuable suggestions.
- as for 20000 in line 6052, I think it has to be removed completely, because making it higher than 20000 would prevent normal values to be sent over MQTT, if I'm not mistaken.
- #ifdef near 6094 added
- I think DHT.read22 has to be executed for each supplied sensor as DHT22s are not on a bus. For OneWire and sensors.requestTemperatures you are right, though.
- strange that it does not work properly with floats, but as I don't have MQTT running here at the moment, I take your code snippet as working and have added it to the code. However, it complains that tmpSign is not declared. Also, why the need to add a "\n" at the end of the sprintf string?

fredlcore, 2019-05-31:
Updated version on GitHub, please check...

seb821, 2019-05-31:
Indeed, tmpSign definition was missing but what you implemented is equivalent.
`char *tmpSign = (t < 0) ? "-" : "";`
As for \n, this is left from some code I found on the internet and thus can be removed.
I have just tested the last modifications and it works, thanks !

closed 2019-05-31 by seb821

fredlcore, 2019-06-01:
Great, thanks for testing!