fredlcore/BSB-LAN #39: Thision S Plus 19

Luposoft63, 2019-08-11:
ich bitte um Hinzufügung folgender Zeilen in die BSB_LAN_defs.h
ich habe dies so in meine eingefügt, und /Q bringt keine Fehlermeldungen mehr.
Die meisten Meldungen konnte ich nicht prüfen, da diese Nummern nicht im Bedienfeld auftauchen (OEM?)
Der Rest stimmt.
Ich hoffe das reicht so.
Gruß Luposoft

#define DEV_188_ALL  188,255 // AVS37.296/141

{0x213D0B43,  CAT_HK1,              VT_BYTE,          857,   STR857,   0,                    NULL,         DEFAULT_FLAG, DEV_195_ALL}, // [ Tage ] - Heizkreis 1 - Estrich Tage erfüllt //FUJITSU
{0x223D0B43,  CAT_HK2,              VT_DAYS,          1157,  STR1157,  0,                    NULL,         FL_RONLY, DEV_195_ALL}, // Estrich Tag erfüllt? logged from ACS700 diagnosis software
{0x113D11A4,  CAT_KESSEL,           VT_MINUTES_SHORT, 2250,  STR2250,  0,                    NULL,         FL_OEM, DEV_195_ALL}, // [ min] Kessel - Pumpennachlaufzeit
{0x093D0F31,  CAT_SITHERM,           VT_CURRENT,      2700,  STR2700_2,0,                    NULL,         FL_RONLY, DEV_195_ALL}, // Ion'strom gefiltert
{0x053D3042,  CAT_SITHERM,           VT_SINT,         2702,  STR2702_2,0,                    NULL,         FL_RONLY, DEV_195_ALL}, // Position Schrittmotor // todo VT_SINT verifizieren
{0x053D3043,  CAT_SITHERM,           VT_SINT,         2703,  STR2703_2,0,                    NULL,         FL_RONLY, DEV_195_ALL}, // Lernwert Gasqualität  // todo VT_SINT verifizieren
{0x093D0F26,  CAT_SITHERM,           VT_SINT,         2705,  STR2705_2,0,                    NULL,         FL_RONLY, DEV_195_ALL}, // R-Wert
{0x053D1388,  CAT_KONFIG,           VT_BYTE,          6221,  STR6221,  0,                    NULL,         DEFAULT_FLAG, DEV_195_ALL}, // Thision 6221 Entwicklungs-Index [?]
{0x053D1193,  CAT_KONFIG,           VT_UINT,          6230,  STR6230_2,0,                    NULL,         FL_RONLY, DEV_195_ALL}, // Konfiguration - Info 1 OEM
{0x053D1193,  CAT_KONFIG,           VT_UINT,          6300,  STR6300_2,0,                    NULL,         DEFAULT_FLAG, DEV_195_ALL}, // Info 1 OEM
{0x053D1AC1,  CAT_STATUS,           VT_ENUM,          8023,  STR8023,  sizeof(ENUM8009),     ENUM8009,     FL_RONLY,     DEV_195_ALL}, // Status Sitherm Pro
{0x053D19F0,  CAT_DIAG_ERZEUGER,    VT_PRESSURE_WORD, 8327,  STR8327,  0,                    NULL,         FL_RONLY,     DEV_195_ALL}, // Wasserdruck
{0x053D0006,  CAT_LPB,              VT_YESNO,         6610,  STR6610,  0,                    NULL,         FL_OEM, DEV_188_ALL}, // Anzeige Systemmeldungen

fredlcore, 2019-08-11:
Der Output von /Q ohne die Änderungen wären hilfreicher gewesen, denn so macht das nicht so viel Sinn, denn in der Define-Zeile ist von der Gerätefamilie 188 die Rede und dann betreffen die meisten Definitionszeilen die Gerätefamilie 195. Mit dem Output von /Q kann ich genau sehen, welche Gerätefamilie Du im Einsatz hast und welche Parameter noch nicht "freigeschaltet" sind.

Luposoft63, 2019-08-12:
Oh,
Ich hatte im Fhem-Forum (Lan-Anbindung für BSB-Bus) die /Q-Ausgabe gepostet. 
#3658 auf Seite 244
die neue Familie 188 ist mein Bedienfeld
Die 195 war schon vorhanden, ist auch die Thision
Brauchst du noch was?

Gesendet mit der GMX Mail App

fredlcore, 2019-08-12:
Oh, sorry, das ist mir durchgerutscht und hatte ich noch nicht eingearbeitet. Ist jetzt fertig und auf GitHub aktualisiert. Danke!

closed 2019-08-12 by fredlcore

Luposoft63, 2019-08-13:
Bisher sehe ich die Änderungen nicht, wo liegt der Fehler?

fredlcore, 2019-08-13:
Dass ich sie nicht nur speichern, sondern auch hochladen muss ;)... Jetzt sollte es gehen.