fredlcore/BSB-LAN #40: Unable to query parameter 8982 via JSON API

hypetsch, 2019-08-20:
Hi,

I am currently implementing an openhab binding which uses the JSON API to periodically query values from the BSB-LAN device. During some tests I noticed that I am unable to query parameter 8982 while it is showing correctly in the web interface.

Executing a `/JQ=8982` just returns an empty JSON. It is also missing in the `/JK=39` output:
```
...
"8980": {
    "name": "Pufferspeichertemperatur 1",
    "possibleValues": [

    ],
    "dataType": 0
  },
  "8981": {
    "name": "Pufferspeichersollwert",
    "possibleValues": [

    ],
    "dataType": 0
  },
  "9000": {
    "name": "Vorlaufsollwert H1",
    "possibleValues": [

    ],
    "dataType": 0
  },
...
```

I assume this is because `findline(...)` cannot find the parameter although it is defined. The cause might be the misordered line https://github.com/fredlcore/bsb_lan/blob/9e36a840ffeef146a3ae4f938996d863c397c996/BSB_lan_defs.h#L7934

Without having it tried I think the same issue exists for
https://github.com/fredlcore/bsb_lan/blob/9e36a840ffeef146a3ae4f938996d863c397c996/BSB_lan_defs.h#L7405

Note: My device is currently flashed the branch from #37 so it might be that it is not reproducible with `master` due to the different number of lines in the command tables.

regards
Peter

fredlcore, 2019-08-20:
Good call! Thanks a lot, the second one is fixed and will be updated in the next push, the first one does not come from me, so I have contacted the contributing user and will hopefully patch this as well soon.

closed 2019-08-20 by fredlcore

hypetsch, 2019-09-09:
Thx, looking forward to the fix.

fredlcore, 2020-01-24:
Sorry, this was fixed a while ago without reporting back here, sorry!