fredlcore/BSB-LAN #41: Housing for BSB LAN Gateway

hacki11, 2019-09-22:
Hi,

just wanted to know if you can recommend any housing/case for the hardware.
Doesn't matter if print by myself or if there is already a product available

hacki11, 2019-10-04:
I took an standard case from conrad.
[image]
[image]
[image]

closed 2019-10-04 by hacki11

fredlcore, 2019-10-05:
FHEM user EPo has provided us with a customized case to print with a 3D printer. The STL files can now be found in the "schematics" subdirectory.