fredlcore/BSB-LAN #45: Thision S 19 Plus - neue Parameter

Luposoft63, 2019-12-29:
Hallo,

ich bin die Q-Liste durchgegangen, davon kann ich nur 3 verifizieren. Der Rest erscheint auf meinem Display nicht.
Ansonsten bin ich für weitere Gruppen durchs Display gegangen.
Anbei meine Änderungswünsche.

Lieben Dank
Eiko

[Parameter.txt](https://github.com/fredlcore/bsb_lan/files/4008028/Parameter.txt)

fredlcore, 2019-12-29:
Hallo Eiko,

vielen Dank - hat das aber etwas mit Deinem anderen Issue zu tun?
https://github.com/fredlcore/bsb_lan/issues/39

Und auf welche Q-Liste beziehst Du Dich? Auf die, die Du hier im FHEM-Forum als CSV aufbereitet hattest?
https://forum.fhem.de/index.php/topic,29762.msg1001983.html#msg1001983

Oder ist das jetzt noch mal was anderes? Mir fehlt da jetzt etwas der Zusammenhang...

Luposoft63, 2019-12-29:
Hallo Frederik,

mit der Q-Liste meinte ich die Abfrage /Q und daraus resultierend Issue 39.
Du hast mich gebeten, diese Parameter nochmals zu verifizieren.
Das sind die ersten 3 Zeilen im File. Mehr ging nicht, da die anderen Parameter nicht im Display angezeigt werden.
Der ganze Rest entstand durch Auswählen der Parameter via Bedieneinheit. Teilweise sind neue ID's dabei bzw. andere Texte.

fredlcore, 2019-12-29:
Ok, danke, aber woher kommen dann die anderen Einträge aus Issue 39? Bzw. was ist jetzt mit denen?

Luposoft63, 2019-12-29:
Oh, da schaue ich nochmal genau nach wie das war.
Da melde ich mich später nochmal.

Luposoft63, 2019-12-30:
Nochmal zu den /Q-Ausgaben, du fragst, was mit denen sei...
das waren alles schon bekannte Parameter und nun gibt es für DEV_195_ALL jeweils einen Eintrag.
Damit läuft /Q ohne Fehler durch. 
Ich hatte es schon im #39 geschrieben. Ich kann diese Daten nicht überprüfen, da sie im Bedienfeld der Heizung nicht auftauchen.
Oder gibt es da noch was anderes was ich tun sollte?
Zur Sicherheit hänge ich meine /Q-Ausgabe vom Anfang nochmal mit ran.
[ErsterStart mit Q.txt](https://github.com/fredlcore/bsb_lan/files/4011104/ErsterStart.mit.Q.txt)

fredlcore, 2019-12-30:
Ah, ok, verstehe, dann war das einfach schon eine von Dir umgesetzte /Q Ausgabe. Gut, dann habe ich jetzt hoffentlich alle Infos und kann dann den Rest demächst nachtragen. Ich schließe dann mal hier, weil ich es dann intern auf die ToDo-Liste setze.

closed 2019-12-30 by fredlcore

fredlcore, 2019-12-31:
Habe die Änderungen jetzt eingepflegt und sind auf GitHub aktualisiert, aber zwei Nachfragen habe ich noch:

1.
Ist 8327 (Wasserdruck) bei Dir bisher nicht über BSB-LAN anzeigbar? Ich frage nur, weil sich die ID nur minimal unterscheidet:
053D3063 (neu bei Dir)
113D3063 (bisher als Default hinterlegt)

2.
Du hast bei Parameter 8805 die Bemerkung "anderer Text" hinterlegt, aber es gibt Parameter 8805 (noch) gar nicht, kannst Du da noch mal schauen? 8804 hat darüber hinaus den Namen "Vorlaufsollwert 3".

Und generell bei bisher nicht vorhandenen Parameternummern: Die können auf DEV_ALL gesetzt werden, damit sie prinzipiell auch für andere Geräte erst einmal zur Verfügung stehen. Die "Ursprungsheizung" kann dann in dem Kommentar am Ende der Zeile festgehalten werden (wie Du es ja auch schon gemacht hast).

Was die anderen Texte angeht, werde ich die erst einmal so belassen, weil bei manchen Geräten der HK 3 als HK P benannt wird. Ggf. ersetze ich die auch alle auf 3/P oder so etwas, da muss ich @1coderookie noch mal fragen, wie er das einschätzt, weil er sich mit der Heizungstechnik un den Unterschieden bei den HKs besser auskennt als ich.

Luposoft63, 2020-01-08:
Doch, der Wasserdruck war anzeigbar.
Hier die beiden WebIF-Ausgaben:
vorher die genutzt:
http://192.168.68.39/Y06,053D19F0 (DEV_195_ALL)
8327 Diagnose Erzeuger - Wasserdruck: 1.8 bar
DC C2 00 0B 06 3D 05 19 F0 24 3E
DC 80 42 0E 07 05 3D 19 F0 00 00 12 DE E8 
nun die:
http://192.168.68.39/Y06,053D3063 (DEV_195_002)
8327 Diagnose Erzeuger - Wasserdruck: 181.3 bar
DC C2 00 0B 06 3D 05 30 63 2B 9A
DC 80 42 0E 07 05 3D 30 63 00 07 15 CA 13 

für DEV_195_002 habe ich den Typ auf VT_SINT1000 geändert.
/1000 für bar gibt es noch nicht.
Ändere bitte das auch im GIT oder falls dein Ordnungssinn das verlangt, baue bitte noch eine /1000 für bar ein.

zu den Parametern 8804/05
den Parameter 8803 gibt es bei mir nicht.
8004 existiert so schon (ist bei mir aber Vorlauftemp 3)
8005 hat die ID wie in defs die 8003 (ist bei mir Vorlaufsollwert 3) - da hatte ich mich mit 'anderer Text' vertan
-ich vermute, da ist mal was durcheinander gekommen. 8803 und 8804 als Vorlaufsollwert scheint nicht logisch

fredlcore, 2020-01-08:
Ich meinte jetzt nicht die ID, die für GF 195 vorgesehen war, sondern die für DEV_ALL gelistet ist: 113D3063
Die unterscheidet sich von Deiner 053D3063 nur in den ersten zwei Ziffern, und da wollte ich sicher gehen, dass die 113D3063 bei Dir nicht geht. Den Typ sowie die 8004/5 trage ich dann entsprechend nach.

Luposoft63, 2020-01-09:
mit 113D3063 funktioniert der Wasserdruck nicht.
WebIF:
http://192.168.68.39/Y06,113D3063
8327 Diagnose Erzeuger - Wasserdruck: error 7 (parameter not supported)
DC C2 00 0B 06 3D 11 30 63 B4 39
DC 80 42 0C 08 11 3D 30 63 07 2C 74

fredlcore, 2020-01-13:
Ok, verstanden. Das mit 8003/4/5 ist seltsam, weil bei HK1 und HK2 die Parameter auch immer auf 3/4 enden. Ich würde es jetzt dieser Logik folgend so einpflegen, würde das dann bei Dir die richtigen Werte anzeigen, auch wenn die Parameternummern dann um eins vermindert wären?
8803 - Vorlauftemperatur 3/P - 0x233D0518
8804 - Vorlaufsollwert 3/P - 0x233D0667

1coderookie, 2020-01-14:
Hi,
also meine folgenden Aussagen beziehen sich immer nur auf die entspr. BDAs. Ich habe jetzt knappe zwei Stunden verschiedene BDAs gesucht und (digital) gewälzt, das Ergebnis sieht so aus:

- RVS43 beim aktuellen Brötje-Öler (= BOB)
- RVS63 beim Brötje-SSR
- LMS14 in einem **Albatros**-Manual
- LMS14 in einem **Brötje**-Manual
Dort sind die Bezeichnungen wie folgt:
8743 VL-Ist 1
8744 VL-Soll 1
8773 VL-Ist 2
8774 VL-Soll 2
8803 VL-Soll 3
8804 VL-Ist 3
-> 8803&8804 sind von der Bezeichnung her also zu den vorhergehenden 'vertauscht'.
---
Beim LMS14 der **Elco**-Modelle scheinen sie sich dann bei 8803&8804 an die vorherige HK1/HK2-Reihenfolge gehalten und es entspr geändert zu haben:
8743 VL-Ist 1
8744 VL-Soll 1
8773 VL-Ist 2
8774 VL-Soll 2
8803 VL-Ist 3
8804 VL-Soll 3

Dass die 3er bei dir (Eike) um eins erhöht sein sollen (also 8804&8805) und 8803 gar nicht vorhanden sein soll, kann ich fast nicht glauben, denn dann würde die entspr Elco-BDA für die aktuelle Thision Plus mit LMS14 (2018) falsch sein. 
8805 gibt es laut BDA dort gar nicht.
Quelle: www.elco.ch/de/resources/downloads/technische_dokumentationen/Betriebsanleitungen/ELCO_THISION-S2_Betriebsanleitung_DE.pdf

Ich habe gerade noch zwei User angeschrieben, die einen identischen Regler verbaut haben (also (LMS14.111B109 = GF195/GV2 mit SW4.3) und sie gebeten, die Parameter 8803, 8804, 8805 auf das Vorhandensein und die jeweilige Bezeichnung an der kesselseitigen Bedieneinheit zu überprüfen. 
 
Die genannten Elco-Bezeichnungen finden sich übrigens auch in älteren
Elco-BDAs wieder - möglich bzw. es scheint so, dass 8803&8804
Elco-spezifisch also generell so benannt sind! 
---
Ich habe in der defs.h bei mir nachgesehen, dort scheint ein genereller
Fehler vorzuliegen:
{0x233D0667,  CAT_DIAG_VERBRAUCHER, VT_TEMP,          8803, (...) -
Diagnose Verbraucher - Vorlaufsol wert P
{0x233D0518,  CAT_DIAG_VERBRAUCHER, VT_TEMP,          8804, (...) -
Diagnose Verbraucher - Vorlaufsol wert P

1. Fehlt dort das zweite "l" bei "Vorlaufsollwert", statt dessen ist ein
Leerezeichen in der Bezeichnung.
2. Sind beide Parameter als VL-Soll bezeichnet, was ja grundsätzlich
sowieso falsch ist - unabhängig davon, nach welcher
Bezeichnungsreihenfolge man sich nun letztlich richtet.

Gruß

1coderookie, 2020-01-14:
Hi nochmal,
so, das ging fix: Ein User hat bereits eine Rückmeldung samt Foto geschickt. Dort ist es so, dass bei 8804 die VL-Temp3 angezeigt wird und in einer weiteren Zeile die VL-Soll3. Also dort werden im Grunde zwei Parameter gleichzeitig angezeigt.
8805 ist dort nicht existent, 8803 als einzelner Parameter ebenfalls nicht. Wäre nun die Frage, ob man in dem Fall 8804 als IST und Zusammenfassung von 8803&8804 ansieht oder als SOLL und Zusammenfassung von 8803&8804.
WENN die Darstellung dort bedeutet, dass 8804 auch hier das VL-Temp3-IST bedeutet und die Zeile darunter (also VL-Soll) aus 8803 kommt, dann wäre die Darstellung identisch mit Brötje & Albatros und die Elco-BDA wäre falsch.
Ich werde nochmal ein wenig recherchieren, kann aber etwas dauern..

Wie sieht das denn bei dir im Display aus Eiko? Und hast du auch die blaue Hintergdunbeleuchtung oder noch eine orangene?

Btw: Können wir die Diskussion hierüber vielleicht per Email fortsetzen? Das ist doch etwas komfortabler, als sich immer hier einzuloggen und wir können auch besser Anhänge mitschicken - was meint Ihr..?

Gruß

fredlcore, 2020-01-14:
Ich fände das jetzt hier ganz passend, weil man dann alles auf einen Blick hat, und Anhänge kann man einfach in das Textfeld ziehen. Einmal eingeloggt muss ich mich auch nicht mehr anmelden.

Diese Parameter-Doppelungen hatten wir bisher schon einige Male, bisher haben wir das so gelöst, dass der oben genannte Parameter der angezeigten Parameternummer zugeordnet wurde und die Zeile darunter der nachfolgenden Parameternummer, falls diese noch nicht belegt war. Das ging meistens, aber nicht immer auf und das müsste man hier auch mal sehen. Wenn das aber so ein Parameter ist, wo es eh' bei mehreren Heizungen unterschiedlich bzw. wie in dem Fall des Users sogar uneindeutig ist, dann würde ich fast eher bei einer Darstellung bleiben, die dann auch konsistent ist, also ...3er für IST und ...4er für SOLL.

1coderookie, 2020-01-14:
[quote]
dann würde ich fast eher bei einer Darstellung bleiben, die dann auch konsistent ist, also ...3er für IST und ...4er für SOLL.
[/quote]
Das wäre dann (nur) bei Elco laut BDAs der Fall, bei den anderen ist es wie gesagt bei 8803&8804 genau andersrum. 
Sehr merkwürdig und gleichzeitg spannend, dass es da anscheinend Abweichungen gibt.. :( ;)

fredlcore, 2020-01-14:
Genau, aber wenn wir es schon nicht einheitlich über verschiedene Hersteller hinkriegen, dann eher so, wie es am sinnhaftesten ist. Wie gesagt, da es sich um einen rein lesbaren Parameter handelt, wird die Verwirrung nicht allzu groß sein. Wenn wir eine klare Auflistung hätten, welche Heizung welche Reihenfolge hat, könnte ich das hinterlegen, aber wenn es dann so Anomalien wie bei dem User gibt, wo zwei Angaben auf eine Nummer gemappt werden (zumindest in der Anzeige), dann kommen wir da sowieso nicht für alle Heizungen auf eine Lösung, die mit der BDA oder dem Display übereinstimmt.

Luposoft63, 2020-01-14:
Anbei auch ein paar Fotos.
Mein Display ist auf jeden Fall eher blau als orange. Ohne die Auswahl
hätte ich grau gesagt ...

Um ganz sicher zu gehen, welcher von den beiden gemeldeten Werten
welcher ist, müßte HK3 aktiv sein...
Wer hat das schon?
Ich für meinen Teil, mit meinem einen HK, werde mit jeder Lösung leben
können. Ich hatte die Diskrepanz auch nur der Ordnung halber mitgeteilt.

Gruß Eiko

fredlcore, 2020-01-14:
Nein, alles gut, das ist für uns ja gerade wichtig, solche Anomalien zu erkennen, gerade dann, wenn sich noch Diskrepanzen gegenüber den Handbüchern ergeben.
Aber kann es sein, dass Du die Fotos nicht angehängt hast? Ich sehe jedenfalls keine...

Luposoft63, 2020-01-15:
Ja, die Bilder hatte ich angehängt.
Jetz nochmal. Die sind insgesamt 3 MB. Das ist doch nicht zuviel, oder?

Gruß Eiko

fredlcore, 2020-01-15:
Sehr seltsam - ich sehe immer noch nichts, siehst Du die denn? Ich versuche es jetzt auch mal mit einem:
[image]

Luposoft63, 2020-01-15:
> LWScreenShot 2020-01-10 at 16 07 41

Ja, deinen Anhang habe ich gesehen.
Ich schicke die Bilder mal direkt an deine Adresse  bsb@code-it.de