fredlcore/BSB-LAN #49: feature request: mDNS support

liudger, 2020-01-17:
support for Multicast DNS
This would make the device discoverable and automatically detectable for a home automation platform.
https://en.wikipedia.org/wiki/Multicast_DNS

fredlcore, 2020-01-17:
Do you have experience with multicast libraries for the Arduino? If you could supply some code snippets including the kind of information you would like to be broadcasted, that would help.

liudger, 2020-01-17:
I am no Arduino expert, when I am finished with the BSB-Lan integration for Home-Assistant I could take a look. But my only language I know  by a tiny bit is Python.

liudger, 2020-01-17:
found this with a quick search. https://gist.github.com/mqu/f8d4f6877703bb39676d68733801871d
the info to send is ip-address, port, device name and maybe base mac address to make it unique?
That would be enough for the beginning.

fredlcore, 2020-01-17:
It would already help to define what kind of information would be necessary for your task that BSB-LAN would need to provide...

fredlcore, 2020-01-17:
The example looks doable, but I wonder which port to use because 8266 sounds more like a demo/example port for the ESP8266. Or can this be choosen freely for your purpose?

liudger, 2020-01-17:
It should be the port that is defined in BSB_lan_config.h `define Port 80`
That is what I guess so the client can configure itself and communicate with BSB-Lan

fredlcore, 2020-01-17:
No, I mean the port that the broadcast is transmitted on (which then contains the ip-address, port, device name and mac address of BSB-LAN). I'm pretty sure that this can't be port 80 because that is reserved for webservers.

liudger, 2020-01-17:
Than your lucky number would be fine I guess;)

fredlcore, 2020-01-17:
I think that won't work because the program you want to use this broadcast with expects the broadcast message on a specific port. So when you want to use it with Home Automation, you should be able to identify the port where Home Automation expects the broadcast and then I would use this port as default.

liudger, 2020-01-17:
From wiki. The default is this it looks like UDP port 5353

```
Packet structure[edit]
An mDNS message is a multicast UDP packet sent using the following addressing:

IPv4 address 224.0.0.251 or IPv6 address ff02::fb
UDP port 5353
When using Ethernet frames, the standard multicast MAC address 01:00:5E:00:00:FB (for IPv4) or 33:33:00:00:00:FB (for IPv6)[a]
```

fredlcore, 2020-01-17:
Ah, thanks, that helps a lot! However, I just found out that the Arduino Ethernet library is not able to send UDP multicast packets, only the ESP could do so. Patches for the Arduino are flying around, but to include these would be futile once people update their libraries...
https://gist.github.com/aallan/9895961

BudBundi, 2020-06-05:
I found [https://github.com/TrippyLighting/EthernetBonjour](url)  and the fork [https://github.com/arduino-libraries/ArduinoMDNS](url) for mDNS on Arduino

fredlcore, 2020-06-05:
Thanks, someone in the forum also came up with a solution, I just haven't found the time to implement it...

hacki11, 2021-01-04:
A more simple approach would be to include the bsblan title in unauthorized html response, too:
Currently:
```
<!DOCTYPE HTML>
<HTML><HEAD><TITLE>Error</TITLE>
</HEAD> <BODY><H1>401 Unauthorized.</H1></BODY></HTML>
```
Solution:
```
<!DOCTYPE HTML>
<HTML><HEAD><TITLE>BSB-LAN Web</TITLE>
</HEAD> <BODY><H1>401 Unauthorized.</H1></BODY></HTML>
```

Now the device can be detected automatically (at least with iobroker)

fredlcore, 2021-01-04:
Should it say "BSB-LAN Web" or just "BSB-LAN"?

hacki11, 2021-01-04:
I would prefer the same title as from authorized site.

fredlcore, 2021-01-04:
I know, I'm just wondering if "BSB-LAN" fits in easier in external systems because this is what people might know. Then I'd change it in all titles.

rroblik, 2021-01-09:
@fredlcore , @hacki11 

does that mean the mDNS option is abandonned ? To my mind that would be the better option to follow. I see the W5500 is compatible for example...

hacki11, 2021-01-09:
Iobroker also supports detection over mdns and it would be the first choice I think. 
But changing the title was low effort for a first solution.

fredlcore, 2021-01-09:
No, it's not abandoned.

fredlcore, 2021-01-09:
Implemented now for Ethernet connections. WiFiSpi library somehow creates neverending repeating error messages. Not sure if this is a general issue or a problem of misconfiguration or bug.

closed 2021-01-09 by fredlcore