fredlcore/BSB-LAN #50: bug/feature: versioning tag support on github

liudger, 2020-01-17:
Make the firmware support semantic versioning.
https://semver.org
This makes updating saver. Now I am scared to update while developing ;)

fredlcore, 2020-01-17:
I agree that this totally makes sense, but the amount of commits which would each require a patch-level increase of the version number would have to be done somewhat automagically. If I were to upload binary versions which I would have to compile first before uploading, this would not be too difficult because I could use preprocessor directives.
If you come up with a (shell) script that would change the patch level automatically when I do a git push, then this could be a solution. Otherwise I know I'll be too lazy to think about this everytime I make a push...

liudger, 2020-01-17:
Could GitHub actions not do this for you?

fredlcore, 2020-01-17:
I don't know. I work with VS Code, if you point me to a resource which explains how (and what) needs to be done, I can try to implement it. The limited time for this project does not allow me to do all the research myself for functions that are not essential (to me ;) ).

liudger, 2020-01-17:
I have the same thing. Just want to do the essential.  The first custom component just worked fine for me and also for Holger (another homey) to some point. Than it broke with the next version. Than I just wanted to do a bit more proper version. Before I know I was writing native integration that needs proper versioning :(

Maybe with a make file. I use it to upload a version to pypi.org
found something in this article. Just scanned it quickly
https://learn.sparkfun.com/tutorials/efficient-arduino-programming-with-arduino-cli-and-visual-studio-code/all

But I am not sure what exactly you do want to script. Everytime you do a push you want to automatically update the version? 
I am stacking pushes till I have a proper working version and than make a version with all the changes and fixes. Maybe you can have a dev branch and merge it once in a while and create a version. The dev branch if for the developers who want to test out some new stuff or new device and master with versions is for the users who are already satisfied ? So this would update less regular.

fredlcore, 2020-01-18:
Yes, I would like to update the patch level version to increase by one each time I push. Often times I push the _defs.h when a new parameter is decoded by a user or if I fix a bug somewhere. I could go down the road of using a dev branch, but still the versioning would apply to that one as well. Haven't really worked with make files yet, so I guess that would be a good reason to do so. Will have a look at the link, thanks.

fredlcore, 2020-01-19:
Thanks for the link again, I found a way to do so not every time I push, but every time I build but that's fine as well. Will be coming with the next push.

closed 2020-01-19 by fredlcore