fredlcore/BSB-LAN #72: Posting to /JS causes

algirdasc, 2020-03-02:
Updated to latest version 0.44, and posting to JS causes my Arduino Mega 2560 to reboot. Using same config on version 0.43 issue could not be reproduced.

Post:
`{"Parameter":"10001","Value":"26.1","Type":"0"}`

Debug:
```
POST /JS HTTP/1.1
/JS
setting line: 10001 val: 06 86 00 
LAN->HEIZ INF      3E2E0215 06 86 00 
DC C2 00 0E 02 3E 2E 02 15 06 86 00 F9 81 
READY
Size of cmdtbl1: 31535
Size of cmdtbl2: 20417
free RAM:3669
Reading EEPROM...
192.168.8.210
Waiting 3 seconds to give Ethernet shield time to get ready...
LAN->HEIZ QUR 6225 Configuration -  Device family: 
DC C2 00 0B 06 3D 05 00 02 52 88 
HEIZ->LAN ANS 6225 Configuration -  Device family: 123
DC 80 42 0E 07 05 3D 00 02 00 00 7B FD 64 
#6225: 123
LAN->HEIZ QUR 6226 Configuration -  Device variant: 
DC C2 00 0B 06 3D 05 00 03 42 A9 
HEIZ->LAN ANS 6226 Configuration -  Device variant: 231
DC 80 42 0E 07 05 3D 00 03 00 00 E7 C9 E5 
#6226: 231
Device family: 123
Device variant: 231
```

fredlcore, 2020-03-02:
@dukess : Could you have a look at this since this seems to be happening since I incorporated your JavaScript changes?

dukess, 2020-03-03:
Sure, i will check it. Looks like variables intersection.

fredlcore, 2020-03-03:
@algirdasc : @dukess has now fixed the error, could you please install the most current master branch and see if it works now? Thanks!

algirdasc, 2020-03-03:
@fredlcore , @dukess looks good:
```
POST /JS HTTP/1.1
/JS
setting line: 10002 val: 04 DC 00 
LAN->HEIZ INF      3F2F0215 04 DC 00 
DC C2 00 0E 02 3F 2F 02 15 04 DC 00 8B 54 
Setting parameter 10002 to 19.45 with type 0
                                            setting line: 10000 val: 05 21 00 
LAN->HEIZ INF      3D2D0215 05 21 00 
DC C2 00 0E 02 3D 2D 02 15 05 21 00 32 5A 
Setting parameter 10000 to 20.53 with type 0
                                            setting line: 10001 val: 06 73 00 
LAN->HEIZ INF      3E2E0215 06 73 00 
DC C2 00 0E 02 3E 2E 02 15 06 73 00 15 B5 
Setting parameter 10001 to 25.8 with type 0
```

fredlcore, 2020-03-03:
Thanks for testing, will close now...

closed 2020-03-03 by fredlcore