fredlcore/BSB-LAN #75: Möglichkeit, das auf einen esp8266 oder esp32 zu portieren?

red-gee, 2020-03-26:
Hallo besteht die Möglichkeit das auch für einen ESP8266/ESP32 zurecht zu stricken? Es wäre schön, da der ESP ja auch die direkte Möglichkeit hat, ihn über WLan anzusprechen.

fredlcore, 2020-03-26:
Theoretisch vielleicht, aber warum? Dem ESP8266 fehlt ein zweiter serieller Port, und ein ESP würde all diejenigen ausschließen, die aus Sicherheitsgründen oder wegen des schlechten WLAN-Empfangs unter dem Dach oder im Keller (oder wo auch immer die Heizung steht) kein LAN verwenden können. Für 10-20 Euro bekommt man eine WLAN-LAN-Bridge, mit der man auch den Arduino ans WLAN bringen kann, und das sehr viel stabiler, als es dem ESP möglich wäre.

closed 2020-03-26 by fredlcore

red-gee, 2020-03-27:
danke für deine Antwort

1coderookie, 2021-02-07:
@red-gee: Mittlerweile sind zwei Möglichkeiten bzgl einer ESP-Verwendung gegeben:
1. Arduino Due + Adapter + ESP8266 anstelle des LAN-Shields
2. ESP32 + Adapter anstelle von Due+LAN-Shield/ESP8266
Wenn du also eine reine WLAN-Anbindung wünschst, dann ist vermutlich die zweite Variante empfehlenswert. Für eine bestimmte ESP32-NodeMCU-Boardvariante gibt es jetzt auch einen spezifischen Adapter. Alle Infos findest du im Handbuch im [Kap. 12](https://1coderookie.github.io/BSB-LPB-LAN/kap12.html)