fredlcore/BSB-LAN #76: Schreiben von Werten

hacki11, 2020-03-28:
Ich möchte beim iobroker Adapter das Schreiben von Werten hinzufügen und hätte ein paar Fragen. Vielleicht könnt ihr mir die ja beantworten.

Ich würde POST http://<IP-Adresse>/JS benutzen.

Ist es möglich über die API zu erkennen, ob ein Wert beschreibbar oder read-only ist? (unabhängig von der Freischaltung in der Firmware) Es gibt einen Errorresponse für read-only, ich würde es aber gerne vor dem Request wissen.

Woher weiß ich, welcher Wert mit INF und welcher mit SET beschrieben werden muss? 

Ihr schreibt, geschrieben werden können nur numerische Werte. Welche Datentypen zählt ihr da genau dazu? "DataType" (0 = Zahl, 1 = ENUM, 2 = Bit-Wert (Dezimalwert gefolgt von Bitmaske gefolgt von ausgewählter Option), 3 = Wochentag, 4 = Stunde/Minute, 5 = Datum/Uhrzeit, 6 = Tag/Monat, 7 = String, 8 = PPS-Uhrzeit (Wochentag, Stunde:Minute))  

Vielen Dank!

fredlcore, 2020-03-28:
> Ist es möglich über die API zu erkennen, ob ein Wert beschreibbar oder read-only ist?
Nein, dazu müsste dies bei jedem Parameter hinterlegt werden, was bei dem begrenzten Speicherplatz nicht sinnvoll wäre.
> Woher weiß ich, welcher Wert mit INF und welcher mit SET beschrieben werden muss?
Alle regulären Parameter werden mit SET geschrieben. INF wird bisher nur sinnvoll beim Übermitteln der Raumtemperatur verwendet.
> Welche Datentypen zählt ihr da genau dazu?
0 direkt, 1 und 2 wenn der Dezimalwert des ENUMs bzw. des Bit-Wertes übermittelt wird, 4 wenn statt des Doppelpunkts ein Punkt verwendet wird. 8 kann sowieso nicht geschrieben werden, da dies von der Heizung kommt.

Alles ohne Gewähr, da bisher noch keiner JSON POST verwendet hat. Über Rückmeldungen was (nicht) geht, freuen wir uns.

closed 2020-03-28 by fredlcore

hacki11, 2020-03-29:
Wie entscheidest du, ob ein Set Button angezeigt wird oder nicht? Müsste dafür nicht irgendwo hinterlegt sein was man beschreiben kann?
[image]

fredlcore, 2020-03-29:
Das wird über ein Flag bei jedem Parameter geregelt, der aber nichts darüber aussagt, ob der Parameter an sich beschreibbar ist, sondern wie der User BSB-LAN konfiguriert hat. Per Default sind alle Parameter auf read-only gesetzt, darüber hinaus können OEM-Parameter als beschreibbar definiert werden oder es können generell alle Parameter schreibbar gemacht werden. Das sagt aber nichts darüber aus, ob man für den Parameter am Ende nicht doch eine Schreib-Fehlermeldung bekommt.

hacki11, 2020-03-30:
Hast du einen Rat bei Parameter 0, dataType: 5

Hab folgende Varianten versucht:

```
{
	"Parameter": "0",
	"Value": "19.47.00",
	"Type": "1"
}

{
	"Parameter": "0",
	"Value": "30.03.2020 19.44.00",
	"Type": "1"
}
```
Jeweils Errorcode 0

fredlcore, 2020-03-30:
Datatype 5 geht nicht.

hacki11, 2020-03-30:
Ok, hab jetzt 0, 1 und 4 erfolgreich getestet.
Bei Bit habe ich nur Status IDs gefunden, ich denke die sind eh nicht beschreibbar.
Ich nehm als Basissatz für read-only Werte mal die Daten aus deinem Standardheader File.

Ich denke damit hab ich das meiste erschlagen und gleichzeitig die meisten readonly Parameter auch so in iobroker.
Vielen Dank, damit sollte man nun auch aus iobroker heraus steuern können.

fredlcore, 2020-03-30:
Gern, Bit-Werte, die man beschreiben könnte, gibt es u.a. im LPB-Menü.