fredlcore/BSB-LAN #79: DateTime über MQTT

hvorragend, 2020-03-31:
In einigen Parameter (bei mir 6800, 6810, 6820, 6830, etc.) hat meine Therme Datums- und Uhrzeit gespeichert. Über die Weboberfläche wird die Uhrzeit auch angezeigt. Über MQTT wird leider nur das Datum übertragen.

Könnte man bitte über MQTT Datum und Zeit übermitteln? 

Danke.

dukess, 2020-03-31:
Hi! 
Try to add parameter No. 0 in log_parameters array in BSB_lan_config.h

hvorragend, 2020-03-31:
Really? How can an additional parameter 0 change the format of another parameter?

dukess, 2020-03-31:
Oops. My mistake.
I look on code. I think this is because the date and time string contains a whitespace.
MQTTClient.publish(MQTTTopic.c_str(), strtok(query(log_parameters[i],log_parameters[i],1)," "));

fredlcore, 2020-04-02:
Yes, that is the problem. Some users wanted the unit to be stripped from the actual value, so the general approach was to drop everything following the first whitespace. In the case of date and time, this obviously does not work and I'll have to hard-code this into the JSON output.

fredlcore, 2020-04-02:
New version online, please check with current master branch if date and time are now transmitted correclty via MQTT and report back here.

hvorragend, 2020-04-04:
Again, thank you very much. It works perfectly fine.

closed 2020-04-04 by hvorragend