fredlcore/BSB-LAN #83: System seems to crash/freeze when SD card is full after extensive logging.

fredlcore, 2020-05-02:
Check if it's possible to query free SD card space before writing (and hope it does not take too long)...

dukess, 2020-06-02:
SdFat library has a function SD.vol()->freeClusterCount() that counts free space.
When #define MAINTAIN_FREE_CLUSTER_COUNT 0 every call spend ~5 sec per call for 16Gb SD card.
freeClusterCount() call time: 4909960 micros
freeClusters: 474234
freeSpace: 15175 MB (MB = 1,000,000 bytes)

When #define MAINTAIN_FREE_CLUSTER_COUNT 1 first call is slow, but all other is faster:
freeClusterCount() call time: 48 micros
freeClusters: 474234
freeSpace: 15175 MB (MB = 1,000,000 bytes)

fredlcore, 2020-06-02:
The latter sounds good for me!

fredlcore, 2020-06-02:
Maybe we could do the first call during setup? Then no one would recognize the delay?

dukess, 2020-06-03:
It is the best. Can we combine counting free space and waiting for network initialization? Something like this

```
  Serial.println(F("Waiting 3 seconds to give Ethernet shield time to get ready..."));  // ...and flash the LED during that time...

  pinMode(LED_BUILTIN, OUTPUT);

  uint32_t diff = 0;
  #if defined LOGGER || defined WEBSERVER
  Serial.print(F("Calculating free space..."));
  uint32_t m = millis();
  uint32_t volFree = SD.vol()->freeClusterCount();
  uint32_t fs = (uint32_t)(volFree*SD.vol()->blocksPerCluster()/2048);
  sprintf(outBuf, "%lu MB free\n", fs);
  Serial.print(outBuf);
  outBufclear();
  diff = (millis() - m)/1000;
  #endif

  for (int i=0; i<(3 - diff + 1); i++) { // 3 seconds guaranteed
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(500);                       // wait for a second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    delay(500);                       // wait for a second
  }
  server.begin();
```

dukess, 2020-06-03:
I can answer this question myself: my version works on Mega. At least for me.

fredlcore, 2020-06-04:
Great, works for me too now, no idea what was the problem back then, but then I'd suggest to completely switch to SdFat.h. For the 3s wait, one could even start blinking the LED before the calculation, store the current millis() and then only wait for the remaining three seconds (if any left). Or, if it's always longer than 3s, we could do the flashing of the LED in an #else statement?

dukess, 2020-06-04:
>Or, if it's always longer than 3s,

I think it depend on SD card size. I can assume that it will be 8 times faster for 2 GB cards.
So i cat propose to rewrite flashing code: turn LED on, then wait for calculation and/or initialization (if calculation is faster than 3 sec i just call delay up to 3 sec), LED off and then blink twice.

```
  Serial.println(F("Waiting 3 seconds to give Ethernet shield time to get ready..."));
  // turn the LED on until Ethernet shield is ready and freeClusterCount is over
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)

  long diff = 3000;
  #if defined LOGGER || defined WEBSERVER
  Serial.print(F("Calculating free space on SD..."));
  uint32_t m = millis();
  uint32_t volFree = SD.vol()->freeClusterCount();
  uint32_t fs = (uint32_t)(volFree*SD.vol()->blocksPerCluster()/2048);
  strcpy_P(outBuf, PSTR("%lu MB free\n"));
  char *p = outBuf + strlen(outBuf) + 1; //add zero to split strings
  sprintf(p, outBuf, fs);
  Serial.print(p);
  outBufclear();
  diff -= (millis() - m); //3 sec - delay
  #endif
  if(diff > 0)  delay(diff);

  //decoration: double blink by LED before start. wait for a second
  for (int i=0; i<2; i++) {
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    delay(250);
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(250);
  }
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  //end of decoration
  server.begin();
```

fredlcore, 2020-06-04:
Great!

dukess, 2020-06-04:
Added in patch https://github.com/fredlcore/bsb_lan/pull/100/commits/b2b767724796a64840a5f87b661ac9ec9594ff91

closed 2020-06-04 by fredlcore