fredlcore/BSB-LAN #86: Graph funktioniert nicht / Graph not working

XL-Reaper, 2020-05-09:
Wenn ich /DG öffne, wird bei mir kein Graph angezeigt. In der Browser Console kann ich folgenden Javascript Fehler sehen:
When opening /DG I don't get any graphed data. I can see the following error in the browser's console:

d3.v3.min.js:1 Uncaught TypeError: Cannot read property 'length' of undefined
    at e (d3.v3.min.js:1)
    at t.parse (d3.v3.min.js:1)
    at DG:104
    at Array.forEach (<anonymous>)
    at DG:103
    at d3.v3.min.js:1
    at Object.<anonymous> (d3.v3.min.js:1)
    at Object.t (d3.v3.min.js:1)
    at XMLHttpRequest.i (d3.v3.min.js:1)

Meine Logdatei einthält aktuell die folgenden Einträge:
My logfile contains this data currently:

3600000;09.05.2020 20:35:51;8700;Außentemperatur;17.5;&deg;C
3600237;09.05.2020 20:35:51;8743;Vorlauftemperatur 1;---;&deg;C
3600444;09.05.2020 20:35:51;8314;Kesselrücklauftemperatur Ist;---;&deg;C
3600660;09.05.2020 20:35:51;20000;Brennerlaufzeit Stufe 1;193
3600663;09.05.2020 20:35:51;20001;Brennertakte Stufe 1;2
3600665;09.05.2020 20:35:51;20002;Brennerlaufzeit Stufe 2;545
3600667;09.05.2020 20:35:51;20003;Brennertakte Stufe 2;2
7200685;09.05.2020 21:36:47;8700;Außentemperatur;15.5;&deg;C
7200917;09.05.2020 21:36:47;8743;Vorlauftemperatur 1;---;&deg;C
7201107;09.05.2020 21:36:48;8314;Kesselrücklauftemperatur Ist;---;&deg;C
7201312;09.05.2020 21:36:48;20000;Brennerlaufzeit Stufe 1;193
7201315;09.05.2020 21:36:48;20001;Brennertakte Stufe 1;2
7201317;09.05.2020 21:36:48;20002;Brennerlaufzeit Stufe 2;545
7201319;09.05.2020 21:36:48;20003;Brennertakte Stufe 2;2

XL-Reaper, 2020-05-11:
This is actually a documented issue in the Handbook:

Die Ausführung des URL-Befehls /D0 sollte außerdem bei der ersten Benutzung erfolgen, da hierdurch die Datei mit dem passenden
CSV-Header initiiert wird

closed 2020-05-11 by XL-Reaper

fredlcore, 2020-05-11:
Ah, ok, what a relief, bugs one cannot reproduce are always the hardest to fix ;)...