fredlcore/BSB-LAN #91: feature request - json request for MAC address and firmware version

liudger, 2020-05-17:
For home-assistant integration I need an identifier so I can check if the integration is already setup or not. For this often a Mac address is used. So it would be handy to get the Mac address just from a json request . (or is this already possible without me knowing?)

Also the firmware would be nice to pull this way. If this is possible it would be awesome.

Now I use the model nr. of the heating system, but it's not the correct way to do it. You could have multiple heating systems and than this setup wouldn't work anymore.

fredlcore, 2020-05-18:
The firmware version would be possible to add to the JSON export as some kind of virtual parameter for example, but the MAC address is a problem. It is set by the software and so far all BSB-LAN installations have the same MAC address ;). This was because so far it was not possible to read the MAC address from the Ethernet Shield (at least version 1, haven't tried with version 2 because I don't have one). 

Also a random generated number would have to be stored in EEPROM, checked if it exists, and if yes, not be generated again. And this part of the EEPROM would have to be protected from a "clean EEPROM" call. Then again, if you want to reset everything, this EEPROM part would also have to be reset.

But maybe it does not hurt that the MAC address is always the same as long as you change it manually (which you can) when you have more than one device running?

liudger, 2020-05-18:
I think this will do just fine. 

thanks

fredlcore, 2020-05-18:
Ok, so I'll send the Mac address via JSON.

fredlcore, 2020-05-19:
@dukess: Since you have been working on the JSON output for a while (and I'm not sure if I would break anything), I was wondering what you think about this? Maybe we could add a new /Jx command for outputting (some of) the configuration, including the MAC address?

dukess, 2020-05-20:
> Now I use the model nr. of the heating system, but it's not the correct way to do it. You could have multiple heating systems and than this setup wouldn't work anymore.

Can we get serial number from heater/cooler with address 0 immediatly after start and set MAC with this value? 

> Maybe we could add a new /Jx command for outputting (some of) the configuration, including the MAC address?
/JI (JSON Information) key will be good idea. 
I think we can send all info like /C function.

liudger, 2020-05-20:
> Can we get serial number from heater/cooler with address 0 immediately after start and set MAC with this value?

I am not sure what you mean? So `Serial Heater``0``mac address`? 
This shouldn't be needed. Home assistant makes a map/network. So if we can identify the bsblan board it shouldn't be an issue that we have 2 the same serial numbers for the 2 heaters. if they are connected 2 different bsblan boards.

liudger, 2020-05-20:
> /JI (JSON Information) key will be good idea.
I think we can send all info like /C function.

sounds great!

fredlcore, 2020-05-20:
I think what @dukess means is not to set the MAC address from the _config.h file but read the serial number of the heater and generate a MAC address from that.
However, this would mean that the query of the heating system would have to happen before the network is being set up. And more importantly, it would not work on PPS heating systems which do not have this kind of parameter (and I think those who use an OCI420 on an LMU64 as well).

dukess, 2020-05-20:
> However, this would mean that the query of the heating system would have to happen before the network is being set up. 

I hope that's not a problem.

> And more importantly, it would not work on PPS heating systems which do not have this kind of parameter (and I think those who use an OCI420 on an LMU64 as well).

For these cases we can use pre-defined MAC from _config.h.

fredlcore, 2020-05-20:
Well, it's at least inconvenient for those (like me ;) ) who have diverted the output from serial to telnet and thus can remotely get the "serial" output. For debugging, the intial sequence is often important.

dukess, 2020-05-20:
This is real problem. So I'll leave this idea and try to make a /JI command.

fredlcore, 2020-05-20:
👍

dukess, 2020-05-21:
I wrote this function, but I can only test it tomorrow or the day after.
https://github.com/dukess/bsb_lan/commits/master/BSB_lan.ino

dukess, 2020-05-22:
Ready for merging: https://github.com/fredlcore/bsb_lan/pull/92

{
  "version": "0.44.11-20200410111601",
  "freeram": 3049,
  "uptime": 115800,
  "MAC": "00:80:41:19:69:90",
  "bus": "BSB",
  "buswritable": 1,
  "busaddr": 66,
  "busdest": 0,
  "monitor": 0,
  "verbose": 1,
  "protectedGPIO": [
    { "pin": 10 },
    { "pin": 11 },
    { "pin": 12 },
    { "pin": 13 },
    { "pin": 50 },
    { "pin": 51 },
    { "pin": 52 },
    { "pin": 53 },
    { "pin": 62 },
    { "pin": 63 },
    { "pin": 64 },
    { "pin": 65 },
    { "pin": 66 },
    { "pin": 67 },
    { "pin": 68 },
    { "pin": 69 }
  ],
  "averages": [
    { "parameter": 8700 },
    { "parameter": 8326 }
  ],
  "loginterval": 3600,
  "logged": [
    { "parameter": 8700 },
    { "parameter": 8314 },
    { "parameter": 8830 }
  ]
}

fredlcore, 2020-05-22:
Wow, that went fast, great, thanks a lot!

fredlcore, 2020-05-22:
Works all in all, I think, but Firefox' JSON parsers gives me the error
`SyntaxError: JSON.parse: unexpected character at line 37 column 3 of the JSON data`

for this output:

```
{
  "version": "1.0.3-20200522224028",
  "freeram": 0,
  "uptime": 9531,
  "MAC": "00:80:41:19:69:90",
  "bus": "BSB",
  "buswritable": 1,
  "busaddr": 66,
  "busdest": 0,
  "monitor": 0,
  "verbose": 1,
  "protectedGPIO": [
    { "pin": 10 },
    { "pin": 11 },
    { "pin": 12 },
    { "pin": 13 },
    { "pin": 50 },
    { "pin": 51 },
    { "pin": 52 },
    { "pin": 53 },
    { "pin": 62 },
    { "pin": 63 },
    { "pin": 64 },
    { "pin": 65 },
    { "pin": 66 },
    { "pin": 67 },
    { "pin": 68 },
    { "pin": 69 }
  ],
  "averages": [
    { "parameter": 8700 },
    { "parameter": 8323 },
    { "parameter": 8326 }
  ],
  "loginterval": 3600,
  "logged": 
  ]
}
```
Line 37 would be the second last line (the closing square bracket). It looks as if this closing bracket is without a corresponding opening bracket?

BudBundi, 2020-05-22:
> Works all in all, I think, but Firefox' JSON parsers gives me the error
> `SyntaxError: JSON.parse: unexpected character at line 37 column 3 of the JSON data`
>
> for this output:
>
> ```
> {
>   "version": [...]

Seems that it shout be opened by logged. But you have no log parameters set.

fredlcore, 2020-05-22:
Exactly.

dukess, 2020-05-22:
Ok. Empty list will be not shown.

fredlcore, 2020-06-04:
MAC address is now shown in /JI so I'll close this one here...

closed 2020-06-04 by fredlcore