fredlcore/BSB-LAN #96: put translations in a separate library

liudger, 2020-05-24:
Hi all,

Isn't it an idea to put the translation in a separate json files? So you don't need to update the firmware for new translations. Only the dependancy version?

Also we could reuse these files in all platforms? For example: when users from homeassistant get German translation they contact me I need to send it to this project and than it needs to update here get released and the user need to update the firmware etc. ( I see this bsblan module more as a hub and I guess the translations should be done by the platforms.)

So I could start doing the translations on Home-assistant and ignore the `desc` output. But this would be double the work and bsblan also doesn't get these translations. So maybe centralising would be a good idea?

What do you think?

dukess, 2020-05-25:
> Isn't it an idea to put the translation in a separate json files? So you don't need to update the firmware for new translations. Only the dependancy version?

Technically, we can make JSON files with strings, but we need to clarify how we will work with them: 1. the language in the settings will have to be set as LANG_C; 2. the interface in this case will look like this:
[image]
3. Linking IDs and localized strings is performed in programs.

> Also we could reuse these files in all platforms? For example: when users from homeassistant get German translation they contact me I need to send it to this project and than it needs to update here [...]

Yes, this is possible, but only if no new functions, enumerations, descriptions, or interface elements are added.

> ...and ignore the `desc` output. 

Do not do this, because, for example, some enumeration elements have strings with IDs that do not match the item numbers.

>But this would be double the work and bsblan also doesn't get these translations. So maybe centralising would be a good idea?

I think yes.

Why didn't I use JSON files in my project? Because parsing these files for non-German localization is a bit more complicated and memory-demanding.

Format can be:
{
"MENU_TEXT_HFK": "Heizungsfunktionen",
"STR1_TEXT": "Stunden/Minuten",
"STR700_TEXT": "Betriebsart",
"ENUM8004_18_TEXT": "Frostschutz aktiv",
}

fredlcore, 2020-05-25:
This would mean we make an SD card a requirement (including the process of putting specific data in specific places/files/directories on that SD card. While I like having the option of what @dukess has created with AJAX etc., I want BSB-LAN to be as easy and with as little action on the user's side as possible. Also, with the Due, we have quite a lot of Flash now available that does not force us to to save any byte possible anymore.

What I don't understand is what you mean with "when users from homeassistant get German translation they contact me I need to send it to this project". What kind of translation from what language do you get from them and why? The parameter names should resemble the ones on the heater, there should be no need to use different translations in different platforms. Or did I miss something here?

dukess, 2020-05-26:
Frederik, can we add another translation format (now in the form of JSON) based on LANG_*. h? People will be able to download these files from the repository themselves and use them at their own discretion. JSON is the standard, unlike my JavaScript trick. Or is there no desire to increase the repository? 
Here is create_undefs with JSON output: https://github.com/dukess/bsb_lan/pull/15

fredlcore, 2020-05-26:
Everything that does not break the current way of having the SD card completely optional is fine with me. But we need to keep track of these new possibilities also in the manual, because once neither me or @1coderookie are using a specific function, its usage could be easily forgotten.

dukess, 2020-05-26:
Understand you. 
https://github.com/fredlcore/bsb_lan/pull/98

dukess, 2020-05-26:
For 8.2.4 section of manual:
There are also localizations in JSON format that can be used when integrating with third-party software. The files are located in the "json" directory of the repository. File format: "string_identifier": "Text", where string_identifier is the value set via the LANG_C.h file. When using" LANG C " in the USB_lan_config file.h instead of strings, their IDs will be issued, for example:
{
  "700": {
    "name": "STR700_TEXT",
    "possibleValues": [
      { "enumValue": "0", "desc": "ENUM700_00_TEXT" },
      { "enumValue": "1", "desc": "ENUM700_01_TEXT" },
      { "enumValue": "2", "desc": "ENUM648_01_TEXT" },
      { "enumValue": "3", "desc": "ENUM700_03_TEXT" }
    ],
    "isswitch": "0",
    "dataType": 1,
    "readonly": 0
  },
}
This way, you can link text in any available language in the user program without changing the device's firmware.

dukess, 2020-06-12:
Here is language resources in JSON: https://github.com/fredlcore/bsb_lan/tree/master/localization/json

fredlcore, 2020-06-13:
Thanks, does it mean we can close this one now, @liudger?

liudger, 2020-06-13:
For now I guess. I am not working on the translations yet. So I have yet to see how well it will work.

Lately too much other stuff is going on. Not much time for development

Liudimation

Willem-Jan L. van Rootselaar
3D Generalist|Character Animator
mobile: +31(0) 6 444 04 363
messages/facetime: +31(0) 6 444 04 363
skype: liudger

www.liudimation.com

fredlcore, 2020-07-31:
Ok, I close this now for the time being, feel free to re-open whenever the need arises.

closed 2020-07-31 by fredlcore