fredlcore/BSB-LAN #102: Enhance network performance and code reusability

loetmeister, 2020-06-04:
Hi,

based on the discussion around #97 I'm opening this issue to not loose some of the good ideas to enhance network traffic and source code clean up, that could be beneficial in future features.
E.g. the consistent use of output buffering, before transmitting the data to the network module could reduce the amount of packets (and TCP packet overhead) in scales of 155% easily.

> Comrades, I decided to stretch my brain and redo the program as I described above. A little testing showed that the idea works, but there may be errors in the code.
> https://github.com/dukess/bsb_lan/commit/8344657b1f0a3eaed116f74cee7ac1c88229a378
> 
> Benefits: + 300 free bytes of RAM.
> Disadvantages: -100 bytes of flash
> 
> _Originally posted by @dukess in https://github.com/fredlcore/bsb_lan/pull/97#issuecomment-636049720_
> 

PS: Revolution, baby! ;-)

cheers

dukess, 2020-06-05:
Revolution, definitely! :-) But first we must defeat the evil bugs: SdFat library doesn't want to work on Due.

dukess, 2020-06-11:
New patch: https://github.com/fredlcore/bsb_lan/pull/108

loetmeister, 2020-06-11:
Very nice. Will try it out... 
I had started to reduce the amount of #ifdef's, like 
https://github.com/loetmeister/bsb_lan/commit/9a4f00a2a0c33c4dc3fdc9579ce835c05e668074. The various defines make the code quite hard to read sometimes. In particular where they are nested.
One other thing I tried... I created a function that handles dht22 read and debug output https://github.com/loetmeister/bsb_lan/commit/c6eee7acd4bb68043a5a14c536004abb78182122, this function seem to be compiled inline, I don't get why..

loetmeister, 2020-07-12:
I've noticed a nice way to avoid use of pointer and reduce code lines, by passing PROGMEM text to sprintf_P in one go...
two lines into one:

```
//            strcpy_P(formatbuf, PSTR("  \"monitor\": %d,\n  \"verbose\": %d")); //note: no \n here! Look for it below
//            outBufLen+=sprintf(outBuf+outBufLen, formatbuf, monitor, verbose);
            outBufLen+=sprintf_P(outBuf+outBufLen, (PGM_P)F("  \"monitor\": %d,\n  \"verbose\": %d"), monitor, verbose);
```
or to replace this, as it previoulsy would sit in RAM:
```
      //outBufLen+=sprintf(outBuf+outBufLen,"<tr><td>\ntemp[%d]: ",i);
      outBufLen+=sprintf_P(outBuf+outBufLen,(PGM_P)F("<tr><td>\ntemp[%d]: "),i);
```

dukess, 2020-07-12:
strange how I missed this function.
This is good idea.
I propose to use PSTR() instead (PGM_P)F()

dukess, 2020-07-21:
https://github.com/fredlcore/bsb_lan/pull/120

dukess, 2020-07-22:
Today i made patch that increase perfomance and making code more readable (i hope for this).
Reviews are welcome.
https://github.com/dukess/bsb_lan/commit/8159a1b0b64cc8761de9bab4d784342460015cd5

loetmeister, 2020-07-23:
Hi,

I like it. Flushing the buffer automatically is helpful.
Not sure about the names of the new functions, they don't reflect what they really do... Buffered print to the web client. However from the current naming scheme, I get the idea.

Would it be helpful to have a replacement of `client.println`, to append the new line automatically?

dukess, 2020-07-23:
thank you!

printToWebClient(), printToWebClient_P(), printToWebClient_PF(), printFmtToWebClient_P()
These names looks better?

About writeln... The easiest thing is to write wrapper functions like this:
```
int printlnToWebClient_PF(uint_farptr_t src){
  return printToWebClient_PF(src) + writelnToWebClient();
}
```
But personally, it's easier for me to put "\n " at the end of the line.

fredlcore, 2020-07-23:
Since we should address the Due in general in the code layout, wouldn't it be better to omit the _P* functions and just use the standard names and add the Mega related adjustments within these functions via #ifdefs?

dukess, 2020-07-24:
I had concerns that there would be a mix of data types, but it seems they were groundless.
Now is only printToWebClient() printFmtToWebClient() functions and macros #define printlnToWebClient(format) (printToWebClient(format) + writelnToWebClient()) istead function above.
https://github.com/dukess/bsb_lan/commit/b1b912cbd73cbe918364f5d9c9dfa70f118bf15f

fredlcore, 2020-07-24:
Looks good to me, thanks! I personally would have also used the "\n" at the end of the line, but both ways are fine with me.

fredlcore, 2020-07-24:
One more thing I was thinking  is whether we should check for the AVR in the ifdef condition and put the SAM code in the #else part. It seems to me that the AVRs are the only ones using PROGMEM and the more modern processors do not have this separation. This way people could compile the code on other platforms like the ESP (if they really want to, but I would not recommend it) without major changes. Or is it the other way round that the SAM is one of the few **not** using PROGMEM? Then of course it would make sense if things stay as they are.

dukess, 2020-07-24:
https://github.com/dukess/bsb_lan/commit/b91b3e03d1fe207395bc553440b6111cf042c91a
No problem but we need to change it everywhere.

fredlcore, 2020-07-24:
Yes, I see that now, so maybe better not to change a running system (until we have enough time to do so ;) )...

dukess, 2020-07-24:
Done it. Just swapped strings where it need. And here only three places with CPU-depended behavior: reset function, EEPROM access.
https://github.com/dukess/bsb_lan/commit/8bf3ea54a4c0ce1c2bde2e02baad610fd97275a0

fredlcore, 2020-07-24:
Wow, awesome! Thanks!

fredlcore, 2020-07-24:
Is it possible to get rid of these compiler warnings or is the compiler just too dumb?

```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void LogTelegram(byte*)':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:2966:11: warning: unused variable 'type' [-Wunused-variable]
   uint8_t type=0;
           ^~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void query(int, int, boolean)':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:4121:32: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      if(decodedTelegram.prognr == -1) continue;
         ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void loop()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:6996:22: warning: unused variable 'c' [-Wunused-variable]
             uint32_t c=0;
```

```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'int findLine(uint16_t, uint16_t, uint32_t*)':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:1086:14: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (temp == line)
              ^
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:1093:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     else if (temp > line)
                   ^
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void LogTelegram(byte*)':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:2966:11: warning: unused variable 'type' [-Wunused-variable]
   uint8_t type=0;
           ^
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void query(int, int, boolean)':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:4121:32: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      if(decodedTelegram.prognr == -1) continue;
                                ^
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void loop()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:6996:22: warning: unused variable 'c' [-Wunused-variable]
             uint32_t c=0;
                      ^
```

dukess, 2020-07-24:
It's not compiler too dumb, it's me: Arduino IDE not shown any warnings by default. :-)

fredlcore, 2020-07-31:
I would like to close this issue now so that the discussion is not spread across several issues and/or pull requests. Everyone, feel free to open a new issue when the need arises!

closed 2020-07-31 by fredlcore