fredlcore/BSB-LAN #106: Move all configuration from BSB_lan_config.h to webinterface, stored on EEPROM

fredlcore, 2020-06-07:
In order to be able to distribute compiled binaries of BSB-LAN, the complete configuration should be possible via the webinterface. Most parameters can already be configured during runtime via URL commands, so these could be called in the background via JavaScript etc.
More problematic could be the various #define configurations, but since we now have enough space on the Due, all functional parts (de-)activated via #defines should be enabled by default and could then be disabled on the Mega if necessary.

All data should be stored on EEPROM which is provided for the Due on the V3 BSB-LAN boards and has the same size as the one built-in in the Mega: 4096 bytes.

The EEPROM storage should be divided into two sections: 
First section: Fixed length configuration options, such as pin numbers for OneWire bus, DHT22, IP-address, PPS parameters etc.
Second section: Divided into two parts for configuration options of (at compile time) unknown length, such as log_parameters array, passkey, exclude_GPIO. First part is a table layouted like this:
Byte 1+2: Offset to actual data
Byte 3: Type of data (may not be necessary if EEPROM.get() is able to determine this in all cases)
Byte 4: Length of data (or length of entries, if EEPROM.put()/get() can be used at all times)
Byte 5 to Byte 5+length: actual data

Since all configuration data is always written at the same time, the actual distribution of data in the second part of the second section can vary, but due to the lookup-table, data can easily be located.

In order to maintain backward compatibility, variables set via BSB_lan_config.h should be written to EEPROM during setup, and then a warning should be printed in the web-configuration that all changes will be overwritten upon reboot as long as the entries in BSB_lan_config.h remain.

Most of the entries probably do not need default values, but for the network part, it would be best to set the device to DHCP mode instead of setting some kind of fixed IP address (although this would probably also be acceptable).

I hope I've covered all caveats, but I guess I have not ;)...

dukess, 2020-06-09:
>In order to maintain backward compatibility, variables set via BSB_lan_config.h should be written to EEPROM during setup, and then a warning should be printed in the web-configuration that all changes will be overwritten upon reboot as long as the entries in BSB_lan_config.h remain.

Just make a flag, that will be set to EEPROM at first run and prevent settings implementation from BSB_lan_config.h

fredlcore, 2020-06-09:
Good idea!

fredlcore, 2020-07-31:
Do you think a struct containing all the settings would be useful? This could probably be written in one go with EEPROM.put and could be read also in one go with EEPROM.read. 

The best way would be to also add information about what kind of data an entry consists of. So whether it is Checkbox, number, dropdown... If the code could then evaluate this and build a website with the correct input fields, it would be very easy to add new configuration options later on. Any ideas?

dukess, 2020-08-01:
The only thing I don't like about the structure idea is that it will be recorded all at once. But the number of write cycles of EEPROM is very limited, AFAIK. Or am I worrying too much about this for nothing?

dukess, 2020-08-01:
We can write a function that, when writing new parameters, compares the contents of the structure byte by byte and writes only the changed data.

fredlcore, 2020-08-01:
No, you are right, of course, because some options will be changed more often than others. But a function like you suggested could take care of that. The (literally) bigger problem would be the amount of RAM needed because the whole structure would have to be kept in RAM (possibly all the time). For the Due that's not an issue (4kB of EEPROM size vs. 96kB of Due RAM), but for the Mega it would not be possible. That would either be a point of departure where the Mega code is kept as it is, or the _config.h would be kept and loaded into RAM after the config is read from the EEPROM.

On my other project (robotan.de) I use a struct to access the EEPROM of the lawn robot so that I know where each configuration option is located. Maybe that's a more feasible option?

dukess, 2020-08-02:
I do not think that there will be any problem with the availability of free memory, since almost all settings stored in RAM now.
About structure: of course, for parameters whose length is known, the structure is the best solution.

fredlcore, 2020-08-02:
Yes, that's true, but if we want to use the whole 4k EEPROM size of the Mega (or the EEPROM on the V3 BSB-LAN board), it would eventually consume all the remaining RAM available. But maybe that's never going to happen...

The other question I have is: Does any kind web-based configuration mena that we have to define a maximum number of entries for one option (such as currently with log_parameters[20]) or if a more flexible approach (such as with ipwe_parameters[]) would be possible. A fixed number of, say, 20 parameters would probably not be a major issue, but the space both in memory and flash would be blocked even if only very few parameters are being used. 

Here's a first draft of a structure, in this case just for the generation of the web-interface, but could be added with the concrete (pointer to the) value of the corresponding option:

```
typedef struct {
	uint8_t id;		// a unique identifier that can be used for the input tag name
	const char* desc;	// pointer to text to be displayed for option - is text length necessary if we just read until NULL?
	uint8_t input_type;	// input type (text, dropdown etc.)
	uint8_t var_type;	// variable type (string, integer, float, boolean etc.), could maybe be derived from input_type or vice versa
	uint8_t category;	// for grouping configuration options
} configuration;
```

We could then have defines to the configuration rows to make access more human-readable; let's say the IP address is stored in row 0 of the struct:
`#define IP_ADDRESS 0`
then we could accesss it via 
`configuration.value[IP_ADDRESS]`
or something like that...

dukess, 2020-08-02:
```
struct __attribute__((__packed__)) settings { //prevent padding
byte readsettingsfromeeprom; //0 - read from _config, 1 - read from EEPROM
byte mac[6]; //MAC address. If all zeros - init with hardware-defined MAC
byte ipaddr[4];
byte mask[4];
byte dns[4];
byte gateway[4];
uint_16t port; //1-65535
boolean logTelegram;
boolean log_unknown_only;         // should we log only unknown commands when logging bus telegrams?
boolean log_bc_only;              // should we log only broadcast commands (dest = 0x7f) when logging bus telegrams?
...
}
```

fredlcore, 2020-08-02:
Maybe using two structs would then be most effective, with the one from my suggestion getting an additional field with a pointer to the position of your struct where the actual value is stored (if that is even possible)?
Because with your struct, we still would have to hard-code a web-interface, and adding a new configuration option would require adding the respective HTML code as well etc. 
With the my struct above, we could just loop through the struct entries and the respective configuration entries would be displayed.

Could/should we just allocate a memory area of X byte which contains all the configuration values in the way you structured it above and then have a (calculated?) pointer that memory area which will then be written to EEPROM?

hacki11, 2020-08-02:
I'm not sure if my input is wanted, but here it is :)
`byte readsettingsfromeeprom`=> `byte settingsSource { EEPROM, FLASH }`

I don't know about the logs but is it possible to introduce a log level instead of several bools?
`uint_8 telegramLogLevel { NONE, UNKNOWN, BROADCAST, ALL }` A higher level includes all lower levels, means if BROADCAST is selected, UNKNOWN will also be written.

Would stick to one naming style, lowerCamelCase or snake_case, but i think it is all just a draft for conceptual issues and you would have changed it in your final version.

hacki11, 2020-08-02:
> 
> 
> We can write a function that, when writing new parameters, compares the contents of the structure byte by byte and writes only the changed data.

Afaik, the EEPROM.update will do exactly that for you automatically, even writing a whole struct.

fredlcore, 2020-08-02:
Ah, thanks, that's a helpful hint - but I'll have to check if the (non-standard) EEPROM library for the serial EEPROM we use on the Due also supports that feature.

fredlcore, 2020-08-02:
...and it does, yeah!
https://github.com/CombiesGit/I2C_EEPROM#void-updateunit16_t-addressuint8_t-value

dukess, 2020-08-02:
> to introduce a log level instead of several bools?

...or bitwise flags: NONE = 0, UNKNOWN = 1, BROADCAST = 2, ALL = 255

fredlcore, 2020-08-02:
I also think bitwise is better because not all options (in the future) may be hierarchical...

dukess, 2020-08-02:
can we place configuration struct in flash memory or we need it in RAM permanently?

fredlcore, 2020-08-02:
I think reading also strains the EEPROM cells, but not as much as writing does. So I would rather keep the structure in RAM - which is not a problem for the Due. Or we would have to copy certain often used parameters from the struct to RAM variables, but that would make it less structured. And that would be a compromise I would not be willing to make for the obsolete Mega infrastructure...

dukess, 2020-08-03:
Add some fields to struct.
typedef struct {
	uint8_t id;		// a unique identifier that can be used for the input tag name
	uint8_t category;	// for grouping configuration options
	const char* desc;	// pointer to text to be displayed for option - is text length necessary if we just read until NULL?
	uint8_t input_type;	// input type (text, dropdown etc.)
	uint8_t var_type;	// variable type (string, integer, float, boolean etc.), could maybe be derived from input_type or vice versa
        uint16_t var_size; //variable size
        uint16_t var_address; //variable address in EEPROM (but how to allocate EEPROM for arrays?)
} configuration;

dukess, 2020-08-03:
> And that would be a compromise I would not be willing to make for the obsolete Mega infrastructure...
Maybe then you can make one more final release with Mega support?
If the format of BSB_lan_defs.h will not change, then we can continue to add new commands to the old release, but without adding new features.

fredlcore, 2020-08-03:
Yes, there will definitively be a final release before we add the first "breaking" feature for the Due. And even as long as the array is just relatively small, it might still run on the Mega. It's just that it might become unstable if you have too many logging/average/etc. parameters...

dukess, 2020-08-03:
Super!

fredlcore, 2020-08-03:
But am I correct in my understanding that something like `ipwe_parameters[] = {...}`is only possible during compile time? So we will have to set a fixed number of parameters for these variables if we want to read them from EEPROM during runtime?

dukess, 2020-08-03:
Correct. Or we should use some kind of dynamic memory allocation.

loetmeister, 2020-08-03:
One comment on the point, to avoid reading all EEPROM data into a struct that will consume RAM. You can use the struct you discussed above, but define with **EEMEM** macro (the struct used with EEMEM should describe the entire EERPOM layout. So it can be a struct of structs)
https://www.avrfreaks.net/forum/tut-c-using-eeprom-memory-avr-gcc?name=PNphpBB2&file=viewtopic&t=38417
With this you can access the EEPROM, it does not consume any RAM, still you need to read the data into variables you want to work with. So you could just allocate the variables you need and read the data, when done (e.g. if this was a function) it frees up the space again)
Some AVRs allow to access EEPROM like any other varible stored in RAM. I think this can be setup with MAPPED_EEPROM_START, however the Mega2560 does not seem to have this feature.

fredlcore, 2020-08-03:
Interesting, does this also apply to the Due's SAM processor and the external i2C serial EEPROM we're using there?

loetmeister, 2020-08-03:
It maybe is possible. I never tried it, I only used internal EEPROM.
But EEMEM just is referencing the EEPROM section (address) So this is not library specific, but handelt by the compiler/linker. You can change the section, not sure if this will work for external EEPROM... like
`#define EEMEM __attribute__((section(".myExtenalEeprom")))`
https://electronics.stackexchange.com/questions/172719/define-eemem-attribute-section-eeprom

If this does not work, you can load everthing in RAM on the Due, just use EEMEM for the Mega?

fredlcore, 2020-08-03:
If we have to read the EEPROM into variables (as you wrote above) and the feature does not work seamlessly on the Due, I would opt against it because it would mean two different sets of variables (one "large" struct for the Due and various independent variables read on-demand from EEPROM on the Mega) and probably quite some work to get it done on the Mega. For me, I won't be doing extra work for the Mega anymore but will create a fork once the first "Mega-breaking" feature for the Due is implemented. Then whoever likes it can implement this (and any other new feature) without making the general (Due-)code unreadable...

loetmeister, 2020-08-03:
It was just a proposal... :-)
If this would work for internal as well as embedded EEPROM, it should not be extra work... however I cannot confirm this.

Just one comment to:
> EDIT: And this behavious actually makes it unsuitable for our project because people will be flashing newly compiled software regularly and then the location in EEPROM will no longer be the same:
> 
Yes, but only `When more EEMEM variables are used` - this should not be done, use just one struct. In this case the layout inside the struct is fixed. The only risk might be that the whole .eeprom section moves...

```
typedef struct {
  uint8_t reserved;
  uint32_t version;
  uint8_t ip_address[4];
  boolean bool_setting;
  configuration conf[20];
// ... other config structs...
} t_eeprom_space;
```

fredlcore, 2020-08-04:
Of course I appreciate any kind of suggestion - didn't mean to disrespect ;)...
And thanks for the clarification, if there is no risk that variables get mixed up after reflashing, then it doesn't matter where in EEPROM the variable is located.
Still, my main priority would be to have a clean solution where (if possible) all configuration options can be accessed transparently and there is no need to copy EEPROM contents to variables etc. (I think that's got me confused from your earlier posting). If that is possible for both systems (with maybe a few extra lines of code encapsulated in #ifdefs) then I'm all in.

loetmeister, 2020-08-04:
Thanks... agree. I would also go for a clean setup, and not trying to implement someting over-sophisticated that has no use.... :-)

On the overall setup.... how will it look like?
The first part you mentioned, with fixed lenth is somewhat clear, but how would you handle this
> Second section: Divided into two parts for configuration options of (at compile time) unknown length, such as log_parameters array, passkey

It sounds like this would create a lot of overhead to track the EEPROM address where this data is actually stored?

On the basic part
If you decide on a layout for the struct(s) and the struct would simply be held in RAM, it's initially empty. So every time the Due boots or you changed the EEPROM you have to (re)read the data to the struct from EEPROM.
If you read or write/update the entire struct each time, you would just need to define the start address when calling eeprom functions. If you want to handle multiple seperate structs, you would need to keep track of the start and end address (i.e. length). 
If this is only a few sections, this should be ok, else this can get complicated and error-prone in future...

I was thinking about a few different structs, based on the type of configuration... and maybe you just allocate some fixed space on a "multi purpose" area, like "log_configuration" example below. So you can use the same EEPROM space for different functions.

```
typedef struct {
  basic_configuration bConf;  // fixed config only needed once, like default bus type, SD-card, etc.
  network_configuration netConf; // fixed config only needed once, like network settings
  log_configuration logConf[40]; // 40 slots for logging settings, avg, ipwe or other stuff?
} my_eeprom

typedef struct {
  byte ipaddr[4];
  byte mask[4];
  byte dns[4];
  ... etc.
} network_configuration;

typedef struct {
  int parameter;
  int avg:1;        // bit flag to include in average calculation
  int log:1;        // bit flag to include in 24h logging
  int ipwe:1;        // bit flag to display in IPWE
} log_configuration;
```

About "passkey".. I would limit it to certain size. Or why not dropping this and use the power of Due to add HTTPS support? :-D

fredlcore, 2020-08-04:
> It sounds like this would create a lot of overhead to track the EEPROM address where this data is actually stored?

Actally, it's not: You have a fixed order of variables that you read from EEPROM one by one, jumping forward the number of bytes in the said locations. But I think the struct makes more sense now.

> So every time the Due boots or you changed the EEPROM you have to (re)read the data to the struct from EEPROM.

Why would you change the EEPROM? And yes, you have to read the EEPROM on every boot, that's the whole idea behind storing the configuration in EEPROM, isn't it? If the EEPROM has not been written yet, default values will be used.

The "struct of structs" looks nice and clean, but I think sticking to one struct would make it easier and with a (still) limited number of configuration options which in the end (like Passkey) can be fixed in length without much inconvenience, it should not be too confusing (I hope ;) ). But configuration options like Passkey or the Auth-Basic credentials show that these will have to be moved from flash memory to RAM to make them configurable and thus will have an impact on the Mega at least.

As for SSL, I had hoped this could be done with the Due, but alas:
https://stackoverflow.com/questions/15830333/arduino-due-https-support

fredlcore, 2020-08-05:
I was thinking that it could be worthwhile to leave BSB_lan_config.h in place, but putting the default struct and its values there. These would be overloaded with values from the EEPROM. This way, Mega users could still use the _config.jh as they were used to (with a one-time reconfiguration, though). And we cound also leave certain configuration options that no one hardly ever has changed so far (GPIO protected pins, bus pins, query retries) as well as options that still will have to be defined on compile-time (such as the language).

fredlcore, 2020-08-05:
I actually still like the "struct of structs" suggestion quite a bit, especially with the bit flags so that we only would have to store a parameter int once and not several times. My main concern is still the approach how to generate the configuration HTML page from such a struct so that adding new configuration options does not require adding (much) more code. I'm not sure if my approach from above makes sense in connection with the different kind of variables we have to configure. What are your thoughts?

loetmeister, 2020-08-05:
Not sure how the HTML form will have to be processed, when submitted by the browser. I guess you have different elements, like text boxes, drop down lists and check boxes. All need to be read and stored in the correct sruct variables.

This was also the point I was asking about reading the EEPROM again, but this is not needed if you keep the whole struct in memory.
Even this is ok on the due, it's not needed.. In particular log Parameters, that will be processed sequencially. I would be simple to loop through the struct array and just process them element by element.
This could also be the way to read the HTML form?

Auth-Basic credentials could also be read directly from EEPROM, they are only needed once in a session..

Keeping BSB_lan_config.h sounds like a good idea. You anyway need a way to populate the initial values.?

fredlcore, 2020-08-05:
For processing the HTML, I thought of several ways until I remembered that it is possible for a while now to use all URL commands also via serial so that BSB-LAN can be used directly attached to a NAS or a FHEM running device such as an Raspberry Pi. For that reason, the configuration should also be possible via URL commands and I was thinking of a syntax like /Cx=y. This would mean we would have individual SET buttons jus as we have with every writeable parameter and a small JavaScript that creates the corresponding URL (again, like with setting other parameters). With a struct consisting of all parameters (as mentioned in my first struct proposal) and their "design aspects" the HTML could be generated on the fly. The struct would also contain the "x" used for /Cx. The receiveing part could then just have a switch-case-block where each received configuration option would be put into the correct variable/struct section. It would be a bit annoying if you wanted to change several parameters at once, but this is probably not going to happen that often. And it would enable configuring the device completely via serial which I think is a big plus if the BSB-LAN module for FHEM is going to be completed sometime in the future...

dukess, 2020-09-07:
List of current config parameters and their (proposal) sizes
```
typedef enum{
  CF_USEEEPROM, //Size: 1 byte. 1 read config from EEPROM. Other values - read predefined values from BSB_lan_config
  CF_FAMILY, //Size: 1 byte. Device family. 0 will be ignored
  CF_VARIANT, //Size: 1 byte. Device variant. 0 will be ignored
  CF_BSBBUS, //Size: 2 bytes. BSB bus pins. 0 will be ignored (DROPDOWN selector?)
  CF_BUSTYPE, //Size: 1 byte. Bus type at start (DROPDOWN selector)
  CF_PROTECTEDGPIO, //Size: 1 * 64 bytes. Array of protected pins.
  CF_MAC, //Size: 6 bytes. MAC address
  CF_DHCP, //Size: 1 byte. DHCP: 0 - disabled, 1 - enabled
  CF_IPADDRESS, //Size: 4 bytes. IP v4 address
  CF_MASK, //Size: 4 bytes. Network mask
  CF_GATEWAY, //Size: 4 bytes. Gateway
  CF_DNS, //Size: 4 bytes. DNS server address
  CF_WWWPORT, //Size: 2 bytes. Port number, 1-65535
  CF_LOGTELEGRAM, //Size: 1 byte. Bitwise: LOGTELEGRAM_NONE = 0, LOGTELEGRAM_UNKNOWN = 1, LOGTELEGRAM_BROADCAST = 2, LOGTELEGRAM_ALL = 255
  CF_LOGAVERAGES, //Size: 1 byte. Log average values. 0 - disabled, 1 - enabled. Program list will be set in CF_AVERAGESLIST
  CF_LOGCURRVALUES, //Size: 1 byte. Log current values. 0 - disabled, 1 - enabled. Program list will be set in CF_CURRVALUESLIST
  CF_LOGCURRINTERVAL, //Size 4 bytes. Unsigned. logging current values interval in seconds
  CF_WEBSERVER, //Size: 1 byte. External (read files from SD card) webserver: 0 - disabled, 1 - enabled
  CF_AVERAGESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_CURRVALUESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_PASSKEY, //Size 64 bytes. String with PASSKEY
  CF_ONEWIREBUS, //Size: 1 byte. One wire bus pin. 0 will be ignored
  CF_DHTBUS, //Size: 2 bytes. DHT temperature/humidity bus pins. 0 will be ignored
  CF_IPWE, //Size: 1 byte. IPWE extension: 0 - disabled, 1 - enabled
  CF_IPWEVALUESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_QUERY_RETRIES, //Size 1 byte. number of retries for the query command
  CF_MAX_IPADDRESS, //Size: 4 bytes. IP v4 address of CUNO/CUNX/modified MAX!Cube
  CF_MAX_DEVICES, //Size 10 * 40 bytes.
  CF_RESET, //Size: 1 byte. Reset function: 0 - disabled, 1 - enabled
  CF_READONLY, //Size: 1 byte. All parameters will be FL_RONLY
  CF_DEBUG, //Size: 1 byte. Debug: 0 - disabled, 1 - debug to serial interface, 2 - debug to telnet
  CF_MQTT, //Size: 1 byte. MQTT: 0 - disabled, 1 - enabled, plain text, 2 - enabled, JSON
  CF_MQTT_IPADDRESS, //Size: 4 bytes. MQTT broker IP v4 address
  CF_MQTT_USERNAME, //Size: 32 bytes.
  CF_MQTT_PASSWORD, //Size: 32 bytes.
  CF_MQTT_TOPIC //Size: 32 bytes.
} cf_params;
```

fredlcore, 2020-09-07:
Wow, thanks, that's great work!
As for DHT, I think 5 is already really maxing it out because you need at least one additional wire for each. 
For 1Wire 10 would be comfortable, I guess, what do you think, @1coderookie ?
For MAX, I think 20 would be be comfortable because you can have more than one heating unit in each room, plus a wall thermostat.

As for the other parameters, I think we could/should keep some of them in _config.h because we won't be able to completely do away with _config.h (at least for the `#define LANG` we will still need it). So some of the more debug-like features should be moved there (or removed completely because hardly anyone has ever configured them):

CF_FAMILY, //Size: 1 byte. Device family. 0 will be ignored
CF_VARIANT, //Size: 1 byte. Device variant. 0 will be ignored
CF_BSBBUS, //Size: 2 bytes. BSB bus pins. 0 will be ignored (DROPDOWN selector?)
CF_PROTECTEDGPIO, //Size: 1 * 64 bytes. Array of protected pins.
CF_QUERY_RETRIES, //Size 1 byte. number of retries for the query command
CF_RESET, //Size: 1 byte. Reset function: 0 - disabled, 1 - enabled

1coderookie, 2020-09-07:
> For 1Wire 10 would be comfortable, I guess, what do you think, @1coderookie ?

Yes I agree - I guess with more than ten devices a dedicated busmaster would be a good idea anyway, so I think 10 is a good amount :) 
Just for my understanding: in case someone would like to connect more sensors - would that still be possible somehow (e.g. by not using the webinterface for configuration) or are we talking about a real limit for the future..?

1coderookie, 2020-09-07:
> Most of the entries probably do not need default values, but for the network part, it would be best to set the device to DHCP mode instead of setting some kind of fixed IP address (although this would probably also be acceptable).

In this point I would suggest to stay with the fixed IP like we already did - using DHCP as a standard could make it hard to find the right IP of the adapter after flashing, especially because it seems that not every kind of router does seem to show the adapter and it's IP in the router's webinterface (e.g. FritzBox). Sure, one could use serial monitor to check the IP, but imho a fixed IP would make it easier in the first row for any newbie.. Or am I wrong..?

fredlcore, 2020-09-07:
As for sensors: Yes, we are talking about real limits now. It may be possible to override the web configuration through _config.h (we're not certain if this will work), but then only in total, not for an individual parameter. We should stick to "comfortable" values, not extreme use scenarios because memory in the EEPROM is fairly limited.
 
As for DHCP: I'm still trying to figure out if/how it could be possible to set a hostname that the device advertises. Then DHCP would be a good option and network newbies wouldn't have to get into their router first to figure out what IP address range they are using, which IPs are outside the DHCP range etc.

1coderookie, 2020-09-07:
Sensors: Ah ok, got it. I agree, more sensors than 10 would be more an extreme scenario, so let's got with 10. 

DHCP: An own hostname for the device would be nice! :)

fredlcore, 2020-09-07:
Yes, unfortunately this does not seem to be supported by the Ethernet library yet :(...
#https://github.com/arduino/Arduino/pull/5701

dukess, 2020-09-07:
Big piece of code.

First we must call initConfigTable() for build config options address table in EEPROM.
After that registerConfigVariable() for linking config options with variables in RAM:
byte ip_addr[4];
...
char topic[32];
registerConfigVariable(CF_IPADDRESS, ip_addr);
...
registerConfigVariable(CF_MQTT_TOPIC , topic);
And now we can read and write EEPROM:
readValueFromEEPROM(CF_MAC);
writeValueToEEPROM(CF_PASSKEY);

At least that's the plan.

```
typedef enum{
  CF_USEEEPROM, //Size: 1 byte. 1 read config from EEPROM. Other values - read predefined values from BSB_lan_config
  CF_BUSTYPE, //Size: 1 byte. Bus type at start (DROPDOWN selector)
  CF_MAC, //Size: 6 bytes. MAC address
  CF_DHCP, //Size: 1 byte. DHCP: 0 - disabled, 1 - enabled
  CF_IPADDRESS, //Size: 4 bytes. IP v4 address
  CF_MASK, //Size: 4 bytes. Network mask
  CF_GATEWAY, //Size: 4 bytes. Gateway
  CF_DNS, //Size: 4 bytes. DNS server address
  CF_WWWPORT, //Size: 2 bytes. Port number, 1-65535
  CF_TRUSTEDIPADDRESS,//Size: 4 bytes. First trusted IP v4 address. 0 - disabled
  CF_TRUSTEDIPADDRESS2,//Size: 4 bytes. Second trusted IP v4 address. 0 - disabled
  CF_LOGTELEGRAM, //Size: 1 byte. Bitwise: LOGTELEGRAM_NONE = 0, LOGTELEGRAM_UNKNOWN = 1, LOGTELEGRAM_BROADCAST = 2, LOGTELEGRAM_ALL = 255
  CF_LOGAVERAGES, //Size: 1 byte. Log average values. 0 - disabled, 1 - enabled. Program list will be set in CF_AVERAGESLIST
  CF_LOGCURRVALUES, //Size: 1 byte. Log current values. 0 - disabled, 1 - enabled. Program list will be set in CF_CURRVALUESLIST
  CF_LOGCURRINTERVAL, //Size 4 bytes. Unsigned. logging current values interval in seconds
  CF_WEBSERVER, //Size: 1 byte. External (read files from SD card) webserver: 0 - disabled, 1 - enabled
  CF_AVERAGESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_CURRVALUESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_PASSKEY, //Size 64 bytes. String with PASSKEY
  CF_BASICAUTH, //Size 64 bytes. String with USER_PASS_B64
  CF_ONEWIREBUS, //Size: 1 byte. One wire bus pin. 0 will be ignored
  CF_ONEWIREBUS_DEVICES, //Size: 8 * 10 bytes. Array of One Wire devices, 8 bytes per device
  CF_DHTBUS, //Size: 2 bytes. DHT temperature/humidity bus pins. 0 will be ignored
  CF_IPWE, //Size: 1 byte. IPWE extension: 0 - disabled, 1 - enabled
  CF_IPWEVALUESLIST, //Size 2 * 40 bytes. Array of prognrs 1-65535. prognr 0 will be ignored
  CF_MAX_IPADDRESS, //Size: 4 bytes. IP v4 address of CUNO/CUNX/modified MAX!Cube. 0.0.0.0 - disabled
  CF_MAX_DEVICES, //Size 10 * 40 bytes.
  CF_READONLY, //Size: 1 byte. All parameters will be FL_RONLY
  CF_DEBUG, //Size: 1 byte. Debug: 0 - disabled, 1 - debug to serial interface, 2 - debug to telnet
  CF_MQTT, //Size: 1 byte. MQTT: 0 - disabled, 1 - enabled, plain text, 2 - enabled, JSON
  CF_MQTT_IPADDRESS, //Size: 4 bytes. MQTT broker IP v4 address
  CF_MQTT_USERNAME, //Size: 32 bytes.
  CF_MQTT_PASSWORD, //Size: 32 bytes.
  CF_MQTT_TOPIC //Size: 32 bytes.
} cf_params;

#define LOGTELEGRAM_NONE 0
#define LOGTELEGRAM_UNKNOWN 1
#define LOGTELEGRAM_BROADCAST 2
#define LOGTELEGRAM_ALL 255

//according to input_type in configuration_struct
typedef enum {
  CPI_TEXT, // general text field
  CPI_SWITCH,
  CPI_DROPDOWN
} cpi_params;

//according to var_type in configuration_struct
typedef enum {
  CDT_BYTE,
  CDT_UINT16, //CPI_TEXT field with format for unsigned integer
[... 116 lines cut ...]
}

void readValueFromEEPROM(uint8_t id){
  if(!options[id].option_address) return;
  int len = 0;
  for(int j = 0; j < sizeof(cf_params)/sizeof(CF_USEEEPROM); j++){
    if(id == config[j].id)
      len = config[j].size;
  }
  for(int i = 0; i < len; i++)
    options[id].option_address[i] = EEPROM.read(i + options[id].eeprom_address);
}

void writeValueToEEPROM(uint8_t id){
  if(!options[id].option_address) return;
  int len = 0;
  for(int j = 0; j < sizeof(cf_params)/sizeof(CF_USEEEPROM); j++){
    if(id == config[j].id)
      len = config[j].size;
  }
  for(int i = 0; i < len; i++)
    EEPROM.update(i + options[id].eeprom_address, options[id].option_address[i]);
}

```

loetmeister, 2020-09-07:
That's a lot of stuff... Not sure how this will all work out :-)

`addressesOfConfigOptions options[size... `

For this one... The basic idea is to only keep an stuct in RAM that has the start addresses from the actual values /structure? 
Where would `configuration_struct` be used?

`const char* desc;	// pointer to text`
If these are char arrays, like the static progmem text, it will have an end of line terminator. No need to know the length... Just the destination array need to be large enough... ;-)

fredlcore, 2020-09-07:
Wow, that's very well thought through! And it will be relatively easy to add new configuration options (not sure though if that's also the case if new data-types come up sometime in the future?). Thanks a lot for wrapping your brain around this, that really looks like a promising point of departure!

dukess, 2020-09-08:
@loetmeister configuration_struct will be used later to create the configuration page interface like prog table now.
'options' array is built dynamically at startup and contains EEPROM addresses and pointers to variables in RAM.
Using an array will allow to speed up the search for value of a variable. We can do without an array, but then we will have to add EEPROM address field to configuration_struct table and carefully monitor absence of intersections.

>...arrays, like the static progmem text...

Yes, sure!

@fredlcore 
> ...though if that's also the case if new data-types come up sometime in the future...

Hope too :)
By the way, we will have to make a large number of changes in the code to make the configuration work, because now most of the settings are made via #define
I'm afraid we won't be able to save compatibility with older versions.

1coderookie, 2020-09-08:
Guys - for me as a non-programmer it's always astonishing how fast you realize certain ideas! Thumbs up! :) 

> I'm afraid we won't be able to save compatibility with older versions.
Imho that's totally fine, the Mega-v2-setup is 'outdated' anyway.. ;)

dukess, 2020-09-08:
@1coderookie I meant with version 1.1 and earlier. Users will must to change the bsb_lan-config.h. By the way, Mega is still a working solution for me at the moment. It still has ~20KB of free space.

fredlcore, 2020-09-08:
I think for such a massive change, it is perfectly fine to expect people to change the config. We can use this opportunity to also change other things which need action by the end-user.
However, I was thinking that we could still use the defines for those Mega users who want to save space by excluding certain segments of code and just use the EEPROM-based configuration nevertheless?

dukess, 2020-09-08:
need to think how to do it

fredlcore, 2020-09-08:
I was just thinking something like this:
We use a specific variable, let's say IP address:
```
#ifdef IPAddr
IPAddress ip(IPAddr);
#endif
```
So now we could write it like this:
```
#ifdef IPAddr
IPAddress ip(IPAddr);
#else
IPAddress ip(options[id].value);
#endif
(Syntax for options variable is not correct of course)

But if this is just me thinking too shallow, then no need to keep these definitions and we make a hard cut with a new major release number. As Ulf said, Mega2560 compatibility is not a must anymore.

loetmeister, 2020-09-08:
> 'options' array is built dynamically at startup and contains EEPROM addresses and pointers to variables in RAM.

Ok.. hope I understand. :-) In this case, this is like an abstraction layer, to not have to keep all EEPROM variables/struct in RAM, but only read an write what is actually needed? Isn't it like the EEMEM "behaviour"?

For the EEPROM layout. I would suggest to add one serial number/magic number in a fixed place which you can update in case the EEPROM layout has to change. In case you change the layout at a later stage, which will not be compatible, you can check by this number and wipe and re-initialize a new layout/default values, if needed..

dukess, 2020-09-09:
> Isn't it like the EEMEM "behaviour"?

Yep. It is. But EEMEM not supported on Duo. And I'm too lazy to write two different pieces of code :)

>...add one serial number/magic number in a fixed place...

Good idea!

dukess, 2020-09-10:
Idea: instead magic number we can use CRC32 for options structure checking.
https://github.com/bakercp/CRC32

fredlcore, 2020-09-10:
Nice idea! But a version number would only cost us one or two bytes and it would be more "human-readable", also when it comes to several different update paths (let's say from version 2 to 3 is different compared to update from version 1 straight to version 3).

hacki11, 2020-09-10:
Yeah, can confirm, keep it simple. There should be something you have to do if you need to do a migration to a higher version as a dev. Should also be a not usual to increase, additions should not be a problem.

dukess, 2020-09-11:
Main problem, as I see it, may be that if we (or user) later want to change size of an option or delete it. In this case, the options addresses will change and data will turn into garbage. To prevent this scenario, we must somehow check the structure. Adding new options should not be a problem if they are will be added to the end of the list.

hacki11, 2020-09-11:
I see this like an api or a database schema. Any change requires a migration method. Additions between options, removals, change in size. Same for content of options. Change of values, removals of values.
For me as a dev it would be more clear to go through readable versions (1-2-3-4...-9) as when I have to go through several random looking CRC hashes. 
That's for a version. 

I think a crc hash which will be calculated on boot and compared with a predefined value in ROM is important, too to avoid reading garbage like you said.

I see them solving two different things, both needed.

fredlcore, 2020-09-11:
> Main problem, as I see it, may be that if we (or user) later want to change size of an option or delete it. In this case, the options addresses will change and data will turn into garbage. To [...]
 
I'm not sure I understand: Why should the user change the size of an option or delete it? I mean if we do it, we know what we're doing (hopefully ;) ), but then we know that we need to take care of it?

[20 more replies]

closed 2020-11-23 by fredlcore