fredlcore/BSB-LAN #107: how to set parameters with mqtt

duncan-valleix, 2020-06-10:
In French
Hello, for English I am sorry but I use a translator, I put into service your kit (arduino version) it works very well, I have everything config in relation to my use but I have not found how set values on it or it parameter, could you tell me if this is possible and how to do it? and of course if this is not possible at the moment would it be possible for you to foresee it because I use Home-Assistant but the integration is not complete, I do not have the possibility of putting modify the parameter of the circuit refresh and unfortunately I do not have the competence to do it myself.
but thank you and you've done a great job.

En Français
Bonjour, pour l'anglais je suis désolé mais j'utilise un traducteur, j'ai mis en service votre kit (version arduino) il fonctionne très bien, j'ai tout config par rapport a mon utilisation mais je n'ai pas trouvé comment set des valeurs sur t'elle ou t'elle paramètre, pouvais vous me dire si cela est possible et comment le faire? et bien sur si cela n'est pas possible a l'heure actuelle serais t'il possible de le prévoir  car j'utilise Home-Assistant mais l’intégration n'est pas complète, je n'ai pas la possibilité de mettre de modifier le paramètre du circuit rafraîchissent et malheureusement je n'ai pas les compétence pour le faire moi même.
mais je vous remercie et vous avez fait une excellent travail.

[image]
[image]
[image]

fredlcore, 2020-06-10:
You can set parameters for example via 
`http://<ip>/S700=20`
which sets parameter 700 to 20 degrees.
Of course you have to set 
`#define DEFAULT_FLAG  0`
in BSB_lan_config.h to make parameters writeable in general. Otherwise BSB-LAN is read-only.
If this is not working, please post your BSB_lan_config.h.

As for integration into Home Assistant, I cannot give you assistance because I don't have this system. A FHEM forum member has worked on this, find more about this here:
https://1coderookie.github.io/BSB-LPB-LAN_EN/chap11.html#1111-home-assistant

fredlcore, 2020-06-10:
By the way: Setting parameters via MQTT is not possible yet, and I don't have plans to implement this in the future, simply because I don't work with MQTT. Maybe some other user will write an implementation, but I don't know if this is going to happen. You can set parameters via JSON, however:
https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#824-retrieving-and-controlling-via-json

duncan-valleix, 2020-06-10:
> You can set parameters for example via
> `http://<ip>/S700=20`
> which sets parameter 700 to 20 degrees.
> Of course you have to set
> `#define DEFAULT_FLAG 0`
> in BSB_lan_config.h to make parameters [...]

In English
 I'm sorry I expressed myself badly, for set the values ​​its working, my question really focused on how to set a value by mqtt

En francais
Je suis navrée je me suis mal exprimé, pour set les valeurs sa fonctionne, ma question porté vraiment sur comment set une valeur par mqtt 

> By the way: Setting parameters via MQTT is not possible yet, and I don't have plans to implement this in the future, simply because I don't work with MQTT. Maybe some other user will write an [...]

En francais
Je vais essayé de modifier le programme pour qu'il surveille MQTT et voir pour intégrer la fonction set

fredlcore, 2020-06-10:
Ok, will close this here for now then.

closed 2020-06-10 by fredlcore

1coderookie, 2021-02-07:
@duncan-valleix: It is now possible to query and set via MQTT! Please see [chapter 8.2.12](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#8212-mqtt) for further informations.