fredlcore/BSB-LAN #112: Expanding /D and /L URL commands

dukess, 2020-06-25:
I prepare new patch with these functions:
/DJ0 - remove journal.txt only
/DD0 - remove datalog.txt only
/DJ - dump journal.txt
/DD - dump datalog.txt
/LE - enable telegrams logging to journal.txt
/LD - disable telegrams logging
New option in BSB_lan_config.h: logTelegram. False - disabled on start, true - enabled on start.
Logging enabling through progNr 30000, /D0  and /D still used for compatibility

What you think?

fredlcore, 2020-06-25:
Sounds good to me! But what exactly would logTelegram enable?

dukess, 2020-06-25:
logTelegram is an explicit parameter that enables raw telegram logging. Its introduction allows us to divide management of datalog.txt and journal.txt. logTelegram will be responsible for working with journal.txt, and log_parameters for working with datalog.txt

fredlcore, 2020-06-25:
Ok, got it. But why removing the 30000 logging? As I said, I wouldn't want to break compatibility if not really necessary. Your suggested changes are cleaner, that's for sure, but not too many people will be noticing ;)...
As for dividing telegram and data log, that could be done, but since logging parameters is so much distributed all over the place, this would probably be a long(er) and error prone task, I guess. But if you want to do it, sure!

dukess, 2020-06-26:
>Ok, got it. But why removing the 30000 logging? As I said, I wouldn't want to break compatibility if not really necessary. Your suggested changes are cleaner, that's for sure, but not too many people will be noticing ;)...

I don't plan to remove 30000 support yet. As you said earlier, it may be possible to remove the legacy in some of the big releases. But I'll probably delete prognr 30000 from the configuration file so that new installations will use the new option.

>As for dividing telegram and data log, that could be done, but since logging parameters is so much distributed all over the place, this would probably be a long(er) and error prone task, I guess. But if you want to do it, sure

Don't panic! :- ) There is only one function for printing telegrams to a file. its behavior is managed by three variables: log_bc_only, log_unknown_only and logTelegram (or log_parameters[0] == 30000). After the new variable logTelegram, all control is now concentrated in a single code fragment.

fredlcore, 2020-06-27:
Ok, great, then let's go :)!

closed 2020-06-29 by dukess