fredlcore/BSB-LAN #118: Backwards Compatibility for ioBroker BSB Adapter

hacki11, 2020-07-14:
Hi there,

really happy to see that great work and effort that is put into the last months!

I am the developer of the ioBroker BSB LAN Adapter to integrate your adapter into iobroker. https://github.com/hacki11/ioBroker.bsblan

Are you looking to maintain backwards compatibility or are there some breaking changes we will run into?
If not i will be independent of your pace and can provide a new version if i find some time.
But if we have some breaking changes, we should synchronize our release plans and document some apis or make a api definition.

Best, hacki11

fredlcore, 2020-07-14:
Thanks! And also thanks for maintaining the ioBroker integration, that really helped to make the project more visible outside its original FHEM ecosystem.
Maintaining backward compatibility is a very important aspect for me, even down to the level of the behaviour of functions in the main code that people may use in the user-defined functions. I also try to maintain output format as much as possible to prevent for example regular expressions from breaking.
Having said that, there are several larger changes in the pipeline (although no timeline has been set), but these will most likely affect the way the configuration file is used (or rather becoming obsolete) as we strive to make BSB-LAN completely configurble via a webinterface.

Maybe it would help if you could define what kind of changes would break your ioBroker integration? So far, the JSON interface will most likely be unaffected in the future (except for maybe adding further details, but not changing existing ones). 

Cheers,

F.

hacki11, 2020-07-14:
Currently only JSON endpoints are used. Only removals or renamings will be an issue, additions should be fine.
I will have a look and can provide the exact endpoints i am using

fredlcore, 2020-07-31:
I close this for now, but feel free to re-open when you have more information for us. Thanks again for your support!

closed 2020-07-31 by fredlcore