fredlcore/BSB-LAN #145: bug after recent changes in BSB_lan_config.h ?

Castellanus, 2020-08-16:
BSB_lan_defs.h:8877:62: error: 'STR8317' was not declared in this scope
 {0x113D0F64,  CAT_DIAG_ERZEUGER,    VT_TEMP,          8317,  STR8317,  0,                    NULL,         FL_RONLY,     DEV_ALL}, // Baxi Luna Platinum+ [°C ] - Diagnose Erzeuger -Schaltpunkt TWW-Betrieb
                                                              ^
BSB_lan_defs.h:8882:62: error: 'STR8321_2' was not declared in this scope
 {0x053D1071,  CAT_DIAG_ERZEUGER,    VT_TEMP,          8321,  STR8321_2,0,                    NULL,         FL_RONLY,     DEV_123_ALL}, // [°C ] - Diagnose Erzeuger -
                                                              ^
exit status 1
'STR8317' was not declared in this scope

fredlcore, 2020-08-16:
We're working on it. Take the last 'official' release from the "Releases" tab in the meantime.

fredlcore, 2020-08-16:
Fixed.

closed 2020-08-16 by fredlcore