fredlcore/BSB-LAN #150: Feature Request: Get 24h average Parameter from JSON

Bratmaxe1981, 2020-09-01:
Hi, 
it is not possible to get the 24h average values ​​via the JSON interface at the moment. 
If this would work, the IO broker adapter could also be expanded accordingly.

thanks

fredlcore, 2020-09-01:
@dukess : Do you see a chance to add the "/A" function to the JSON output without much work or would this require more substantial changes?

dukess, 2020-09-01:
Like this?
```
{
  "8700": {
    "name": "STR8700_TEXT",
    "value": "21.3",
    "unit": "UNIT_DEG_TEXT"
  },
  "8326": {
    "name": "STR8326_TEXT",
    "value": "0",
    "unit": "UNIT_PERC_TEXT"
  }
}
```

fredlcore, 2020-09-01:
Yes, like this, but probably somehow with a marker that these are the 24h average values? I don't know it "A8700" instead of "8700" makes sense or if this is no good JSON or if we would have to create a new URL command for that?

dukess, 2020-09-01:
I would prefer to create a new command, for example /JA

fredlcore, 2020-09-01:
Sounds good to me - would it make sense to add other functions like temperature/humidity sensors as well or would this be a different approach (which we could/should leave until somebody really needs it)?

hacki11, 2020-09-01:
Would be great to have the following information for such a parameter:
- name
- datatype
- value
- unit
- desc 

Currently for getting these information for values i have to use a combination of JQ and JC.

dukess, 2020-09-02:
> Sounds good to me - would it make sense to add other functions like temperature/humidity sensors as well or would this be a different approach (which we could/should leave until somebody really needs it)?

I think if we will access these sensors as virtual programs, we can do everything in one function.

> Would be great to have the following information for such a parameter:

it is possible the fields 'datatype' and 'desc' will be useless, since the average is a numeric value.

dukess, 2020-09-03:
https://github.com/dukess/bsb_lan/commit/beecee6da3e222dd71238cde1c87b85f1cc82016 - patch with /JA command. I haven't tested it yet, since I can get access to the device on weekends only.

dukess, 2020-09-04:
@hacki11 @fredlcore @Bratmaxe1981 https://github.com/fredlcore/bsb_lan/pull/152

fredlcore, 2020-09-04:
Great, thanks! 
@Bratmaxe1981 : I've merged the new command, can you try it out?

hacki11, 2020-09-04:
you guys are pretty active right now, i like it!
Can you post an example output for me to code against?

`Request: GET /JA=8700,8701`
```
{
  "name": ???, 
  "value": ???,
  "unit": ??,
 anything else?
}
```

dukess, 2020-09-04:
Request: GET /JA

Answer:
```
{
  "8700": {
    "name": "STR8700_TEXT",
    "value": "18.1",
    "unit": "UNIT_DEG_TEXT"
  },
  "8326": {
    "name": "STR8326_TEXT",
    "value": "2",
    "unit": "UNIT_PERC_TEXT"
  }
}
```

Bratmaxe1981, 2020-09-04:
That looks great, I'll test it soon. Thanks for your work!

Bratmaxe1981, 2020-09-05:
Is working great, using it with the new IO-Broker Adapter V0.2.

Thanks for your great work!

fredlcore, 2020-09-05:
Thanks for testing, will close this now.

closed 2020-09-05 by fredlcore