fredlcore/BSB-LAN #157: Bad JSON in MQTT message

fl42, 2020-10-08:
Hi,

I tried to publish some values using MQTT with JSON.

Unfortunately the published JSON looks invalid:
`{"BSB-LAN":{"status":{"8700":"17.6","8830":"51.9","8310":"57.8","8744":"---","8950":"57.8","8331":"14568",}}`
(note the missing '}')

My broker is mosquitto version 1.6.12
I'm using BSB Lan version 1.1.39-20201005105852 on Arduino Due.

My config (logging part):
```h
/* activate logging on SD-card. Requires a FAT32-formatted Micro-SD card inserted into the Ethernet-Shield's card slot */
// #define LOGGER
// Logging von "rohen" Bus-Datentelegrammen
// Telegrams logging. False - disabled on start, true - enabled on start.
#define LOGTELEGRAM false
// You can enable logging with URL /LE command or disable it with /LD command when run time.
// Data will be write to "journal.txt"
boolean log_unknown_only = 1;         // should we log only unknown commands when logging bus telegrams?
boolean log_bc_only = 0;              // should we log only broadcast commands (dest = 0x7f) when logging bus telegrams?

// Logging data from parameters
// Interval and list of parameters can be redefined through /L command when run time
// Data will be write to "datalog.txt"
int log_parameters[20] = {
  8700,
  8830,
  8310,
  8744,
  8950,
  8331
};
unsigned long log_interval = 30;    // logging interval in seconds

// Activate sending log_parameters to MQTT broker every log_interval seconds
#define MQTTBrokerIP 192,168,2,1	// Please use commas instead of dots!!!
//#define MQTTUsername "User" // Set username for MQTT broker here or comment out if no username/password is used.
//#define MQTTPassword "Pass" // Set password for MQTT broker here or comment out if no password is used.
#define MQTTTopicPrefix "bsb_lan" 	// Optional: Choose the "topic" for MQTT messages here
#define MQTT_JSON 					// Optional: Use this if you want a json package of your logging information printed to the mqtt topic
//#define MQTTDeviceID "MyHeater"	// Optional: Define a device name to use as header in json payload. If not defined, BSB-LAN will be used.
// Payload will be of the structure: {"MQTTDeviceID": {"status":{"log_param1":"value1","log_param2":"value2"}, ...}}
```

Have any of you ever had the problem?
How can I help identify the root cause?

Many thanks

fredlcore, 2020-10-08:
@dukess : Do you have any idea? Could this have to do with the things you fixed before?

dukess, 2020-10-09:
The problem is more deeper than shown here: program 20000-20200 never sent as JSON since adding the MQTT JSON function. I can offer fast fix for ordinary programs because in my pull request any external sensor can be accessed through programs.

dukess, 2020-10-09:
https://github.com/fredlcore/bsb_lan/pull/159

fredlcore, 2020-10-09:
Thanks! I would have said it is ok until we include all your major changes, but since you were so fast, I have merged your pull request right now, thanks a lot for this speedy fix!

dukess, 2020-10-09:
:-)

fl42, 2020-10-09:
Thank you for the quick fix, it works!
Should I leave this issue open until it is fixed for program > 20000?

fredlcore, 2020-10-18:
@dukess : I just got an e-mail from a user that the current master branch does not compile anymore when MQTT is enabled:
```
/Users/linge/Documents/bsb_lan/BSB_lan.ino: In function 'void loop()':
BSB_lan:7013:49: error: 'MQTTPayload' was not declared in this scope
          if(is_first){is_first = false;} else {MQTTPayload.concat(F(","));}
```

Now I'm confused because @fl42 said that it now works - can one of you enlighten me (and if possible fix the remaining bug)?

fl42, 2020-10-18:
Hi @fredlcore,

I just tried to compile master branch using Arduino IDE (with MQTT enabled) and it works.
Were you able to reproduce his problem on your side?

lingej, 2020-10-18:
This issue was reported by me ;)

Version: 1.1.26-20200908101658 is currently running with MQTT enabled

i have done a git pull and tried to recompile the code.

Now i get

```
/Users/linge/Documents/bsb_lan/BSB_lan.ino: In function 'void loop()':
BSB_lan:7013:49: error: 'MQTTPayload' was not declared in this scope
          if(is_first){is_first = false;} else {MQTTPayload.concat(F(","));}
```

Do i have to add some new libs?

lingej, 2020-10-18:
The error accours if `#define MQTT_JSON` is NOT set in config_local.h

fl42, 2020-10-18:
@lingej Does your configuration file BSB_lan_config.h have the same format as BSB_lan_config.h.default ? (no missing #define?)

lingej, 2020-10-18:
@fl42 i will check ....

fl42, 2020-10-18:
Ok, I'm able to reproduce
Actually it does not work if you want MQTT without JSON payload
I may propose a fix

PR #161 (2020-10-18): MQTTPayload exists only if MQTT_JSON is defined
Fix #157 
@dukess Any opinion on this fix?
closed 2020-10-18 by fredlcore