fredlcore/BSB-LAN #164: GitHub Actions / CI creating problems

fredlcore, 2020-10-22:
Guys, (@fl42, @dukess and others), I have not used GitHub Actions / CI before, and I'm fine with anything new as long as it does not disturb my workflow. But in addition to the breaking errors we've had recently, I now get this result when pushing minor changes in BSB_lan_defs.h (code compiles fine locally):
https://github.com/fredlcore/bsb_lan/actions/runs/322877524

I don't know what these errors mean and where I should start fixing them, but other than that it might be a cool feature, so far I have not seen any benefits in having these GitHub Actions / CI added to the project, especially when we are actually working on several major changes which in itself will probably require our attention in the near future to get them all running properly.

Unless I have overseen some little bug or switch I now have to use, I would suggest we revert to the pre-CI area. Or is this not feasible?

Cheers,

F.

fredlcore, 2020-10-22:
Ok, now I'm confused: When I clicked "re-run jobs" they now all passed fine, whereas before the first three had failed before.
This is what came in the e-mail:
```
Run failed for master (c275001)

Repository: fredlcore/bsb_lan
Workflow: PlatformIO CI
Duration: 13.0 seconds
Finished: 2020-10-22 20:50:14 UTC

View results

Jobs:
build (3.6, BSB_lan_config_default.h) failed (1 annotation)
build (3.6, BSB_lan_config_mqtt.h) failed (1 annotation)
build (3.6, BSB_lan_config_mqtt_json.h) failed (1 annotation)
build (3.7, BSB_lan_config_default.h) cancelled (2 annotations)
build (3.7, BSB_lan_config_mqtt.h) cancelled (2 annotations)
build (3.7, BSB_lan_config_mqtt_json.h) cancelled (2 annotations)
build (3.8, BSB_lan_config_default.h) cancelled (2 annotations)
build (3.8, BSB_lan_config_mqtt.h) cancelled (2 annotations)
build (3.8, BSB_lan_config_mqtt_json.h) cancelled (2 annotations)
```

fl42, 2020-10-22:
Hi @fredlcore,

Continuous Integration should be a means to detect bugs more quickly and not a constraint.
In any case CI can fail but this should not stop you from doing anything (push new code, publish a release, etc). However if the CI fails, other people may have the same issue so it's worth checking why it failed.

As far as I know I can't see the logs of failed builds as new builds were triggered.
CI should be fully deterministic (i.e. a build should not pass or fail without code change).
If it's not the case, we should investigate why.
At the moment the CI only ensures the code can be built in a clean (i.e. reproducible) environment.

How does CI prevent you from completing your workflow?

fredlcore, 2020-10-22:
Thanks for the explanation - my workflow here is that I have a script which upon a keypress in VSCode does the following:
- compile for Due
- compile for Mega
- update version number in bsb_version.h
- generate language files (UNDEFs etc.)
- do a git pull from master branch
- push newly staged files to master

This all went fine with my last minor update in BSB_lan_defs.h, but a few minutes later I got the abovementioned e-mail. The first three lines show the tests that failed (I think one said that some file was missing?). I then ran my script again (the only resulting code change would have been the updated version number in BSB_version.h), again with no errors whatsoever locally. Then there was no further e-mail, but when I revisited the link above, the first three were still marked as failed. I then clicked re-run jobs, they all passed. Strange.

So for now, it is not an obstacle any more, but when it's raising errors where there are none (or none where I could identify the location or root cause), then it's somewhat annoying. 
But maybe it was just a glitch, in that case I have no objections anymore ;)...

closed 2020-10-22 by fredlcore

fl42, 2020-10-23:
Feel free to ping me on this thread on CI failure, so I may have a look.

fredlcore, 2020-10-23:
Thanks, will do!