fredlcore/BSB-LAN #167: Bug in displaying PPS time programs

fredlcore, 2020-11-09:
@dukess , I hope you can help me here:
Since a few versions back, the displaying of the time programs is no longer working properly for PPS bus.
Just to recall: the encoding for hour/minutes is different from BSB and is counted in 10 minutes intervals. The time "23:30" therefore is encoded as 141 (0x8D), to decode it one has to divide it by six to get the hours and multiply the remainder with 10 to get the minutes, i.e. 141/6 = 23(,5) and 141 mod 6 = 3 ( * 10 = 30 minutes).
The thing now is that it still works in the webinterface if you use full hours (i.e. 23:00). Also, the value to be sent to the heater is correctly calculated in all cases, so the above mentioned 141 is sent correctly to the heater. 

But in the web interface, it seems that this is what goes wrong: The 141 is divided by 6 (= 23) and to that the remainder times 10 (= 30) is added to the hour, so it is displayed as "53:00".

I have tried to figure out where to change this, but since I haven't had time to familiarize myself with the new approach with decodedTelegram.value etc., I couldn't figure out where this failed conversion takes place.

It would be great if we could fix this error before we merge the major changes you have been working on in the past few weeks, so that I can make a stable current release before merging the new. Thanks!

fredlcore, 2020-11-09:
It could be that part of the problem are these lines of code:
```
//            case VT_HOUR_MINUTES: temp_val = ((pps_values[line-15000] / 6) * 256) + ((pps_values[line-15000] % 6) * 10); break;
            case VT_HOUR_MINUTES: temp_val = (pps_values[line-15000] / 6) + ((pps_values[line-15000] % 6) * 10); break;
```
in combination with the printTime() function. PrintTime() sets decodedTelegram.value without doing any further calculations for all bus types. However, for the debug output (which is, however, also wrong), it does divisions and modulo calculations.
Off the cuff, I would assume that the above commented-out calculation for temp_val is correct rather than the currently active one.

fredlcore, 2020-11-09:
Ok, I think the problem has to do with my handling of PPS commands, I think I'll need to dig further before I get back to you (or will have hopefully solved the matter by myself :) )...

fredlcore, 2020-11-09:
Ok, I think I found at least something that I don't understand:
In this commit
https://github.com/fredlcore/bsb_lan/commit/e02236d971a849d365152be40c565491c963dfcc
you made several changes in the printTelegram function, including the removal of a lot of conditions in line 2263 at that time. I'm not sure whether this is the reason, but I just realized that for some time now PPS telegrams are no longer decoded to plain text in the serial monitor when they are received from the heater. This may have to do with the fact that the four byte command ID for PPS telegrams actually does not exist, rather it is just one byte which I put into the second byte from left (0x00FF0000) and then check the command table for it. 

On the other hand, the calculations of that fake PPS command ID don't really make sense when printTelegram is called with -1 as the second parameter, so maybe that's (one of) the reason(s).

I hope I make a bit of sense here. Maybe I just have to install an earlier version and see if it works then in order to have a better idea when the problem started...

dukess, 2020-11-10:
Got it. I'm trying to figure out the problem.

dukess, 2020-11-10:
>lot of conditions in line 2263 at that time.

These conditions are actually contained in lines 2237-2240: pps_cmd is transformed to cmd, instead of cmd to pps_cmd.

> It could be that part of the problem are these lines of code:
>
> //            case VT_HOUR_MINUTES: temp_val = ((pps_values[line-15000] / 6) * 256) + ((pps_values[line-15000] % 6) * 10); [...]

time is set as 
pps_values[cmd_no] = h * 6 + m / 10;
so, If i understand correctly, calculation must be these:
case VT_HOUR_MINUTES: temp_val = pps_values[line-15000]; break;

I am very weak in understanding PPS bus, so I try not to change anything in it.
For this reason, the query() function did not return anything to decodedTelegram.value, so we can add
```
          switch (decodedTelegram.type) {
//            case VT_TEMP: temp_val = pps_values[(c & 0xFF)] * 64; break:
            case VT_BYTE: temp_val = pps_values[(line-15000)] * 256; sprintf_P(decodedTelegram.value, PSTR("%d"), temp_val); break;
            case VT_ONOFF:
            case VT_YESNO: temp_val = pps_values[(line-15000)] * 256; sprintf_P(decodedTelegram.value, PSTR("%d"), temp_val); decodedTelegram.isswitch = 1; break;
//            case VT_HOUR_MINUTES: temp_val = ((pps_values[line-15000] / 6) * 256) + ((pps_values[line-15000] % 6) * 10); break;
            case VT_HOUR_MINUTES: {
              temp_val = pps_values[line-15000];
              sprintf_P(decodedTelegram.value, PSTR("%02d:%02d"), pps_values[line-15000] / 6, (pps_values[line-15000] % 6) * 10); break;
            }
            default: temp_val = pps_values[(line-15000)]; sprintf_P(decodedTelegram.value, PSTR("%d"), temp_val);  break;
          }
```

fredlcore, 2020-11-10:
Thanks a lot for your input, I'll have a closer look at it - I'm just wondering: If you haven't set decodedTelegram.value before, how comes that when querying a PPS parameter, it still gets the value (even though its calculation is a bit off)? For that, decodedTelegram.value would have to be filled, isn't it?

dukess, 2020-11-10:
For the PPS bus, values are stored in a separate array that does not exchange data with other variables. That's why I couldn't figure out what data to show to the user and left everything as it is, without changes. So I need a big consultation.

fredlcore, 2020-11-10:
Found the problem, it was ineed all my fault ;) - the condition you removed was already faulty because at the beginning I converted the 1 byte PPS command IDs to the 4 byte BSB command ID style so that I could use the same cmdtbl. I just put the PPS byte into position 0x00FF0000 as mentioned above. But later on, I also encoded the pps_value variable index into 0x000000FF position. So the old condition could never match anymore, so in the serial monitor there were no longer decoded telegrams. But now it works again after just one line change ;).

As for the display in the web interface, it was similarly trivial: I could just do away with the special case for VT_HOUR_MINUTES and just leave the pps_value content untouched. I only had to change the sprintf part in printTime() like the one a few lines below because it doesn't fit the scheme that we use for BSB.

I'm sorry that the PPS implementation is so much piggybacking onto the BSB-LAN part, but the good thing is that once it runs, there won't be any need to improve it further because there are no new machines or parameters coming up ;)...

closed 2020-11-10 by fredlcore