fredlcore/BSB-LAN #168: DS18B20 sensors routines may be optimised

Youpimatin, 2020-11-09:
Hi,
here is my setup :
- I have 10 DS18B20 sensors connected on the arduino board
- I have 17 log_parameters total (DS18B20 included)
- I've set BSB-lan to log parameters with a log_interval of 10 seconds (it's short, but it's for testing)

When I hook up my oscilloscope on the 1-wire, I can see this :  
[image]

Timing are (19.92s total cycle time) :
- 8.08s 1-wire data
- 800ms doing something else (or waiting)
- 720ms 1-wire data
- 1.84s doing something else (or waiting)
- 960ms 1-wire data
- 7.52s doing something else (or waiting)

I suspect the LONG polling time of 8.08s is issuing a **sensors.requestTemperatures();** each time one of the 10 sensors are to be polled, I _think_ it could be done once, not multiple times.

Youpimatin, 2020-11-09:
Code investigations :

I think the problem resides in the logger section :

#ifdef LOGGER
...
#ifdef ONE_WIRE_BUS
            if (log_parameters[i] >= 20200 && log_parameters[i] < 20300) {
              int log_sensor = log_parameters[i] - 20200;
              **sensors.requestTemperatures(); // Send the command to get temperatures**
              float t=sensors.getTempCByIndex(log_sensor);
              dataFile.print(F("1W Temperature "));
              dataFile.print(log_sensor);
              dataFile.print(F(";"));
              dataFile.println(t);
            }
#endif

**sensors.requestTemperatures();** is called each time a sensor temperature is logged

fredlcore, 2020-11-09:
If you have a code improvement, please submit a pull request, thanks!

closed 2020-11-09 by fredlcore

Youpimatin, 2020-11-09:
The coldness of the answer does not really give me enthusiasm.
I wasn't here to complain, but to help.

fredlcore, 2020-11-09:
Sorry, I didn't see it as a complaint, and a "please" and "thanks" in the same sentence doesn't sound cold to me, but it's late in the evening and as a full-time working father of three kids, I have to limit my time to hobby projects like this and get to matters in a focused way, especially when it concerns "add-on features" that are not part of the core functionality of BSB-LAN (to which I would count external sensors). 
But to make it clear: It seems you have found a significant way to improve the polling of the sensors, and once you have found a way how this should be done in a better way, I'm honestly looking forward to get to know how this could be improved and I'd be grateful for that. A pull request would be the standard for this here on GitHub. If in the end it's just a one-liner, you can also post it here and I'll add it myself, but anything larger should be submitted through proper channels.

Youpimatin, 2020-11-09:
OK, I understand (and you beat me with three, as I have two)
Thanks a lot for taking time to read and answer.

I think I don't feel comfortable enough to publicly modify a program that is not my "baby".
And you know the way your code works far better than I would, even if I took hours to study it.

It may be a one-liner (guilty : on my copy of BSB-lan I just commented the line-in-bold, and it seems to work better _for me_, as it now happily throws MQTT frames every 10 seconds, but I know it's not the way it should be done)

If I find a way to make it better I'll tell you.

Other little things about BSB-lan :
I've also done some work on the LANG_FR.h file, but I don't feel to publish it as i'm not 100% sure of each and every line I've translated, and I can take my mistakes, as long as they don't ... "impact" anyone

Ah, and i _think_ (still careful) I've seen errors in the BSB-lan v3 PCB, if I remember well :
- the BC547 transistor should be labelled CBE from left to right on the silkscreen on the PCB
- the BC557 should be the turned upside-down (with the flat part up) and labelled EBC from left to right on the silkscreen on the PCB (and I know it works, even if it is mounted in reverse from the schematics, but it works a little less well)

fredlcore, 2020-11-09:
Thanks a lot for your comprehensive reply and taking your time for this project. I'm always looking forward to improvements, especially since I myself do not have all the add-ons working that have been added to BSB-LAN in the past (and this includes the DS18B20 sensors which I don't have and thus can't test). Therefore, I don't know in that case whether the polling is optional in each run or what time-frame should be observed so that most up-to-date values are available. But you are most likely right that once per polling interval should be enough and I'll look into that.

As for the language files: If there are terms not yet translated, then feel free to submit them, it's better to have something rather than nothing. As for the technical terms, I always double check with the manual to figure out what terms are used there. That's most likely the best option.

And thanks a lot especially for the suggestion regarding the PCB layout - the first error is already corrected in the current PCB (have a look here: https://github.com/fredlcore/bsb_lan/blob/master/schematics/BSB-Board%20plain.jpg
However, the second one did not occur to me until you mentioned it. Just to be sure: To my understanding, the collector in the BC557 in the schematic is the pin with the arrow pointing to the base and the emitter is going to the R3 / 300 Ohm resistor (https://www.elektronik-kompendium.de/sites/bau/schalt/02012912.gif)
The BC557s that I'm using have the collector on the left and emitter on the right when looking at the flat side of the transistor:
https://cdn-reichelt.de/documents/datenblatt/A100/BC556_BC557_BC558-CDIL.pdf
So, to my understanding both the labeling as well as the facing of the transistor should be correct? 
Or do you suggest that the schematics are wrong? Because if the schematics are correct, EAGLE does not give me the option to just turn around the silk screen. I could only change the labels of the pins (which I added myself). But to my understanding, using the above mentioned transistor type, these should be correct?

But in any case, thanks for highlighting this!

Youpimatin, 2020-11-10:
Oh yes pin labels are OK on the V4 ! (I made my board from the V3)

"To my understanding, the collector in the BC557 in the schematic is the pin with the arrow pointing to the base and the emitter is going to the R3 / 300 Ohm resistor"

I think the error comes from the previous sentence, because for bipolar transistors (NPN and PNP), on the symbol, the arrow is always on the Emitter (pointing out on the NPN ; pointing towards the base on the PNP)

(And the PNP pins naming has always been weird for me, because the pin that "collects" current is the Emitter, and the pin that "emits" current is the "Collector"))

- The BC557 is OK on the shematic
- The datasheet is OK too

But, the PCB (even the V4) is wrong, as the pin that goes to the 300 Ohm resistor is the Emitter, and it should be the Collector.
So, The BC557 should be mounted upside down to conform to the shematic.

fredlcore, 2020-11-10:
Thanks a lot for the clarification! What I don't understand is why EAGLE come up with a wrong silkscreen if the underlying schematic is correct. Or do they assume that a PNP transistor has pins in the order of Emitter, Base, Collector when facing the flat side of the transistor? Then I would have to recommend people to order a different brand of BC557 so that it fits the silkscreen.

Youpimatin, 2020-11-10:
Doing a quick search leads me to :

[https://www.eevblog.com/forum/eagle/eagle-pcb-layout-editor-pnp-transistor-e-and-c-pins-swapped/](url)

There seems to exist errors in EAGLE transistor libraries.

fredlcore, 2020-11-10:
Thanks a lot for the link, so I'll have to adjust the part in EAGLE then. But could you explain to me why it is still working if the transistor is effectivley inserted the wrong way? Or to put it differently, what would be the (measurable) improvement when it's done right?

Youpimatin, 2020-11-10:
See [Wikipedia](https://en.wikipedia.org/wiki/Bipolar_junction_transistor).

In brief form:  Emitter and Collector are P-doped, but Emitter is largely doped compared to Collector, so there is a transistor effect in reverse, but it is weak compared to the "normal" way to use the transistor.

In long form, with your schematic, as it is designed:
Say Veb is 0.7V when saturated, you have 0.55 mA throuh the base
With a typical gain of 150, you get 83mA max saturation current you _could_ use in the opto.
Since the opto gets about 5.8 mA from the 300 Ohms resistor,
So you could use a transistor with a lower gain (of about 10)

... and if I remember well, it's what is your reversed transistor, a transistor with a lower gain (but I don't know by what amout lower)

fredlcore, 2020-11-10:
Ok, thanks a lot, now I get the idea... But that would mean using the same transistor with the same gain, a stronger current would flow through the opto and its LED. So I would have to adjust the resistor accordingly? Or is the resistor calculated correctly and it was just that the LED in the opto was working on its lower limit?

Youpimatin, 2020-11-10:
Two hypothesis in changing the transistor orientation :

- If the transistor was in his linear region, it will now flow more current through the LED
- If the transistor was saturated, it won't flow more current through the opto's LED

But in any case, the current will never be more than what allows the 300 Ohms resistor, so you don't have to change it.

fredlcore, 2020-11-10:
Great, thanks a lot again, I learned a lot in these past few posts :)...

Youpimatin, 2020-11-10:
You're welcome

fredlcore, 2020-11-10:
I also moved the sensor request code outside that loop and it should work as this is how it's done in other parts of the code as well. Thanks again for highlighting this matter!

dukess, 2020-11-10:
@Youpimatin 
in new proposed version sensors.requestTemperatures() called once in 25 seconds (time can be ajusted).

Youpimatin, 2020-11-10:
Seems promising !

That was the best idea I had (associated with a caching of the values for the other functions if there's some free RAM) but I wouldn't know how to do it !

Since most of the time temperatures can't really change during a few seconds, I think it should be perfect if it works this way.

As soon as I can test it, I'll report what my 'scope says

Thank !

dukess, 2020-11-10:
Thanks for idea. I think it can be realized after leaving in past compatibility with Mega. Mega just not have enough RAM.

Youpimatin, 2020-11-10:
You say "in new proposed version", I can't beleive ... is it already in the master branch ?
or is it work in progress on your local node ?

dukess, 2020-11-10:
It closer than you can imagine :-)
https://github.com/fredlcore/bsb_lan/pull/154

Youpimatin, 2020-11-10:
Ah okay, in fact it was already "in the pipe"
You had already changed the 1-wire in late september !

"* I changed the logic to poll One wire bus: command to request new values is now send once in  ONE_WIRE_REQUESTS_PERIOD ms (15 sec) instead command sending every time when temperature is request.
  Does it make sense to poll sensors more often than once a minute?"

"...called once in 25 seconds (time can be ajusted)."

I think I'll use 15 or 30 seconds, and making this configurable is awesome !

_Reading the discussion, I'm just scared if it will work without external EEPROM because I don't have any (yet)_

dukess, 2020-11-10:
We need testers, because i have not Dallas too. :)

Youpimatin, 2020-11-10:
I can do that ! (keeping my current working version nearby in case of problem)

I have 10 DS18B20, but I don't have any DHT22 (I have some BME280 instead)

dukess, 2020-11-10:
Note: DS18B20 now is "programs", so if You need log it You need add these programs to log list.

Youpimatin, 2020-11-10:
OK, I've tried the last version tonight, and it doesn't work at all for me.
I think I configured well the BSB_lan_config.h

the serial monitor gives :

23:35:45.691 -> READY
23:35:45.691 -> EEPROM not ready
23:35:45.691 -> Reading EEPROM...done.
23:35:45.691 -> Size of cmdtbl1: 45984
23:35:45.691 -> Size of cmdtbl2: 44856
23:35:45.691 -> free RAM: 85863
23:35:45.738 -> Init One Wire bus...
23:35:45.738 -> Starting SD..ok
23:35:46.277 -> 192.168.1.200
23:35:46.277 -> Waiting 3 seconds to give Ethernet shield time to get ready...
23:35:46.277 -> Calculating free space on SD...30082 MB free

...nothing more

I can ping the IP of BSB-lan, but I get no HTTP.

I flashed my older version, it works well :

23:38:27.497 -> READY
23:38:27.497 -> Size of cmdtbl1: 44952
23:38:27.497 -> Size of cmdtbl2: 43440
23:38:27.497 -> free RAM:85879
23:38:27.497 -> Reading EEPROM...
23:38:27.497 -> Starting SD..ok
23:38:28.059 -> 192.168.1.200
23:38:28.059 -> Waiting 3 seconds to give Ethernet shield time to get ready...
23:38:28.059 -> Calculating free space on SD...30082 MB free
23:38:31.763 -> numSensors: 10
23:38:32.138 -> LAN->HEIZ QUR 6225 Configuration - Famille d'appareil: 
23:38:32.185 -> DC C2 00 0B 06 3D 05 00 02 52 88 
23:38:32.185 -> HEIZ->LAN ANS 6225 Configuration - Famille d'appareil: 80
23:38:32.185 -> DC 80 42 0E 07 05 3D 00 02 00 00 50 68 6D 
23:38:32.232 -> #6225:  80
23:38:32.419 -> LAN->HEIZ QUR 6226 Configuration - Variante d'appareil: 
23:38:32.419 -> DC C2 00 0B 06 3D 05 00 03 42 A9 
23:38:32.466 -> HEIZ->LAN ANS 6226 Configuration - Variante d'appareil: 127
23:38:32.466 -> DC 80 42 0E 07 05 3D 00 03 00 00 7F CB 54 
23:38:32.466 -> #6226:  127
23:38:32.513 -> Device family: 80
23:38:32.513 -> Device variant: 127
23:38:35.091 -> DISP->HEIZ QUR 8310 Diagnostic générateur - T° de chaudière: 
23:38:35.091 -> DC 8A 00 0B 06 3D 0D 05 19 4F 8C 
23:38:35.185 -> HEIZ->DISP ANS 8310 Diagnostic générateur - T° de chaudière: 37.9 &deg;C
23:38:35.185 -> DC 80 0A 0E 07 0D 3D 05 19 00 09 7C BA 08 
23:38:35.419 -> HEIZ->01 INF      0D0901F6 00 00 00 00 00 
23:38:35.419 -> DE 80 01 10 02 0D 09 01 F6 00 00 00 00 00 C4 5E 
23:38:35.747 -> HEIZ->01 INF      0D0901F6 00 00 00 00 00 
23:38:35.794 -> DE 80 01 10 02 0D 09 01 F6 00 00 00 00 00 C4 5E 
23:38:36.122 -> HEIZ->01 INF      0D0901F6 00 00 00 00 00 
23:38:36.122 -> DE 80 01 10 02 0D 09 01 F6 00 00 00 00 00 C4 5E 
23:38:42.404 -> POST ...
...

fredlcore, 2020-11-11:
I have the same problem and it seems to have to do with the fixed IP. If you enable DHCP, it works. Just a workarounds for the moment...

Youpimatin, 2020-11-14:
I've just taken the time to test the last version :

Good news is the fixed IP works, now, but I have bad news for the DS18B20 sensors.
The program sees the 10 sensors ...
[image]

...but indexes are 0, 2, 4, etc... instead of 0, 1, 2, etc...
[image]

And when page settings is accessed, BSB-lan locks.

Oscillocope says there is 480ms activity when starting the program, but no more activity can be observed later.

Notes, I updated and changed some settings from my last post :
byte UseEEPROM = 0;
byte DHT_Pins[10] = {2, 7};

Back to last BSB-lan woking version for the moment, but ready to test a new one when needed !

dukess, 2020-11-14:
I prepare patch that fix incorrect numbers, but... Even now, something must be showing.
When You flash new version, can You show output for programs 20300-20499?
In new version DHT, Dallas and Max sensors can be accessed individually by program numbers.
If 20300-20499 values will be empty then i did something wrong when rewrite code.

fredlcore, 2020-11-14:
If you don't have sensors attached, then the values should be empty, right?

Youpimatin, 2020-11-14:
It's better with the very last modifications :
`
20000 One Wire, DHT & MAX! Sensors - Burner runtime level 1: 0 s |   |  
-- | -- | --
20001 One Wire, DHT & MAX! Sensors - Burner cycles level 1: 0 |   |  
20002 One Wire, DHT & MAX! Sensors - Burner runtime level 2: 0 s |   |  
20003 One Wire, DHT & MAX! Sensors - Burner cycles level 2: 0 |   |  
20004 One Wire, DHT & MAX! Sensors - DHW runtime: 0 s |   |  
20005 One Wire, DHT & MAX! Sensors - DHW cycles: 0 |   |  
20050 One Wire, DHT & MAX! Sensors - 24h average. Température extérieure locale: 0.0 °C |   |  
20100 One Wire, DHT & MAX! Sensors - DHT22 Sensor ID #1: (parameter not supported) |  
20101 One Wire, DHT & MAX! Sensors - DHT22 Sensor Temperatur #1: °C (parameter not supported) |  
20102 One Wire, DHT & MAX! Sensors - DHT22 Sensor Luftfeuchtigkeit #1: % (parameter not supported) |  
20103 One Wire, DHT & MAX! Sensors - DHT22 Sensor Abs Luftfeuchtigkeit #1: % (parameter not supported) |  
20104 One Wire, DHT & MAX! Sensors - DHT22 Sensor ID #2: (parameter not supported) |  
20105 One Wire, DHT & MAX! Sensors - DHT22 Sensor Temperatur #2: °C (parameter not supported) |  
20106 One Wire, DHT & MAX! Sensors - DHT22 Sensor Luftfeuchtigkeit #2: % (parameter not supported) |  
20107 One Wire, DHT & MAX! Sensors - DHT22 Sensor Abs Luftfeuchtigkeit #2: % (parameter not supported) |  
20300 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #1: 18.81 |   |  
20302 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #2: 22.38 |   |  
20304 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #3: 18.62 |   |  
20306 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #4: 18.69 |   |  
20308 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #5: 18.31 |   |  
20310 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #6: 18.75 |   |  
20312 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #7: 18.87 |   |  
20314 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #8: 18.31 |   |  
20316 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #9: 18.75 |   |  
20318 One Wire, DHT & MAX! Sensors - DS18B20 Sensor ID #10: 18.50
`

'Sensors' page :
`
1w_temp[0] 18.81:
--
1w_temp[1] 22.31:
1w_temp[2] 18.62:
1w_temp[3] 18.69:
1w_temp[4] 18.31:
1w_temp[5] 18.75:
1w_temp[6] 18.87:
1w_temp[7] 18.31:
1w_temp[8] 18.75:
1w_temp[9] 18.50:
`

Settings page extract :
`
Logging the following parameters every 15 seconds: Yes

8700 - Température extérieure locale
8740 - Température ambiante 1
8741 - Consigne actuelle de température Ambiante
8310 - Température de chaudière
8311 - Consigne de température de chaudière
8743 - Température de départ circuit de chauffe 1
8744 - Consigne de départ H1
20300 - DS18B20 Sensor ID
20301 -
20302 - DS18B20 Sensor ID
20303 -
20304 - DS18B20 Sensor ID
20305 -
20306 - DS18B20 Sensor ID
20307 -
20308 - DS18B20 Sensor ID
20309 - 
`

I did not imagine that sensors were skipping numbers, so I set 20300 to 20309 and not 20300, 20302... 20310.
I don't know if it's a feature or a bug.

fredlcore, 2020-11-14:
The skipping is probably a bug?

dukess, 2020-11-14:
Thank you!
The idea was this: programs 20301, 20303 etc. should return temperature values, and programs 20300, 20302 etc. should return sensor numbers (addresses) so that you can somehow distinguish. Same behavior for DHT and MAX.
But here I see that the temperature is returned in even programs, which is wrong. I'm looking for an error.

fredlcore, 2020-11-14:
For DHT we also have two types of values (temperature and humidity), how would that fit into the numbers?

dukess, 2020-11-14:
4 types in cycle:
20100, 20104, etc - DHT pins
20101, 20105, etc - temperature
20102, 20106, etc - humidity
20103, 20107, etc - abs humidity

dukess, 2020-11-14:
For MAX 4 types in cycle too:
20500, 20504, etc - device name
20501, 20505, etc - current temperature
20502, 20506, etc - target temperature
20503, 20507, etc - valve

fredlcore, 2020-11-14:
Thanks! @1coderookie : This is the correct enumeration, not the one that I mentioned, sorry!

Youpimatin, 2020-11-14:
Ah, smarter !

OK, testing the up-to-date version...

Youpimatin, 2020-11-14:
Ah, this time no more temperatures, but IDs are OK !
[image]

[image]

(On the oscilloscope, there is activity on the 1-wire pin)

dukess, 2020-11-14:
More better! :)

dukess, 2020-11-14:
Can i ask you send to me debug (Serial or telnet) output?
really, i can't find problem.

Youpimatin, 2020-11-14:
No problem :

21:46:42.230 -> READY
21:46:42.230 -> EEPROM not ready
21:46:42.230 -> Using settings from config file
21:46:42.230 -> done.
21:46:42.230 -> EEPROM schema v.0 Program schema v.2
21:46:42.230 -> Address EEPROM option 0: 0
21:46:42.230 -> Address EEPROM option 1: 1
21:46:42.230 -> Address EEPROM option 2: 2
21:46:42.230 -> Address EEPROM option 3: 6
21:46:42.230 -> Address EEPROM option 4: 226
21:46:42.230 -> Address EEPROM option 5: 306
21:46:42.230 -> Address EEPROM option 6: 418
21:46:42.230 -> Address EEPROM option 7: 419
21:46:42.230 -> Address EEPROM option 8: 420
21:46:42.230 -> Address EEPROM option 9: 421
21:46:42.230 -> Address EEPROM option 10: 422
21:46:42.230 -> Address EEPROM option 11: 423
21:46:42.230 -> Address EEPROM option 12: 427
21:46:42.277 -> Address EEPROM option 13: 428
21:46:42.277 -> Address EEPROM option 14: 588
21:46:42.277 -> Address EEPROM option 15: 589
21:46:42.277 -> Address EEPROM option 16: 593
21:46:42.277 -> Address EEPROM option 17: 753
21:46:42.277 -> Address EEPROM option 18: 759
21:46:42.277 -> Address EEPROM option 19: 760
21:46:42.277 -> Address EEPROM option 20: 764
21:46:42.277 -> Address EEPROM option 21: 768
21:46:42.277 -> Address EEPROM option 22: 772
21:46:42.277 -> Address EEPROM option 23: 776
21:46:42.277 -> Address EEPROM option 24: 778
21:46:42.277 -> Address EEPROM option 25: 782
21:46:42.277 -> Address EEPROM option 26: 786
21:46:42.277 -> Address EEPROM option 27: 850
21:46:42.277 -> Address EEPROM option 28: 914
21:46:42.277 -> Address EEPROM option 29: 915
21:46:42.324 -> Address EEPROM option 30: 916
21:46:42.324 -> Address EEPROM option 31: 926
21:46:42.324 -> Address EEPROM option 32: 927
21:46:42.324 -> Address EEPROM option 33: 1087
21:46:42.324 -> Address EEPROM option 34: 1088
21:46:42.324 -> Address EEPROM option 35: 1092
21:46:42.324 -> Address EEPROM option 36: 1093
21:46:42.324 -> Address EEPROM option 37: 1094
21:46:42.324 -> Address EEPROM option 38: 1095
21:46:42.324 -> Address EEPROM option 39: 1099
21:46:42.324 -> Address EEPROM option 40: 1131
21:46:42.324 -> Address EEPROM option 41: 1163
21:46:42.324 -> Address EEPROM option 42: 1195
21:46:42.324 -> Address EEPROM option 43: 1227
21:46:42.324 -> Size of cmdtbl1: 32784
21:46:42.324 -> Size of cmdtbl2: 24936
21:46:42.324 -> Size of cmdtbl3: 33264
21:46:42.324 -> free RAM: 85807
21:46:42.324 -> Init One Wire bus...
21:46:42.793 -> numSensors: 10
21:46:42.793 -> PPS settings:
21:46:42.793 -> Starting SD..ok
21:46:43.402 -> 192.168.1.200
21:46:43.402 -> 255.255.255.0
21:46:43.402 -> 192.168.1.1
21:46:43.402 -> Waiting 3 seconds to give Ethernet shield time to get ready...
21:46:43.402 -> Calculating free space on SD...30003 MB free
21:46:46.590 -> Start network services
21:46:47.012 -> LAN->HEIZ QUR 6225 Configuration - Famille d'appareil: 
21:46:47.012 -> dc c2 00 0b 06 3d 05 00 02 52 88 
21:46:47.012 -> HEIZ->LAN ANS 6225 Configuration - Famille d'appareil: 80
21:46:47.012 -> dc 80 42 0e 07 05 3d 00 02 00 00 50 68 6d 
21:46:47.012 -> #6225:  80
[... 134 lines cut ...]
21:47:04.778 -> HEIZ->01 INF      0D0901F6 00 00 00 00 00 
21:47:04.778 -> de 80 01 10 02 0d 09 01 f6 00 00 00 00 00 c4 5e 
21:47:05.106 -> HEIZ->01 INF      0D0901F6 00 00 00 00 00 
21:47:05.106 -> de 80 01 10 02 0d 09 01 f6 00 00 00 00 00 c4 5e 
21:47:06.278 -> POST /XXXXXXXXXXXXXXXX/JQ?Parameter=8740,710,700 HTTP/1.1

21:47:06.278 -> /XXXXXXXXXXXXXXXX/JQ?Parameter=8740,710,700
21:47:06.278 -> URL: JQ?Parameter=8740,710,700
21:47:06.466 -> LAN->HEIZ QUR 8740 Diagnostic consommateurs - Température ambiante 1: 
21:47:06.466 -> dc c2 00 0b 06 3d 2d 05 1e 51 a7 
21:47:06.466 -> HEIZ->LAN ANS 8740 Diagnostic consommateurs - Température ambiante 1: 19.2 &deg;C
21:47:06.466 -> dc 80 42 0e 07 2d 3d 05 1e 00 04 cb 1c e2 
21:47:06.466 -> #8740:  19.2 &deg;C
21:47:06.653 -> LAN->HEIZ QUR  710 Circuit de chauffe 1 - Consigne confort temp.ambiante CC1: 
21:47:06.653 -> dc c2 00 0b 06 3d 2d 05 8e d2 1e 
21:47:06.653 -> HEIZ->LAN ANS  710 Circuit de chauffe 1 - Consigne confort temp.ambiante CC1: 16.0 &deg;C
21:47:06.653 -> dc 80 42 0e 07 2d 3d 05 8e 00 04 00 b2 5a 
21:47:06.700 -> #710:  16.0 &deg;C
21:47:06.841 -> LAN->HEIZ QUR  700 Circuit de chauffe 1 - Régime Circuit de chauffe  1: 
21:47:06.841 -> dc c2 00 0b 06 3d 2d 05 74 9c 4b 
21:47:06.841 -> HEIZ->LAN ANS  700 Circuit de chauffe 1 - Régime Circuit de chauffe  1: 1 - Automatik
21:47:06.888 -> dc 80 42 0d 07 2d 3d 05 74 00 01 a7 52 
21:47:06.888 -> #700:  1 - Automatik

If you need anything else, please feel free to ask, if I can help you I'll do :)

dukess, 2020-11-14:
Bugfix: https://github.com/fredlcore/bsb_lan/pull/177

Youpimatin, 2020-11-15:
Harder, better, faster, stronger !

It works as expected, I'm switching to the last version on my boiler permanently.

Changed the "Home Assistant" configuration to get the new sensors ID, perfect !

If if find bugs I'll report.

THANK YOU !

dukess, 2020-11-15:
Thank you too! Your reports was very helpful!

Youpimatin, 2020-11-16:
A little step back for the moment, returning to the 1.1 version.

Maybe I've done something wrong,
or maybe there is a breaking change,
but the BSB-lan thermostat integration in "Home Assistant" doesn't allow me anymore to change the temperature.

I'll check later if I can find what happens

dukess, 2020-11-16:
If you use PPS bus now then here was bug with terminal type selection (QAA70 or QAA50) until yesterday.