fredlcore/BSB-LAN #170: Please remove leading spaces via MQTT

FunkOdyssey2001, 2020-11-10:
I have several problems,  because there seems to be leading spaces in the MQTT output. 
For example all error strings (e.g. 6800). 
Please see my forum topic here, too: https://forum.fhem.de/index.php/topic,115705.msg1100143.html#msg1100143

fredlcore, 2020-11-10:
Since the other contributors here don't speak German, it's not helpful to link to a German forum thread. Please describe the errors here in detail as well as observations you may have made. Thanks.

FunkOdyssey2001, 2020-11-10:
Sorry, but this is really simple. There are always leading spaces before string MQTT output. 

The link should just show the way how I was pointed to this behavior.

FunkOdyssey2001, 2020-11-11:
Perhaps references with #116

dukess, 2020-11-13:
In new version i will propose to change DHT22 requests: accessing to sensors will do through programs, i.e. even programs will returns temperature, odd - humidity

FunkOdyssey2001, 2020-11-16:
> In new version i will propose to change DHT22 requests: accessing to sensors will do through programs, i.e. even programs will returns temperature, odd - humidity

I don't understand this in detail.  Does this have anything to do with the leading spaces?

fredlcore, 2020-11-16:
Maybe. The whole way of accessing sensors is now through parameter numbers, so these are no longer handled differently. Therefore, the leading space should hopefully disappear. Please check.

FunkOdyssey2001, 2020-11-16:
I will try the new v2.0 branch soon.

FunkOdyssey2001, 2020-11-18:
In Version 2.0.14-20201117142051 no differences. There are still leading spaces.

`byte mqtt_mode = 1;` 

```
2020-11-18_16:12:10 bsb History01_FAPhase:  16 - not found
2020-11-18_16:12:07 bsb Betriebsart:  1 - Automatik
```

There are no leading spaces for numeric parameters:

```
2020-11-18_16:02:22 bsb Raumtemperatur1: 21.0
2020-11-18_16:02:22 bsb Vorlaufsollwert1: 19.0
```

fredlcore, 2020-11-19:
Latest update should fix this.

closed 2020-11-19 by fredlcore

FunkOdyssey2001, 2020-11-19:
Thank you very much.