fredlcore/BSB-LAN #173: http rfc compliant line ending = \r\n

matmoeh, 2020-11-13:
I'm using  homeassistant and tried to get values from bsb-lan. I am using some code i found, to get the values via the REST interface:

https://community.home-assistant.io/t/bsb-lan-integration/113501/61

But this did not work, because homeassistant does not like the BSB-LAN line ending: \n

Described here:
https://github.com/home-assistant/core/issues/43152#issuecomment-726832232

So i've changed all \n to \r\n and it is now working.

sr-as2, 2020-11-16:
`So i've changed all \n to \r\n and it is now working.`

Hi matmoeh,
i'm having the same problem, how did you do that ?

matmoeh, 2020-11-17:
I downloaded the source and replaced all single \n to \r\n.
This is not the best way, because i just wanted to replace Webserver/REST specific values. ... but i did not wanted to figure this out, so i replaced as many i found.

sr-as2, 2020-11-18:
Thanks, maybe @fredlcore can update this code ?
I update to BSB-Lan v2 and it seems better in Home Assistant 0.117 even if i see many "unavailable" for some BSB values. I don't know really why.

fredlcore, 2020-11-18:
Please test it with the new code, both of you. I think we have made the changes everywhere, but let us know if (and where) this is not the case.

sr-as2, 2020-11-18:
I'm using Version 2.0.9-20201116222422 and it's working...for some values.
For exemple, this one is working some times (lots of empty values):

```
  - platform: rest
    resource: http://192.168.0.39/JQ=8413
    name: pac_compresseur
    unit_of_measurement: "%"
    timeout: 30
    value_template: "{{ value_json['8413']['value'] }}"
```

[image]

dukess, 2020-11-20:
@sr-as2 Do you using Mega?

sr-as2, 2020-11-20:
Hi @dukess 
yes, it's a ArduinoIDE Mega2560 Due

dukess, 2020-11-20:
In this case i guess problem in addressing of third table "optbls". 
If you will, you can take fresh changes from my fork: https://github.com/dukess/bsb_lan

Today or tomorrow i will to make pull request for including last changes in main branch.

sr-as2, 2020-11-20:
ok great, thank you, i'll try it!

sr-as2, 2020-11-20:
I've just installed your fork but it's the same, some values are still unavailable with my rest sensors (but ok in the BSB lan interface)

dukess, 2020-11-20:
Thank you! Which sensors do you mean?
If you mean Dallas/DHT22 then in v 2.0 access to these was changed and work through programs 20100 - 20699

sr-as2, 2020-11-20:
I'm don't know if it's the same issue or if it's a Home Assistant issue...
It's start here:  https://github.com/home-assistant/core/issues/42589
Since a Home assitant update, rest sensors with BSB-lan were broken. Then i update to bsb-lan v2 and rest sensors are now working BUT some values are show as unavailable (not every time) in Home assistant.

sensors: 8000, 8001, 8413, 8323....

dukess, 2020-11-20:
Understood. I'll look at it tomorrow.

dukess, 2020-11-21:
If you want, you can try new commit: https://github.com/dukess/bsb_lan/commit/5ddf0ce3717f3001f0f52dfd4f552075df943d40

sr-as2, 2020-11-21:
Thank you, just try your new version. It seems to be worst than before...

```
Update of sensor.pac_pointdeconsignewp is taking over 10 seconds
Update of sensor.ecs_temperature is taking over 10 seconds
Update of sensor.ecs_state is taking over 10 seconds
Update of sensor.pac_consignedeparth2 is taking over 10 seconds
Update of sensor.pac_consignedeparth1 is taking over 10 seconds
```

dukess, 2020-11-21:
Thank you for testing!
Very strange, I did not add any significant changes that slow down the work.
Every request require ~650-750 ms with Mega.
How many parameters you request?

sr-as2, 2020-11-21:
I'm requesting 20 parameters. 
Now, with 15 it's working. Maybe there is one paramet that it doesn't like...
I'll try to add them one by one

For now, it's working! Thanks a lot @dukess

dukess, 2020-11-21:
If there's any news, tell me. :)

closed 2020-11-23 by fredlcore