fredlcore/BSB-LAN #175: Issues and questions still open from merging to version 2.0

fredlcore, 2020-11-14:
@dukess, I think I collect the small questions and issues I still have here in this issue and number them so that it's easier to reference them in answers.

fredlcore, 2020-11-14:
1. EEPROM-based configuration and webinterface: I have set useEEPROM to 1 in _config.h, so that values are stored in EEPROM. In the webinterface, however, "Read config from EEPROM" is still set to "off". Is that correct?

fredlcore, 2020-11-14:
2. For 1-Wire, there is only one configuration option which reads "Use One Wire bus on pin", but the selection here is on/off and not a pin number.

fredlcore, 2020-11-14:
3. Maybe we discussed this before, but the values for MAX! are only in the config, not in the webinterface. But turning it on and off is in the webinterface. Maybe we should do it either completely in the web or completely in _config.h?

dukess, 2020-11-14:
> EEPROM-based configuration and webinterface: I have set useEEPROM to 1 in _config.h, so that values are stored in EEPROM. In the webinterface, however, "Read config from EEPROM" is still set to "off". Is that correct?

It is correct until first save in EEPROM: you must enable "Read config from EEPROM" and then save, /CS. Since this moment configuration will be read from EEPROM.

If you set useEEPROM to 0, then values from config file will be used. You can edit config and save it in EEPROM, but at startup config file will be used.

>For 1-Wire, there is only one configuration option which reads "Use One Wire bus on pin", but the selection here is on/off and not a pin number.

This is a typo error. Sure, here must be text field. Catch the patch.

>Maybe we discussed this before, but the values for MAX! are only in the config, not in the webinterface. But turning it on and off is in the webinterface. Maybe we should do it either completely in the web or completely in _config.h?

In both cases EEPROM used for store MAX devices names and their addresses. In current version we can set list MAX devices through web interface. But i not implemented names lists conjunction from EEPROM and config file.

fredlcore, 2020-11-14:
Ok, just so that I get it right: 
In order to store **and** read configuration from EEPROM, I **must** set useEEPROM to 1?
Or is "useEEPROM" only for storing configuration in EEPROM, and in order to read it at startup, I have to set the webinterface "Read config from EEPROM" to "on"?

As for how to proceed after submit changes: I think the user should decide on whether restart or not. But parameters such as log and average parameters will take effect immediately, correct?

As for MAX: You say that EEPROM is used to store MAX device names (like "KEQ0502326"?) and addresses - what would be the address then? In current version, I can only find "on/off" for MAX. Can you explain a bit more?

dukess, 2020-11-14:
useEEPROM variable affects how configuration is read from EEPROM at startup. 0 - not read. 1- read.
Configuration can be write with any value.
```
 if(UseEEPROM){ //Read EEPROM when it allowed from config file. This value was set in config file. 
    SerialOutput->print(F("Reading EEPROM..."));
    readFromEEPROM(CF_USEEEPROM); //Now we read value from EEPROM into UseEEPROM.
  } else {
    SerialOutput->print(F("Using settings from config file\r\n"));
  }
  if(UseEEPROM == 0x96){//Read EEPROM when EEPROM contain magic byte (stored configuration)

```
i.e. you must set UseEEPROM to non-zero in config file **and** set "Read config from EEPROM" to "on" for reading config from EEPROM.

>But parameters such as log and average parameters will take effect immediately, correct?
You right.
I think we can add handlers for most numbers of parameters in future for change program behavior on the fly. (btw, bus type can't be switched without reboot yet).

>As for MAX: You say that EEPROM is used to store MAX device names (like "KEQ0502326"?) and addresses - what would be the address then? In current version, I can only find "on/off" for MAX. Can you explain a bit more?

These addresses stored in max_devices array automatically (after pressing "pair" button on device AFAIK). This array hidden from end user and cannot be changed directly.

dukess, 2020-11-14:
Note: i built code for Mega and run it. Mega starts sucessfully but not displayed programs above 140.
We need someone who can deal with addressing space in Mega.

fredlcore, 2020-11-14:
Ok, I think I got it with useEEPROM and "Read config from EEPROM" now :).

As for having to reboot when changing bus type: Why is it necessary to reboot because (at least previously) I could change the bus type via /P command on the fly?

MAX: All I see in config is this:
General | Enable MAX | on/off
General | CUNO/CUNX/modified MAX!Cube IP address | 192.168.1.5

The thing with the MAX device names (such as "KEQ502326") is that yes, they are only added to the array once they are detected, but the detected IDs were stored to EEPROM before, so you didn't have to repair them with every reboot. Also the IDs in max_device_list act as a limit to which devices should be monitored. (Previously, ) devices not in that list were not processed in case they sent messages (for example because they belong to the neighbour's flat). So I was wondering whether there is a web-based configuration option for the max_device_list variable?

fredlcore, 2020-11-14:
As for the Mega: I really think it's then time to drop support for it. It is also hard for me to test the code because it no longer compiles in my configuration, unless I disable logging. And with disabled logging, I won't notice any issues that come from log-related problems.

However, I would like to give Mega users the option to still be able to use the current BSB_lan_defs.h file so that they can still get updated parameter configurations. Would it be possible to keep the file with just two cmdtbl sections and not three as it seems to be now? Because for three cmdtbl sections, we would have to update the code for version 1.1 (which still only has two cmdtbl).

On the other hand, as soon as we have to introduce a new data type in the _defs.h, compilation will fail anyways because the main code does not know about these. Hm... so maybe it's better to make a clean cut here and now? Version 1.1 is good enough for most users anyway?

dukess, 2020-11-15:
> As for having to reboot when changing bus type: Why is it necessary to reboot because (at least previously) I could change the bus type via /P command on the fly?

At this time all vaues just writes to RAM (and EEPROM). For chaning bus type we need call a function with new value. So, as i told early, we can make handler little bit later.

>The thing with the MAX device names (such as "KEQ502326") is that yes,...

This option is exist. And it works on my device.  Very curiously why it not shown on your device.

>Mega

I tried to add support for three tables in the 1.1 version code, but everything led to assembler errors (not enough address space?). So i think revert back to two tables support.
With two tables 1.1 version can use current _defs with minimal modifications (Need to add one definition which non exist in new version to .ino).

fredlcore, 2020-11-15:
Reboot: Ah, got it, so /P still works because it direclty calls the bus change function.

Mega: That sounds great! With two tables, there should still be left space for several dozen new parameters, I think. Afterwards, we'll have to see...

MAX: 
This is how my settings look like:
[image]
Where should the device names appear?

fredlcore, 2020-11-15:
This is the _config.h section for MAX:
```
boolean enable_max_cul = true;                //enable or disable connection to CUNO/CUNX/modified MAX!Cube;
byte max_cul_ip_addr[4] = {192,168,1,5};     //  IP of CUNO/CUNX/modified MAX!Cube. Please use commas instead of dots!!!
char max_device_list[20][11] = {               // list of MAX! wall/heating thermostats that should be polled
  "KEQ0502326",
  "KEQ0505080",
  "KHA0002948",
  "KHA0002927"
};
```

dukess, 2020-11-15:
Looks like one more bug issue. List of MAX! devices is placed below MAX IP Address.
[image]

dukess, 2020-11-15:
Can you show source of config page? It may be error in http page.

fredlcore, 2020-11-15:
Sure:
```
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN">
<html><head>
<meta http-equiv="content-type" content="text/html; charset=UTF-8">
<meta name="viewport" content="width=device-width, initial-scale=1.0,  minimum-scale=1.0">
<title>BSB-LAN Web</title>
<style>A:link  {color:blue;text-decoration: none;} A:visited {color:blue;text-decoration: none;} A:hover {color:red;text-decoration: none;background-color:yellow} A:active {color:blue;text-decoration: none;} A:focus {color:red;text-decoration: none;}
table {width: 100%;  max-width: 1024px;  margin: auto;}
td {vertical-align: top;}
td.header {vertical-align: middle;}
input {width: 100%; box-sizing: border-box;} select {width: 100%;}</style>
</head><body>
<script>function set(line){
var value = document.getElementById('value'+line).value.replace(":", ".");
value = value.replace("---", "");
if(isNaN(value)==false || value == ''){
window.open(document.getElementById('main_link').href+'S'+line+'='+value,'_self');
}}
function setbit(line){
var x=document.getElementById('value'+line); var value=0;
for (var i=0; i<x.options.length; i++) {
if(x.options[i].selected){
value=value+eval(x.options[i].value);
}}
window.open(document.getElementById('main_link').href+'S'+line+'='+value,'_self');
}</script>
<font face='Arial'>
<center><h1><A ID=main_link HREF='/4444/'>BSB-LAN Web</A></h1></center>
<table align=center><tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/4444/K'>Heizungsfunktionen</a></td><td class="header" width=20% align=center><a href='/4444/T'>Sensoren</a></td><td class="header" width=20% align=center><a href='/4444/DG'>Zeichne Logdatei</a></td><td class="header" width=20% align=center><a href='/4444/Q'>Prüfe auf neue Parameter</a></td></tr>
<tr bgcolor=#f0f0f0><td class="header" width=20% align=center><a href='/4444/C'>Einstellungen</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN/anhang_b.html' target='_new'>URL-Befehle</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN/inhaltsverzeichnis.html' target='new'>Handbuch</a></td><td class="header" width=20% align=center><a href='https://1coderookie.github.io/BSB-LPB-LAN/kap15.html' target='_new'>FAQ</a></td></tr></table><p></p><table align=center><tr><td class="header">
Einstellungen<BR><BR>
Version: 2.0.1-20201114001531<BR>
Freier Speicher: 81263 Bytes <BR>
Uptime: 124361858<BR>
Bus-System: 
BSB (66, 0) Lesen/Schreiben<BR>
Monitor Modus: 0<BR>
Verbositäts-Level: 1<BR>
1-Wire Bus Pins: 
7, Sensoren: 0
<BR>
DHT22 Bus Pins: 
2, 3
<BR>
Gesperrte GPIO Pins: 
0 1 3 10 11 12 13 18 19 20 21 22 23 50 51 52 53 62 63 64 65 66 67 68 69 <BR>
MAC-Adresse: 
00:80:41:19:69:90<BR><BR>
Installierte Module: <BR>
WEBSERVER, IPWE, RESET, MQTT: Disabled, Verbose DEBUG: Serial, LOGGER, VERSION_CHECK: enabled<BR><BR>
Free space: 15269 MB<br>free clusters: 1954505<BR>freeClusterCount() call time: 2 microseconds<BR><br>
Berechnung von 24h-Mittelwerten für die folgenden Parameter: <BR>
8700 - Außentemperatur: 20050<BR>
8323 - Gebläsedrehzahl: 20051<BR>
8326 - Brennermodulation: 20052<BR>
<BR><BR>Loggen der folgenden Parameter alle 
3600 Sekunden: Ja<BR>
<BR>
8700 - Außentemperatur<BR>
8743 - Vorlauftemperatur 1<BR>
8314 - Kesselrücklauftemperatur Ist<BR>
BSB-Datentelegramme (unabhängig vom Intervall)<BR>
Nur unbekannte Bus-Telegramme loggen: Ja<BR>
Nur Broadcast Bus-Telegramme loggen: Ja<BR>
<BR>
<BR>
<form id="config" method="post" action="/4444/CI"><table align="center"><tbody>
<tr><td>General</td><td>
Read config from EEPROM</td><td>
[... 128 lines cut ...]
<option value='1'>Ja</option>
<option value='26989'>/K</option>
</select></td></td>
<tr><td>MQTT settings</td><td>
MQTT broker IP address</td><td>
<input type=text id='option_39' name='option_39' pattern='((^|\.)((25[0-5])|(2[0-4]\d)|(1\d\d)|([1-9]?\d))){4}' VALUE='192.168.1.20'></td></td>
<tr><td>MQTT settings</td><td>
User name</td><td>
<input type=text id='option_40' name='option_40'  VALUE='User'></td></td>
<tr><td>MQTT settings</td><td>
Password</td><td>
<input type=text id='option_41' name='option_41'  VALUE='Pass'></td></td>
<tr><td>MQTT settings</td><td>
Topic prefix</td><td>
<input type=text id='option_42' name='option_42'  VALUE='BSB-LAN'></td></td>
<tr><td>MQTT settings</td><td>
Topic prefix</td><td>
<input type=text id='option_43' name='option_43'  VALUE='MyHeater'></td></td>
</tbody></table><p><input type="submit"></p>
</form>
</td></tr></table>
</body>
</html>

```

dukess, 2020-11-15:
Stupid question: do you enable MAX_CUL definition?

dukess, 2020-11-15:
Hey! I look in config.default, MAX_CUL is not enabled by default.

fredlcore, 2020-11-15:
You're right, how embarrassing ;).

BTW: I would set up the config in this way: Enable all modules by default and then add a conditional for AVR/MEGA where the memory intensive modules are removed again using #undef. 
Because currently, it's a bit confusing for new users who think they have disabled a function and forget to look all the way at the bottom to see that the value is actually activated for the Due down there.

fredlcore, 2020-11-15:
One more thing: The QAA type is stated as QAA50 for me, although I have defined is as 0x53 for QAA70 in the _config.h. Is this a bug?

1coderookie, 2020-11-15:
I join this conversation with some small issues I just found.
FYI: I didn't change anything in the standard configuration within BSB_lan_config.h (besides the useage of the fix IP 88). 
This is what I found:
1) PPS bus write is OFF (default), QAA-type should be 0x53/QAA70 as it is pre-set in the config.h. But: Within the webconfig the QAA50 is diplayed instead of the QAA70, see screenshot below. In the screenshot from @fredlcore above, the QAA-type/-model isn't displayed at all with the setting 'PPS bus write ON' (but I didn't check that in my case!).
[image]

 2) Just to make it clear, maybe it would be nice to add "(seconds)" behind "Log interval" in this line:
[image]

3) I'm missing the option to grant general write-access (in config.h it's "#define DEFAULT_FLAG FL_RONLY" or "0") - maybe that would be nice to have. 

4) In the listing of the installed modules at the top of the /C-page, imho the listing could be a bit more easier to read, if the lines with the disabled and enabled modules would be separated. Right now it looks like this:
[image]
 -> probably better would be something like this:
Installierte Module: 
WEBSERVER, IPWE, RESET, MQTT: Disabled 
Verbose DEBUG: Serial, LOGGER, VERSION_CHECK: enabled 

But in summary: Great job @dukess, thanks a lot for this really cool feature! :)

fredlcore, 2020-11-15:
As for 3 - this is an option that is fixed during compilation, no possibility to change it during run-time (sorry, I missed that in an earlier conversation).

1coderookie, 2020-11-15:
Another point:
I changed useDHCP in config.h to 'false' to start off with the fix default-IP 192.168.178.88.
When I set "Use DHCP" within the webinterface to "On" afterwards, the serial monitor of the Arduino IDE shows me this after clicking 'send data':
12:58:09.204 -> Option value: 192.168.178.5 
12:58:09.204 -> Option ID: 37 
12:58:09.204 -> Option value: 0 
12:58:09.204 -> Option ID: 38 
12:58:09.204 -> Option value: 192.168.178.20 
12:58:09.204 -> Option ID: 39 
12:58:09.204 -> Option value: User 
12:58:09.237 -> Option ID: 40 
12:58:09.237 -> Option value: Pass 
12:58:09.237 -> Option ID: 41 
12:58:09.237 -> Option value: BSB-LAN 
12:58:09.237 -> Option ID: 42 
12:58:09.237 -> Option value: MyHeater 
First, this is kinda weird up to me, cuz a) there are listed two different IPs now and b) User& Pass (securityfunctions) and BSB-LAN&MyHeater (MQTT-topics) are also shown here. If this is meant to be this way, then don't understand that..? 
Second: The adapter is still and only reachable at the IP 88. After a reboot the "Use DHCP" within the webinterface is shown as "On", but still the IP 88 is used. So that doesn't seem to work - or am I missing something here..?

1coderookie, 2020-11-15:
> As for 3 - this is an option that is fixed during compilation, no possibility to change it during run-time (sorry, I missed that in an earlier conversation).

Ah ok, thanks!

1coderookie, 2020-11-15:
..just if it's important to know: I did these tests with BSB-LAN v.2.0.3-20201114002548..

dukess, 2020-11-15:
> Enable all modules by default and then add a conditional for AVR/MEGA where the memory intensive modules are removed again using #undef.
Because currently, it's a bit confusing for new users who think they have disabled a function and forget to look all the way at the bottom to see that the value is actually activated for the Due down there.

We need to try this idea, even though it looks a bit counterintuitive. 

>One more thing: The QAA type is stated as QAA50 for me, although I have defined is as 0x53 for QAA70 in the _config.h. Is this a bug?

Most likely, this means that the variable is 0. I added checking at startup and setting with correct value.

dukess, 2020-11-15:
> As for 3 - this is an option that is fixed during compilation, no possibility to change it during run-time (sorry, I missed that in an earlier conversation).

I thought about it, but I was too lazy to implement it because changes were too big.
We can add some unused bit to the DEFAULT_FLAG (if possible) and check for its presence at each request. If it is set, then the program accepts further write permission according with variable.

dukess, 2020-11-15:
>Just to make it clear, maybe it would be nice to add "(seconds)"

I see no problem

>a) there are listed two different IPs now and

These addresses is for different services (MQTT and MAX)

>b) User& Pass (securityfunctions) and BSB-LAN&MyHeater (MQTT-topics) are also shown here. 

I'm thinking of disabling the output of this debugging information in the future

>After a reboot the "Use DHCP" within the webinterface is shown as "On", but still the IP 88 is used. So that doesn't seem to work - or am I missing something here..?

If you set UseEEPROM to 1 in config and set "Read from EEPROM" to "on" and run /CS, then it is problem.

fredlcore, 2020-11-15:
Ok, it's getting a bit cluttered here, that's why I was numbering my points in the beginning ;)...

As for disabling modules for the Mega, I agree, it is a bit counter-intuitive. Maybe we should just do it the way it was before and enable/disable everything in the line where it is explained (and generally enable all/most definements that won't hurt other functionalities) and then make recommendations at the end for Mega users?

I don't think we need to mask User/Pass on the serial monitor. Whoever has control to that, can also control BSB-LAN. So better save that memory and energy for other things ;). But just my opinion...

And no, we don't need to change DEFAULT_FLAG at run-time. Since this is a function where people significantly change access to the heater, I'd like to keep it as a compile-time configuration.

fredlcore, 2020-11-15:
As for saving to EEPROM: I understood only now that you have to call /CS so that it is actually saved to EEPROM - is that correct? So submitting the form only saves the values into the run-time variables (because they are persistent until reboot) and /CS commits them to EEPROM, correct? If so, shouldn't we just save them to EEPROM once the form is submitted? Or how would people know about /CS?

1coderookie, 2020-11-15:
> > Just to make it clear, maybe it would be nice to add "(seconds)"
> 
> I see no problem

No, it's not a problem - I just think that it could be helpful for some users if there would be the note that the unit is 'seconds'.

dukess, 2020-11-15:
>now that you have to call /CS so that it is actually saved to EEPROM - is that correct?

Yep.

> If so, shouldn't we just save them to EEPROM once the form is submitted? 

This is exactly what I was talking about yesterday(?) asked (about automatic save to EEPROM and reboot). So here is no problem to uncomment code (strings 7208-7215).

>No, it's not a problem - 

Sorry. I mean "no problem to add "seconds".

fredlcore, 2020-11-15:
Oh, sorry, then I didn't understand it yesterday - so yes, please uncomment those lines. As long as we have the useEEPROM = 0 fallback, there is no harm in writing to the EEPROM right away. If we do so, do we still need the /CS URL command?

dukess, 2020-11-15:
> do we still need the /CS URL command?
When we changing parameters through URL commands (/L, /D, etc) only.

fredlcore, 2020-11-15:
Ah, ok, got it!

dukess, 2020-11-16:
Propose to remove these strings from config page (because info displayed below):

Bus type
MENU_TEXT_BUS: BSB (66, 0) MENU_TEXT_BRW
Monitor mode
MENU_TEXT_MMD: 0
Verbosity level
MENU_TEXT_VBL: 1
One wire
MENU_TEXT_OWP: 3, MENU_TEXT_SNS: 10
DHT22
MENU_TEXT_DHP: 2, 3
MAC
MENU_TEXT_MAC: 00:80:41:19:69:90
Averages
MENU_TEXT_AVT:
8700 - STR8700_TEXT: 20050
8326 - STR8326_TEXT: 20051

Interval
MENU_TEXT_LGP 3600 MENU_TEXT_SEC: MENU_TEXT_YES

Log list
8700 - STR8700_TEXT
8314 - STR8314_TEXT
8830 - STR8830_TEXT
8740 - STR8740_TEXT
MENU_TEXT_BDT
MENU_TEXT_BUT: MENU_TEXT_YES
MENU_TEXT_LBO: MENU_TEXT_YES

Leave this info on config page:
Version
MENU_TEXT_VER: 2.0.3-20201114002548
free RAM
MENU_TEXT_RAM: 81271 Bytes
Uptime
MENU_TEXT_UPT: 112953614
Protected pins
MENU_TEXT_EXP: 0 1 3 10 11 12 13 18 19 20 21 22 23 50 51 52 53 62 63 64 65 66 67 68 69

Compiled modules
MENU_TEXT_MOD:
WEBSERVER, IPWE, RESET, MQTT: JSON, Verbose DEBUG: Telnet, LOGGER, VERSION_CHECK: enabled

Free space: 14806 MB
free clusters: 473805
freeClusterCount() call time: 2 microseconds

fredlcore, 2020-11-16:
You can read my mind :) - I was just about to suggest the same thing. Yes, it would be great (and look much better) if we remove all the text-only information from the configuration page which is already printed in the form. I think the protected can go, too. I think hardly ever uses these, but if you think it's better to have them here, then that's fine with me, too.
As for the difference between free space and free clusters, I think one of the two would be enough? And is the freeClusterCount call time relevant for any debugging?

dukess, 2020-11-16:
I think free space is enough. freeClusterCount call time not need now, but i think leave it commented in code. List of protected pins is helpful for One Wire and DHT sensors.

fredlcore, 2020-11-16:
Agree!

dukess, 2020-11-16:
Did two variant config info: with and without WEBCONFIG.

dukess, 2020-11-16:
Enough changes for today.
Now testing and fix bugs.

fredlcore, 2020-11-16:
Great, so Mega users who cannot use web-config can still get the info. However, web-config display should be default?

dukess, 2020-11-16:
Yes. For Due it compiled by default.

fredlcore, 2020-11-16:
In the FHEM forum there was a user mentioning just now that /D0 no longer clears the log file but shows the log file in plain text instead. I tried this with my setup and it seems to be the case. Or has this to do with changed URL commands?

fredlcore, 2020-11-16:
I checked the source code just now and the new commands /DD, /DJ, /DD0 and /DJ0 work as expected. /D followed by any other character dumps datalog.txt. Should we keep it that way and force people to distinctly say which file they want to remove? Or remove datalog.txt with /D0? Or both files? Open question for me...

1coderookie, 2020-11-16:
Are both commands /DD0 and /DJ0 also creating a new csv-header, as /D0 did? We suggested to use that for the first time of usage of a microSD card also.
I would suggest to give the opportunity to delete both logfiles and create a new csv-header at the same time with that one command /D0.

dukess, 2020-11-16:
I remove /D0, sorry. Reason: here is now two file: datalog.txt for program value logs and journal.txt for telegram dumps.
So we shoud to use /DD0 and /DJ0 instead /D0. /DD and /DJ instead /D

fredlcore, 2020-11-16:
The question is whether there are situations where you only want to remove one file and not the other. At least in my experience, I always remove both of them. For that purpose /D0 would be helpful because any external program would only have to issue one http request and not two...

dukess, 2020-11-16:
Clear. Return /D0 right now.

fredlcore, 2020-11-16:
Great, thanks a lot!

One more thing: I think we should erase the EEPROM the first time the board boots up. Otherwise it's filled with 0xFF which is especially confusing for PPS users. I think it's easy to detect if we just check for a few consecutive EEPROM addresses for 0xFF.
The other thing is that the current erasing of EEPROM takes about a minute or so. For the Mega, this is what it is, but for the Due, we use the I2C_EEPROM library which offers the pageWrite() function where 4096 bits can be written in one go. Or at least the fastBlockWrite function, which is not limited to the page size of a given EEPROM, so it's maybe more compatible. What do you think?

fredlcore, 2020-11-16:
I have added the fastWriteBlock() function for Due and it's done in an instant :). As for the initial clearing of the EEPROM, I'm not quite sure where this should be put as there are many EEPROM-related functions in the setup function. For the checking for 0xFF, the I2C_EEPROM library's fastReadBlock function could be used, if more than just a few locations would need to be checked for 0xFF.

dukess, 2020-11-16:
> I think we should erase the EEPROM the first time the board boots up. Otherwise it's filled with 0xFF which is especially confusing for PPS users. 

Don't worry: PPS settings initialized in setup with this code
```
printToDebug(PSTR("PPS settings:\r\n"));
  for (int i=PPS_TWS;i<=PPS_BRS;i++) {
    if(pps_values[i] == 0xFFFF) pps_values[i] = 0;
    if (pps_values[i] > 0 && pps_values[i]< 0xFFFF && i != PPS_RTI) {
      printFmtToDebug(PSTR("Slot %d, value: %u\r\n"), i, pps_values[i]);
    }
  }
//Fallback init with predefined room unit type
if (pps_values[PPS_QTP] == 0) pps_values[PPS_QTP] = QAA_TYPE;
```
So pps_values will be set to 0 on the first read of the EEPROM

fredlcore, 2020-11-16:
Ah, ok, great!

fredlcore, 2020-11-17:
Compiling (on Mega at least) fails if LOGGER definement is not activated:
```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:708:19: error: variable or field 'transmitFile' declared void
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:708:19: error: 'File' was not declared in this scope
```

dukess, 2020-11-17:
but here is not transmitFile declaration

fredlcore, 2020-11-17:
Yes, the line numbers are somehow wrong, don't know why - does compiling work for you when you select the Mega?

dukess, 2020-11-17:
Yep. I tried  to disable LOGGER, WEBSERVER and both.

fredlcore, 2020-11-17:
Hm, I checked again:
On the Mega, it fails as soon as WEBSERVER is disabled. LOGGER does not matter.
On the Due, it fails as soon as LOGGER is disabled. WEBSERVER does not matter.
The line numbers in both cases are the same as above (and not accurate, apparently...).

fredlcore, 2020-11-17:
...and I see that it does not make sense, because all seven occasions of transmitFile are perfectly fine encapsulated into #ifdefs...

fredlcore, 2020-11-17:
Anyway, if it runs fine on other computers, maybe it's just here or a VSCode issue...

dukess, 2020-11-19:
I need a "Guinea pig" for an experiments with MQTT: I want to send values not in a single packet, but in a series of small packets. Potential advantages: you can make the number of values sent unlimited without depending on the maximum packet size. Disadvantages: large overhead.

[19 more replies]

closed 2020-11-23 by fredlcore