fredlcore/BSB-LAN #183: Please incread MQTT_MAX_PACKET_SIZE to 2048

FunkOdyssey2001, 2020-11-16:
https://github.com/fredlcore/bsb_lan/blob/3f323b269b14cca07f4605a1c7813a3ff9f51fa4/src/PubSubClient/src/PubSubClient.h#L26

I've configured 80 parameters to be send by MQTT.
The complete output needs about 1900 characters and the default value is a way too small.

FunkOdyssey2001, 2020-11-16:
Oh, it looks like the behavior has already been changed. The buffer size is now set dynamically.

https://github.com/fredlcore/bsb_lan/commit/d5467f74775037919c892c0bf84e25a8f52a26a5

fredlcore, 2020-11-16:
Thanks for checking this out - so basically you just want the size increased from 1024 to 2048? Should definitively be doable on the Due, not so sure for the Mega...

FunkOdyssey2001, 2020-11-16:
I wonder how all the other MQTT users do this? 
The problem is that I cannot specifically query/subscribe one parameter, but have to specify all parameters in the configuration. This makes the json string really long.

Please imagine that you would output the error category via MQTT. This is going to be very long.

I have used 2048 in my older Arduino Mega setup, too.

fredlcore, 2020-11-16:
Polling the whole error category makes little sense to me because I'm only interested in the current, most recent error, not the whole error history. So that's probably why most other users don't have an issue with this ;). It may have run fine on the Mega before, but all the new functions are consuming quite some RAM. @dukess: What do you think if we add this as a _config.h-only definement variable?

dukess, 2020-11-16:
@FunkOdyssey2001 
>This makes the json string really long.

I also don't like the option where all parameters are sent in one packet. But I do not know how brokers with multiple packages with the same topic name will work. If this allowed by protocol and brokers, we can split packet for few parts.

@fredlcore 
>What do you think if we add this as a _config.h-only definement variable?

Because MQTT 2.8 allowed packets dynamic resize, we can add one more option in (web-) config.

fredlcore, 2020-11-16:
I think adding it as a definement or variable just in _config.h would be sufficient. We should not overload the web-config with options that in the end almost nobody will be using. But if we have it in _config.h, people can still adjust it if necessary.
The advantage of a definement would be that it would not hurt older installations (well, now just a few days old ;) ) and people would not have to update their _config.h because we could just check if the definement is active, and if not, we use the default value.

dukess, 2020-11-19:
Great!
I make little change: https://github.com/dukess/bsb_lan/commit/36ecd8060de0c7414398540475ddff9ef3107e72
Can you take bsb_lan from my fork and try MQTT JSON?
https://github.com/dukess/bsb_lan

dukess, 2020-11-19:
Yes, JSON was added in 0.4x

FunkOdyssey2001, 2020-11-19:
I am getting errors while compiling:

```
BSB_lan:7290:28: error: 'resetBoard' was not declared in this scope
                 resetBoard();
                            ^

```

fredlcore, 2020-11-19:
You have to use the config from @dukess branch. The RESET Definement has been removed in Master

FunkOdyssey2001, 2020-11-19:
I know. I do not have any reset definements.

FunkOdyssey2001, 2020-11-19:
I have to remove the `#ifdef RESET` and then it works. Freetz don't have this in his code: https://github.com/fredlcore/bsb_lan/blob/master/BSB_lan.ino#L5516

```
#ifdef RESET
void resetBoard(){
...
}
#endif
```

fredlcore, 2020-11-19:
That's what I mean, you have to use @dukess config where RESET is still defined. You seem to have used the current _config.h where #define RESET has been removed because it is no longer necessary in the master branch, but it is necessary in @dukess' snapshot.

dukess, 2020-11-19:
Frederik, i see you returned code with 3 cmdtbls. Did it work on Mega?

dukess, 2020-11-19:
Pull last changes.

fredlcore, 2020-11-19:
I have only added the variable in _defs.h because the code in .ino was still making references to cmdtbl3. So I thought you only removed cmdtbl3 from _defs.h, but not from .ino, so I added the empty variable back into _defs.h, so that it compiles. 
Or did I forget to merge a pull request where you removed cmdtbl3 from the functions in the main code?

dukess, 2020-11-19:
It looks like there was an edit conflict. I removed support for 3 tables here.
https://github.com/fredlcore/bsb_lan/commit/8c25c8049a082f9a9c74e114012fcfcde18aa33e#diff-1bd647b02bf0ccbe434d8cc3d6596e22497c78ff95f4317eaeae9bef0e7cef49

fredlcore, 2020-11-19:
You're right - is there a way to fix this still now?

dukess, 2020-11-19:
I think it can wait little bit. I suggest testing MQTT first.

fredlcore, 2020-11-19:
Ok, your call! I think it would probably be enough just to copy over the block with the calc_XXX functions from your patch because nothing else was changed in that part, I guess...

FunkOdyssey2001, 2020-11-19:
The current code only runs for a very short time. Then the Arduino freezes.
In addition, I cannot do anything with the received data in the MQTT server. New values are apparently always **appended**. The JSON is getting **longer**.

```
...
LAN->HEIZ QUR  710 Heizkreis 1 - Komfortsollwert:
dc c2 00 0b 06 3d 2d 05 8e d2 1e
HEIZ->LAN ANS  710 Heizkreis 1 - Komfortsollwert: 20.0 &deg;C
dc 80 42 0e 07 2d 3d 05 8e 00 05 00 81 6b
#710: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"710":"20.0"}}}
LAN->HEIZ QUR  712 Heizkreis 1 - Reduziertsollwert:
dc c2 00 0b 06 3d 2d 05 90 21 e1
HEIZ->LAN ANS  712 Heizkreis 1 - Reduziertsollwert: 18.0 &deg;C
dc 80 42 0e 07 2d 3d 05 90 00 04 80 9a 2f
#712: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"712":"18.0"}}}
LAN->HEIZ QUR  720 Heizkreis 1 - Kennlinie Steilheit:
dc c2 00 0b 06 3d 2d 05 f6 2d 81
HEIZ->LAN ANS  720 Heizkreis 1 - Kennlinie Steilheit: 0.56
dc 80 42 0e 07 2d 3d 05 f6 00 00 1c 6a 95
#720: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.56
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"720":"0.56"}}}
LAN->HEIZ QUR 6705 Fehler - SW Diagnosecode:
dc c2 00 0b 06 3d 05 00 99 60 5a
HEIZ->LAN ANS 6705 Fehler - SW Diagnosecode: 0
dc 80 42 0e 07 05 3d 00 99 00 00 00 ea 18
#6705: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6705":"8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found"}}}
LAN->HEIZ QUR 6710 Fehler - Reset Alarmrelais:
dc c2 00 0b 06 3d 05 05 d6 26 84
HEIZ->LAN ANS 6710 Fehler - Reset Alarmrelais: 0 - Nein
dc 80 42 0d 07 05 3d 05 d6 00 00 5c c5
#6710: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6710":"0"}}}
LAN->HEIZ QUR 6740 Fehler - Vorlauftemperatur 1 Alarm:
dc c2 00 0b 06 3d 21 06 9d d0 7e
HEIZ->LAN ANS 6740 Fehler - Vorlauftemperatur 1 Alarm: --- min
dc 80 42 0d 07 21 3d 06 9d 01 78 d5 42
#6740: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6740":"---"}}}
LAN->HEIZ QUR 6741 Fehler - Vorlauftemperatur 2 Alarm:
dc c2 00 0b 06 3d 22 06 9d 89 2e
HEIZ->LAN ANS 6741 Fehler - Vorlauftemperatur 2 Alarm: --- min
dc 80 42 0d 07 22 3d 06 9d 01 78 1b a2
#6741: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min--- min
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6741":"---"}}}
LAN->HEIZ QUR 6743 Fehler - Kesseltemperatur Alarm:
dc c2 00 0b 06 3d 11 05 de 38 2f
HEIZ->LAN ANS 6743 Fehler - Kesseltemperatur Alarm: --- min
dc 80 42 0d 07 11 3d 05 de 01 78 25 ef
#6743: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min--- min--- min
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6743":"---"}}}
LAN->HEIZ QUR 6745 Fehler - Trinkwasserladung Alarm:
dc c2 00 0b 06 05 25 0b 10 93 ed
HEIZ->LAN ANS 6745 Fehler - Trinkwasserladung Alarm: --- h
dc 80 42 0d 07 25 05 0b 10 01 08 e3 3a
#6745: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min--- min--- min--- h
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6745":"---"}}}
LAN->HEIZ QUR 6800 Fehler - Historie 1 Datum/Zeit:
dc c2 00 0b 06 3d 05 06 d3 23 72
HEIZ->LAN ANS 6800 Fehler - Historie 1 Datum/Zeit: 05.03.2020 06:20:00
dc 80 42 14 07 05 3d 06 d3 00 78 03 05 00 06 14 00 04 f5 9f
#6800: 8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min--- min--- min--- h05.03.2020 06:20:00
Output topic: BSB/json
Payload Output : {"BSB_LAN_TEST":{"status":{"6800":"8742 Diagnose Verbraucher - Raumtemperatur 1 Modell: 14.4 &deg;C1 - Automatik1 - Automatik20.0 &deg;C18.0 &deg;C0.560 - not found0 - not found0 - Nein--- min--- min--- min--- h05.03.20}
...
```

dukess, 2020-11-19:
Thank you for logs!

FunkOdyssey2001, 2020-11-19:
Could it be that you have changed the behavior for MQTT text output, too?
With your code, I cannot revert the mqtt_mode from 2 to 1 with the old behaviour.

dukess, 2020-11-19:
Here was bug which introduced when "leading spaces via MQTT" was fixed.

dukess, 2020-11-19:
>With your code, I cannot revert the mqtt_mode from 2 to 1 with the old behaviour. 

You can't switch to plain text in web config?

FunkOdyssey2001, 2020-11-19:
I am not using the web config, because I always have problems with loading the config from the EPROMs after re-flashing. 

I changed the mqtt mode in the config file and the output looks crazy. 
But first you can ignore this. I am still able to revert to the master branch.

dukess, 2020-11-19:
Most likely problem was caused buffer overflow.

dukess, 2020-11-20:
@fredlcore @FunkOdyssey2001 BTW: for ENUMs MQTT send value and description (1 - Automatik). Maybe, for beauty, it should send description only? (Automatik). Or somebody can parsing string for value extraction?

fredlcore, 2020-11-20:
We only introduced the description upon @FunkOdyssey2001's request, IIRC. The numerical value should stay because that one is easier to display in a graph, for example when monitoring the burner status etc.

dukess, 2020-11-20:
Ok, clear!

FunkOdyssey2001, 2020-11-20:
It doesn't really matter now, but couldn't the JSON be structured more logically in such cases?

`bsb-lan/bsb {"parameterid": "8005", "parameterstring": "Status Kessel", "value": 25, "string": "Aus"}`

Somehow I like this structure. It would just make the json mapping (and json2nameValue in FHEM) more complex on the receiving end.

FunkOdyssey2001, 2020-11-20:
By the way: Your current code is working fine in JSON mode:

```
Output topic: BSB/json
Payload Output : {"BSB_LAN":{"status":{"700":"1 - Automatik"}}}
LAN->HEIZ QUR  710 Heizkreis 1 - Komfortsollwert:
dc c2 00 0b 06 3d 2d 05 8e d2 1e
HEIZ->LAN ANS  710 Heizkreis 1 - Komfortsollwert: 20.0 &deg;C
dc 80 42 0e 07 2d 3d 05 8e 00 05 00 81 6b
#710: 20.0 &deg;C
Output topic: BSB/json
Payload Output : {"BSB_LAN":{"status":{"710":"20.0"}}}
LAN->HEIZ QUR  712 Heizkreis 1 - Reduziertsollwert:
dc c2 00 0b 06 3d 2d 05 90 21 e1
HEIZ->LAN ANS  712 Heizkreis 1 - Reduziertsollwert: 18.0 &deg;C
dc 80 42 0e 07 2d 3d 05 90 00 04 80 9a 2f
#712: 18.0 &deg;C
Output topic: BSB/json
Payload Output : {"BSB_LAN":{"status":{"712":"18.0"}}}
LAN->HEIZ QUR  720 Heizkreis 1 - Kennlinie Steilheit:
dc c2 00 0b 06 3d 2d 05 f6 2d 81
HEIZ->LAN ANS  720 Heizkreis 1 - Kennlinie Steilheit: 0.56
dc 80 42 0e 07 2d 3d 05 f6 00 00 1c 6a 95
#720: 0.56
Output topic: BSB/json
Payload Output : {"BSB_LAN":{"status":{"720":"0.56"}}}
LAN->HEIZ QUR 6705 Fehler - SW Diagnosecode:
dc c2 00 0b 06 3d 05 00 99 60 5a
HEIZ->LAN ANS 6705 Fehler - SW Diagnosecode: 0
dc 80 42 0e 07 05 3d 00 99 00 00 00 ea 18
#6705: 0 - not found
Output topic: BSB/json
Payload Output : {"BSB_LAN":{"status":{"6705":"0 - not found"}}}
LAN->HEIZ QUR 6710 Fehler - Reset Alarmrelais:
dc c2 00 0b 06 3d 05 05 d6 26 84
HEIZ->LAN ANS 6710 Fehler - Reset Alarmrelais: 0 - Nein
dc 80 42 0d 07 05 3d 05 d6 00 00 5c c5
#6710: 0 - Nein
```

dukess, 2020-11-21:
> This could be nicer

You mean to remove empty JSON strings?

> bsb-lan/bsb {"parameterid": "8005", "parameterstring": "Status Kessel", "value": 25, "string": "Aus"}

Nice idea but need discuss with @fredlcore - this way break compatibility with old version.... However v 2.0 already broke compatibility in many functions.

fredlcore, 2020-11-21:
Sorry, I may not really get the benefit, but if you (@FunkOdyssey2001) say yourself that 
> It would just make the json mapping (and json2nameValue in FHEM) more complex on the receiving end.
then that in itself is not convincing to me if it means breaking backward compatibility (which we could do, if it was really an improvement).
For the moment, I get the impression that we're changing the code for very special use cases that so far have only affected you. I would like to prioritize the bread and butter issues that still affect version 2.0.
But since @dukess is doing all the heavy lifting at the moment, I would leave it up to him to decide.

FunkOdyssey2001, 2020-11-21:
It was just an idea. 
I am happy with the MQTT text mode. You don't have to change that for me!

I'm not really a MQTT pro either, but the current Json structure still looks kind of strange (status-prefix).
But there are experienced MQTT professionals somewhere who can judge that. That is not me. 

Please don't change this "for me". It was just a suggestion.

FunkOdyssey2001, 2020-11-21:
> You mean to remove empty JSON strings?

Yes

dukess, 2020-11-21:
Gentlemans, we can add "mqtt_mode = 3" in config and do "JSON with format" output. 
If @fredlcore will accept "one program value per MQTT JSON message" then we just need to add 10-15 strings of code.

>  remove empty JSON strings

@FunkOdyssey2001 what is better: send to broker string like "---" instead empty string or do not send nothing?
If "rich MQTT JSON" will be accepted we can send error code to broker.

fredlcore, 2020-11-21:
Fine with me!

FunkOdyssey2001, 2020-11-22:
I would like it better to send strings like "---".

dukess, 2020-11-23:
Done!

dukess, 2020-11-23:
@fredlcore Do you have any preferences for fields naming?
Now is {"parameterid": one_of_logvalues, "parametername": "name", "value": "query_result", "desc": "enum value description", "unit": "unit of measurement", "error", error_code}
@FunkOdyssey2001 quite rightly points out is not the best choice of names for the fields.

We can set names according to Home Assistant or any other software.

fredlcore, 2020-11-23:
I'm not sure that I understand why there is a need to rename the fields. But if there is a need, what would be a better solution? To me, matching "parametername" to "name" sounds quite right, and so does "desc" and "enum value description" as well as the others. Do you have any suggestions, @dukess and @FunkOdyssey2001 ?

We need to bear in mind that any change here will most likely result in all MQTT/JSON users having a broken system if they forget to read the Changelog (which most of them do ;) ). So if we do this step, it should really be a benefit in the long run, and not just "looking better".

FunkOdyssey2001, 2020-11-23:
The news names are just for the new MQTT mode 3.

fredlcore, 2020-11-23:
Ok, but does it make sense to have two different naming schemes? I would like to avoid that, also for @1coderookie 's sake who will have to have two different chapters/sections in the manual to maintain. As I said, if there is a real benefit, I'm fine with it, but so far I simply have not understood the problem, I guess...

dukess, 2020-11-23:
In this case we can use {"id": one_of_logvalues, "name": "name", "value": "query_result", "desc": "enum value description", "unit": "unit of measurement", "error", error_code} as in JSON commands.

fredlcore, 2020-11-23:
Yes, that would be better. But could somebody explain to me the difference between the current way MQTT/JSON is handled and what the new mode 3 would bring? I'm sorry that I still don't understand it, but I'm neither using MQTT nor JSON for that matter...

1coderookie, 2020-11-23:
Sorry for jumping into this discussion guys, but I just stumbled across the note about the latest patch in the BSB_lan.ino: "MQTT: If query() get any error then MQTT send to broker "---" as value." and found the source of that within this discussion here.

I remember that within certain parameters the heater sided display unit shows a value like "---", if a parameter is not set. In these cases BSB-LAN shows the same output "---" as it is a 'regular' value for these parameters. 
I just did a quick check - see screenshot of 632&633 as an example.
[image]

I'm not using MQTT either, but I'm thinking that it might not be the best solution to use the same output as an error message.
If you guys don't see a problem here, that's fine with we. Just wanted to mention it if you are maybe not aware of this possible 'regular' output within some parameters..

fredlcore, 2020-11-23:
Good point!

FunkOdyssey2001, 2020-11-23:
I think that we should use the same ouput as on the webserver. 
We cannot ignore these values and should not send nothing. Or do you mean an empty string or a null value?

dukess, 2020-11-23:
Other way: "20000":"0" when ok and "20000":"" (empty value) when error.

dukess, 2020-11-23:
Oh, i found what a problem was here: because (log_parameters[i] > 0) stayed on old place it cut content but leave wrapper.

1coderookie, 2020-11-23:
> I think that we should use the same ouput as on the webserver.

Exactly - but in my understanding the '---' as a regular 'parameter-not-set-message' is something different then an 'error', which is caused by "If query() get any error". 
But maybe I didn't get it right what you guys meant, so it's up to you.

> We cannot ignore these values and should not send nothing. Or do you mean an empty string or a null value?

No, I meant something different. Again, just to make it clear: 
The output "---" is a 'regular' output within certain parameters, if that parameter isn't set. Up to me that's something different then an 'error'-message which is caused by "If query() get any error". 
But if I'm wrong in this and the regular '---' (= parameter not set) should be handled as an 'error'-message within MQTT, then thats fine with we. 
But IF it should be handled different (what imo would make more sense), then something else then '---' would be more sufficient (up to me); e.g.: 'xxx' oder 'error' or whatever..

FunkOdyssey2001, 2020-12-08:
Can we close this issue?
It looks like there are no problems anymore with MQTT_MAX_PACKET_SIZE.

fredlcore, 2020-12-08:
Great to hear that there aren't any issues anymore. The reasons this is still open was that @1coderookie is right that we cannot use "---" for an error or empty value, because "---" stands for "disabled". @dukess, is that the way you implemented it?

dukess, 2020-12-08:
If query() return error then mqtt client will send for plain text empty 
string as value,  for json - "", for rich json - "" and error code.

fredlcore, 2020-12-08:
Ok, great, that should be fine, thanks! I'll close this now then.

closed 2020-12-08 by fredlcore