fredlcore/BSB-LAN #184: BSB stops sending MQTT output after a few days

FunkOdyssey2001, 2020-11-16:
I have already noticed several times that after a few days, BSB no longer sends data via MQTT.

If I execute a reset via / NE or / N, the MQTT-output appears again.

I've already tried different MQTT servers. The behavior is identical.

The web access works fine.

I am currently using BSB-LAN Web, Version 1.1.50-20201104231010 on a Arduino Due.

fredlcore, 2020-11-16:
This information is too little to work on. First of all, version 2.0 is only out for a few days, so it's probably not possible that this has happened repeatedly (after a few days) under version 2.0. Other than that, does BSB-LAN still send out MQTT messages? Anything relevant to this in the serial monitor output? Without further information, there is nothing much we can do.

FunkOdyssey2001, 2020-11-16:
I will of course do more tests. I know that with this information it is not that easy. I was hoping that you could give me tips on how to find a solution. That's how you've done it.

I just don't know how I can keep the serial monitor running for several days. Do you have any idea?

By the way: I never said that I am currently using v2.0 :-)

fredlcore, 2020-11-16:
> By the way: I never said that I am currently using v2.0 :-)

Ok, that's an important information, because many things have changed under the hood, including updated libraries. So please check if this still applies for v2.0.
As for the serial monitor: I have hooked up my Due to my Synology NAS. Then I log on to the NAS via ssh and use miniterm on the serial port that the USB of the Due creates. Then I just leave the terminal session open for as long as I want.

FunkOdyssey2001, 2020-11-28:
No problems anymore since more than a week. Thank you.

closed 2020-11-28 by FunkOdyssey2001

fredlcore, 2020-11-28:
Glad to hear!