fredlcore/BSB-LAN #188: Check for new parameters is broken

fredlcore, 2020-11-21:
We had a similar issue a while ago, but now the "Check for new parameters" function (/Q) is broken again. This is the current output on my machine, which I ran after I received a similar output from a user:
```
Version: 2.0.14-20201117142051
 Scanne nach Geräten...
 Geräteadresse gefunden: 0
 Geräteadresse gefunden: 3
 Geräteadresse gefunden: 10
 
 Teste Geräteadresse 0...
 Gerätefamilie: 163 
 Gerätevariante: 5 
 Geräte-Identifikation: 1635LMS15.001A100LMS15.001A100 
 Software-Version: 1635LMS15.001A100LMS15.001A1004.24.2 
 Entwicklungs-Index: 00006B - decoding error 
 Objektverzeichnis-Version: 00006B - decoding error1.01.0 
 Bootloader-Version: (parameter not supported) 
 EEPROM-Version: (parameter not supported)------ 
 Konfiguration - Info 2 OEM: (parameter not supported)------00 
 Zugangscode Inbetriebnahme?: (parameter not supported)------0000 
 Zugangscode Fachmannebene ?: (parameter not supported)------000000 
 Zugangscode OEM?: (parameter not supported)------00000000 
 Zugangscode OEM2?: (parameter not supported)------0000000000 
 Bisher unbekannte Geräteabfrage: (parameter not supported)------00000000002020 
 Hersteller-ID (letzten vier Bytes): (parameter not supported)------00000000002020180190038180190038 
 Bisher unbekannte Geräteabfrage: 04011200C8 - unknown type 
 Außentemperatur (10003): 04011200C8 - unknown type8.8 °C8.8 °C 
 Außentemperatur (10004): 04011200C8 - unknown type8.8 °C8.8 °C8.8 °C8.8 °C 
 
 6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
 163;5;LMS15.001A100;4.2;;1.0;---;0;0;0;0;0;20;180190038;04011200C8;
 
 
 Starte Test...
 6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
 163;5;LMS15.001A100;4.2;;1.0;---;0;0;0;0;0;20;180190038;04011200C8;
 
 
 Starte Test...
 00C8 - unknown type8.8 °C8.8 °C8.8 °C8.8 °C 
 
 decoding error1.01.0 
 Bootloader-Version: (parameter not supported) 
 EEPROM-Version: (parameter not supported)------ 
 Konfiguration - Info 2 OEM: (parameter not supported)------00 
 Zugangscode Inbetriebnahme?: (parameter not supported)------0000 
 Zugangscode Fachmannebene ?: (parameter not supported)------000000 
 Zugangscode OEM?: (parameter not supported)------00000000 
 Zugangscode OEM2?: (parameter not supported)------0000000000 
 Bisher unbekannte Geräteabfrage: (parameter not supported)------00000000002020 
 Hersteller-ID (letzten vier Bytes): (parameter not supported)------00000000002020180190038180190038ter>
```
It seems that the returned parameter value is not overwritten and new values are just appended at the end - until a "parameter not supported" comes up. This seems to clear the variable and the process begins again...
The above user's version is from 17 November, but mine from today has the same/similar output. Could you please check this, @dukess?

fredlcore, 2020-11-21:
Also, my Due hangs when running /Q after parameter 7708, I assume a buffer overflow due to too the long appended values. Here are the last few lines in serial monitor:
```
53d04a2                                                                         
LAN->HEIZ QUR 7708 Ein-/Ausgangstest - Modulationssignal QX3:                   
dc c2 00 0b 06 3d 05 04 a2 2b a6                                                
HEIZ->LAN ANS 7708 Ein-/Ausgangstest - Modulationssignal QX3: 0 &#037;          
dc 80 42 0d 07 05 3d 04 a2 00 00 2e b9                                          
LAN->HEIZ QUR 7708 Ein-/Ausgangstest - Modulationssignal QX3:                   
dc c2 00 0b 06 3d 05 04 a2 2b a6                                                
HEIZ->LAN ANS 7708 Ein-/Ausgangstest - Modulationssignal QX3: 0 &#037;          
dc 80 42 0d 07 05 3d 04 a2 00 00 2e b9                                          
#7708:  (parameter not supported)25.0325.101. 06:00 - 22:00 2. --:-- - --:-- 3.;
```

dukess, 2020-11-22:
Can't reproduce error. Can you add some debug outputs in build_pvalstr()?

fredlcore, 2020-11-22:
That did the trick - I thought I had synced before several times, but apparently not ;). Sorry for the confusion!

closed 2020-11-22 by fredlcore