fredlcore/BSB-LAN #193: Writing of type VT_SECONDS_WORD doesn't work

Magublafix, 2020-11-27:
Hi Guys,

First of all thank you for this great project! ;)

I tried to change the parameter 836, but received an error that the parameter is not writeable. I looked up the code and it seems that there is no routine for parameters of type VT_SECONDS_WORD in the set method.
Would it be possible to add this feature? I would try it on my own but kinda afraid of destroying something ;)

regards,
Gerald

Magublafix, 2020-11-28:
Ok i was to curious and had to try it ;) it didn't work with the VT_MINUTES_WORD block as the seconds parameter is a float. But with those lines it works just fine.

`
case VT_SECONDS_WORD:
      {
      uint16_t t=atof(val);
      param[0]=0x01;
      param[1]=(t >> 8);
      param[2]= t & 0xff;
      param_len=3;
      }
      break;
`

regards,
Gerald

dukess, 2020-11-28:
Thank you for report!
You can take fix ( https://github.com/dukess/bsb_lan/commit/5e096f9cd6a22aa28b1269fe8b3d548fecb6a084 ) from my fork and trying it when you want. Or you can wait for new version here.

dukess, 2020-11-28:
Tried it. Not enough.

setting line: 746 val: 06 00 0a
HEIZ->LAN NACK  746 ENUM_CAT_0b_TEXT - STR746_TEXT: WORD len error len!=3: 06
set failed NACK
Setting parameter 746 to 10 with type 1
setting line: 746 val: 06 00 0a
HEIZ->LAN NACK  746 ENUM_CAT_0b_TEXT - STR746_TEXT: WORD len error len!=3: 06
set failed NACK
Setting parameter 746 to 10 with type 1

Magublafix, 2020-11-28:
@dukess 
Yeah the first try didn't work, because the case block uses atoi not atof. 
You have to use those lines, I think I didn't find an existing block like this:
`
case VT_SECONDS_WORD: 
{ 
uint16_t t=atof(val); 
param[0]=0x01; 
param[1]=(t >> 8);
param[2]= t & 0xff; param_len=3; } 
break;
`

dukess, 2020-11-28:
we need @fredlcore .

dukess, 2020-11-28:
Bus data for prog 834. Looks like it not like 0x06 and 0x05 flags.

RGT1->HEIZ QUR  834 ENUM_CAT_0b_TEXT - STR834_TEXT:
dc 86 00 0b 06 3d 21 06 5a 80 68
HEIZ->RGT1 ANS  834 ENUM_CAT_0b_TEXT - STR834_TEXT: 30 UNIT_SEC_TEXT
dc 80 06 0e 07 21 3d 06 5a 00 00 1e 8c 05
RGT1->HEIZ QUR  834 ENUM_CAT_0b_TEXT - STR834_TEXT:
dc 86 00 0b 06 3d 21 06 5a 80 68
HEIZ->RGT1 ANS  834 ENUM_CAT_0b_TEXT - STR834_TEXT: 30 UNIT_SEC_TEXT
dc 80 06 0e 07 21 3d 06 5a 00 00 1e 8c 05
RGT1->HEIZ QUR  833 ENUM_CAT_0b_TEXT - STR833_TEXT:
dc 86 00 0b 06 3d 21 06 5c e0 ae
HEIZ->RGT1 ANS  833 ENUM_CAT_0b_TEXT - STR833_TEXT: 2.0 UNIT_DEG_TEXT
dc 80 06 0e 07 21 3d 06 5c 00 00 80 c9 eb
RGT1->HEIZ QUR  834 ENUM_CAT_0b_TEXT - STR834_TEXT:
dc 86 00 0b 06 3d 21 06 5a 80 68
HEIZ->RGT1 ANS  834 ENUM_CAT_0b_TEXT - STR834_TEXT: 30 UNIT_SEC_TEXT
dc 80 06 0e 07 21 3d 06 5a 00 00 1e 8c 05
RGT1->HEIZ SET  834 ENUM_CAT_0b_TEXT - STR834_TEXT: --- UNIT_SEC_TEXT
dc 86 00 0e 03 3d 21 06 5a 01 00 20 95 f9
HEIZ->RGT1 ACK  834 ENUM_CAT_0b_TEXT - STR834_TEXT:

So it was true. Fix it.

setting line: 834 val: 01 00 1f
LAN->HEIZ SET  834 ENUM_CAT_0b_TEXT - STR834_TEXT: --- UNIT_SEC_TEXT
dc c2 00 0e 03 3d 21 06 5a 01 00 1f 47 d7
HEIZ->LAN ACK  834 ENUM_CAT_0b_TEXT - STR834_TEXT:
dc 80 42 0b 04 21 3d 06 5a 9b b8
Setting parameter 834 to 31 with type 1
setting line: 834 val: 01 00 1f
LAN->HEIZ SET  834 ENUM_CAT_0b_TEXT - STR834_TEXT: --- UNIT_SEC_TEXT
dc c2 00 0e 03 3d 21 06 5a 01 00 1f 47 d7
HEIZ->LAN ACK  834 ENUM_CAT_0b_TEXT - STR834_TEXT:
dc 80 42 0b 04 21 3d 06 5a 9b b8
Setting parameter 834 to 31 with type 1

dukess, 2020-11-28:
Fix in my fork: https://github.com/dukess/bsb_lan/commit/05b9c3941c82a621e69d0f585243fbe647d3ef4c
Fix in pull request: https://github.com/fredlcore/bsb_lan/pull/194

fredlcore, 2020-11-28:
Sorry for getting in here late, but does your fix work now, @dukess ?
Otherwise, I would need a serial monitor log from @Magublafix when you set the parameter on the heater itself, so we can exactly identify how certain parts of the telegram have to be set. Sometimes assuming/copying the blocks from other data types works, but not always.

Magublafix, 2020-11-28:
I'm still on an mega 2560 so i didn't try @dukess version.
But I can attach a serial monitor output when I change the value with the code change I mentioned. Is that what you need?

[austriaemail_Set836.txt](https://github.com/fredlcore/bsb_lan/files/5610870/austriaemail_Set836.txt)

dukess, 2020-11-28:
at least it work with my boiler.

dukess, 2020-11-28:
@Magublafix
/S836=300.00
set ProgNr 836 = 300.00
setting line: 836 val: 01 01 2C 
LAN->HEIZ SET  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: DC--- s
DC C2 00 0E 03 3D 21 06 59 01 01 2C E9 0A 
HEIZ->LAN ACK  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: 
DC 80 42 0B 04 21 3D 06 59 AB DB 
LAN->HEIZ QUR  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: 
DC C2 00 0B 06 3D 21 06 59 49 B6 
HEIZ->LAN ANS  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: DC300 s
DC 80 42 0E 07 21 3D 06 59 00 01 2C ED D6 
#836: 300 s 

I see "setting line: 836 val: 01 01 2C" so my patch must work with your system.

Magublafix, 2020-11-28:
@dukess well as i said this output was **NOT** recorded with your version of the code. Instead I used the workaround which I suggested above.  
I can't try your version because I still have no arduino due and the program is already too huge to fit on the mega 2560.

dukess, 2020-11-28:
It is not a problem.
But i recommend place VT_SECONDS_WORD between 
```
    // ---------------------------------------------
    // 16-bit unsigned integer representation
```
    case VT_SECONDS_WORD:
```
    // Temperature values
    case VT_TEMP_WORD:
    case VT_CELMIN:
```
because VT_SECONDS_WORD is integer.

1coderookie, 2020-11-28:
> I tried to change the parameter 836, but received an error that the parameter is not writeable. 

Guys, I might be wrong, but I think it's just not 'writeable' because **it's an OEM parameter**:

{0x213D0659,  CAT_HK1,              VT_SECONDS_WORD,  836,   STR836,   0,                    NULL,         **FL_OEM**, DEV_ALL}, // [ s ] - Mischer Nachstellzeit Tn

Therefore it's not writeable, even if BSB-LAN got the regular write access granted.
To  make it writeable, see the note in the beginning of the defs.h about OEM parameters:

#define FL_OEM      5    // Known OEM parameters are set to read-only by default. If you want to have general write-access (not recommended!) to OEM parameters, set FL_OEM to 4.

I tried this morning and I was able to set the parameter. I guess it's also possible if one changes 'FL_OEM' to '0' (or -if already implemented- FL_WRITEABLE) to allow only this one OEM parameter to be set, or even 'DEFAULT_FLAG' to make it writeable together with the other default-flagged parameters and then give general write access to BSB-LAN.
However, (@Magublafix) keep in mind that a change like this in your defs.h would have to be done again if the file gets overwritten due to a complete update of BSB-LAN.

Magublafix, 2020-11-28:
@fredlcore 
Sry should have mentioned that I already set the parameter to 0 instead of DEFAULT_FLAG like I do for all Parameters I want to write at the beginng. But as writing didn't work either I checked the serial output and found this line "Unknown type or read-only parameter". I found this output only in the default case block and added further output "DebugOutput.println(get_cmdtbl_type(i));" which  in turn gave me the index of VT_SECONDS_WORD. From there it was clear to me that the problem is a missing case statement. 

Next I tried to add the case block infront of the other WORD blocks (VT_MINUTES_WORD, ...) but this actually didn't work because the parameter is a float value and no int at least for my system. Not sure if this is valid for all parameters of this type. Therefor I copied the type and changed the atoi to an atof, afterwards I could set the param.

fredlcore, 2020-11-28:
I think you were replying to @1coderookie , and while I think it's both issues at the same time, DEFAULT_FLAG as 0 still does not allow you to write FL_OEM parameters. But I will merge @dukess fix and see how it goes!

1coderookie, 2020-11-28:
> ... DEFAULT_FLAG as 0 still does not allow you to write FL_OEM parameters...

That's why I wrote "I guess it's also possible...". 
But as I wrote before that, I tried as described in the defs.h in the beginning and set "#define FL_OEM 5" to "4" - after that, I was able to set the parameter. At least the webinterface showed me the new value and didn't tell me that there was a problem - I didn't check at the heater itself if it was successful though..

Magublafix, 2020-11-28:
@1coderookie 
hmm ok very strange. I tried it now too with a clean 0.44 download (only have an old arduino) and only changed the ip adress and set FL_OEM to 4. But still I get the same error. Nevertheless I'm happy with the solution I found so maybe it makes no sense to discuss this further ;)

[image]

`GET /S836=120.00 HTTP/1.1

/S836=120.00
set ProgNr 836 = 120.00
Unknown type or read-only parameter
GET /8412 HTTP/1.1

/8412`

dukess, 2020-11-28:
With 836 program i have same issue (OEM) but 834 is allow me to change their value (If somebody want to ensure).
836 can be changed only when QAA70 entering in "OEM mode".

fredlcore, 2020-11-28:
@1coderookie : I really wonder why you did not get the error that @Magublafix got, because the data type VT_SECONDS_WORD was not defined in the set() function until now. So even with setting FL_OEM to 4, the error should have come, as it stands for both read-only parameter as well as "unknown type". Could you check that again and maybe do a SerMo log when setting this parameter?

As for @dukess' 836 and the QAA70, I don't really get that, because QAA70 is PPS, for which this parameter is not available - and how do you enter OEM mode there? 834 is not a OEM parameter, so it is generally writeable, but it should only be able to work AFTER your recent addition of VT_SECONDS_WORD to the set() parameter, not before, right?

dukess, 2020-11-29:
@fredlcore Sorry for misrepresentation. I mean room device QAA70.

fredlcore, 2020-11-29:
But are you sure that the PPS related QAA70 has an effect on whether you can change parameter 836? That would be really strange...

1coderookie, 2020-11-29:
@dukess: Do you really mean the QAA70 room unit which is for old systems using PPS, or do you mean the QAA75 which is for the newer systems and connected via BSB?

@fredlcore: Just checked again right now with 836 and sure you're right - it doesn't work, I'm getting the error message as you said.. :( 
I'm really sorry for any confusion or inconvenience that my posting caused - I'm still trying to figure out what happened there yesterday that it didn't appeared in the first time..

fredlcore, 2020-11-29:
No worries, highlighting the FL_OEM flag was certainly helpful. And sometimes there are oddities that are hard to explain ;)...

dukess, 2020-11-29:
My room device work as BSB device.
It have selector for changing mode: for end-user, engineer, OEM. If you know password then you can switch from end-user to engineer (more allowed parameters) or OEM (much more  allowed parameters). When i select "engineer" 836 program just not shown. When i select "OEM" 836 program shown and can be changed. 

By the way, i enhance software-controlled read-only mode today.

1coderookie, 2020-11-29:
> My room device work as BSB device.
> It have selector for changing mode: for end-user, engineer, OEM.

The you are talking about the QAA75. It looks like this one shown in the manual, right? https://1coderookie.github.io/BSB-LPB-LAN_EN/chap03.html#362-qaa75--qaa78

dukess, 2020-11-29:
Yep, it is QAA75.91x

fredlcore, 2020-12-08:
To get back to the original problem (and for the time being close this issue), this is the code that now runs in 2.0:
```
     {
      if(val[0]!='\0'){
        uint16_t t=atoi(val);
        param[0]=0x01;  //enable
        param[1]=(t >> 8);
        param[2]= t & 0xff;
      }else{
        param[0]=0x00;  // disable
        param[1]=0x00;
        param[2]=0x00;
      }
      param_len=3;
      }
      break;
```
@Magublafix , could you please check if that code snippet works for your, too?

closed 2020-12-08 by fredlcore

Magublafix, 2020-12-09:
@fredlcore yep that works fine! 
Just because i'm interested ;) when is 0x01 used for enabling end when 0x06? Because when I understood it correctly the problem was not the atoi() vs. atof() operation but the enable flag is different for VT_MINUTES_WORD.

fredlcore, 2020-12-09:
We don't know. That's why we need to have a log for the SET telegram for each parameter.

Magublafix, 2020-12-09:
@fredlcore ah ok, thank you for the quick answer! :)
btw: I already ordered the arduino due ;)

fredlcore, 2020-12-09:
@Magublafix: Could you send us a serial monitor log when setting parameter 836 on your heating system? Then we know exactly how the telegram should be formed and can adjust it accordingly. Thanks!

Magublafix, 2020-12-09:
@fredlcore sure

`setting line: 836 val: 01 00 78 
LAN->HEIZ SET  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: DC--- s
DC C2 00 0E 03 3D 21 06 59 01 00 78 C0 4A 
HEIZ->LAN ACK  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: 
DC 80 42 0B 04 21 3D 06 59 AB DB 
LAN->HEIZ QUR  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: 
DC C2 00 0B 06 3D 21 06 59 49 B6 
HEIZ->LAN ANS  836 Heizkreis 1 -  Mischer Nachstellzeit Tn: DC120 s
DC 80 42 0E 07 21 3D 06 59 00 00 78 C4 96 
#836: 120 s
`

fredlcore, 2020-12-09:
Ok, then the enable byte is 0x01. This is also the way it is in version 2.0, so all should be fine and I'll close this issue now. Please feel free to reopen it if there are still any problems.