fredlcore/BSB-LAN #196: Testing the recent version: problems, probable issues, questions...

1coderookie, 2020-12-01:
Hi guys,
I opened this issue as a possibility to report and discuss several problems, probable issues and questions of/about the recent version(s) of BSB-LAN. I hope it's ok for you ;)

@dukess: Because I always had some issues that seemed to only appear within my system (@fredlcore had none of these problems), I just tested the recent version (2.0.14-20201117142051) with a complete fresh system: a not yet used Due and a new EEPROM (straight from the seller). Neither of these had ever contact with BSB-LAN, so I just imitated the absolute 'new' user, who starts over at the first time ever.

I used the default settings of the config.h file, so I didn't change anything there.
The flashing itself via Arduino IDE was flawless. 
After the reboot I switched to the configuration page (url/C). 
Then I changed two settings via the webinterface: 
1. I set 'Read config from EEPROM' to 'Ein' (= on) and
2. I set 'Write parameters to system' from 'Aus' (= off) to 'Ein (Engineer + OEM)'. 
Then of course I clicked on 'Daten absenden' (= send data / save config), the button at the end of the page to save the settings.
The URL changed to url/CI as usual, showing me the new settings. 

Because one of the problems in the past was that the settings obviously weren't saved, I clicked on 'Einstellungen' (= settings; url/C) again to see, if the new settings now have been saved.  
But: The system doesn't react anymore. The page seems to be loading (judged by the item of the browser tab) and the serial monitor shows "Reading EEPROM..." - but nothing happens anymore since then. 
See the full output of this sequence of the serial monitor, which stops after 'Reading EEPROM...':
```
14:28:45.582 -> EPROM option 9: 421
14:28:45.582 -> Address EEPROM option 10: 4Addressress EEPROM option 11: 423
14:28:45.582 -> Address EEPROM option 12: 427
14:28:45.582 -> Address EEPROM option 13: 428
14:28:45.582 -> Address EEPROM option 14: 588
14:28:45.582 -> Address EEPROM option 15: 589
14:28:45.582 -> Address EEPROM option 16: 593
14:28:45.582 -> Address EEPROM option 17: 753
14:28:45.582 -> Address EEPROM option 18: 759
14:28:45.582 -> Address EEPROM option 19: 760
14:28:45.582 -> Address EEPROM option 20: 764
14:28:45.582 -> Address EEPROM option 21⸮READY
14:28:45.774 -> Reading EEPROM...done.
14:28:45.774 -> EEPROM schema v.0 Program schema v.3
14:28:45.774 -> Address EEPROM option 0: 0
14:28:45.774 -> Address EEPROM option 1: 1
14:28:45.774 -> Address EEPROM option 2: 2
14:28:45.774 -> Address EEPROM option 3: 6
14:28:45.774 -> Address EEPROM option 4: 226
14:28:45.774 -> Address EEPROM option 5: 306
14:28:45.774 -> Address EEPROM option 6: 418
14:28:45.774 -> Address EEPROM option 7: 419
14:28:45.774 -> Address EEPROM option 8: 420
14:28:45.808 -> Address EEPROM option 9: 421
14:28:45.808 -> Address EEPROM option 10: 422
14:28:45.808 -> Address EEPROM option 11: 423
14:28:45.808 -> Address EEPROM option 12: 427
14:28:45.808 -> Address EEPROM option 13: 428
14:28:45.808 -> Address EEPROM option 14: 588
14:28:45.808 -> Address EEPROM option 15: 589
14:28:45.808 -> Address EEPROM option 16: 593
14:28:45.808 -> Address EEPROM option 17: 753
14:28:45.808 -> Address EEPROM option 18: 759
14:28:45.808 -> Address EEPROM option 19: 760
14:28:45.808 -> Address EEPROM option 20: 764
14:28:45.893 -> Address EEPROM option 21: 768
14:28:45.893 -> Address EEPROM option 22: 772
14:28:45.893 -> Address EEPROM option 23: 776
14:28:45.893 -> Address EEPROM option 24: 778
14:28:45.893 -> Address EEPROM option 25: 782
14:28:45.893 -> Address EEPROM option 26: 786
14:28:45.893 -> Address EEPROM option 27: 850
14:28:45.893 -> Address EEPROM option 28: 914
14:28:45.893 -> Address EEPROM option 29: 915
14:28:45.893 -> Address EEPROM option 30: 916
14:28:45.893 -> Address EEPROM option 31: 926
14:28:45.893 -> Address EEPROM option 32: 927
14:28:45.893 -> Address EEPROM option 33: 1087
14:28:45.893 -> Address EEPROM option 34: 1088
14:28:45.893 -> Address EEPROM option 35: 1092
14:28:45.893 -> Address EEPROM option 36: 1093
14:28:45.893 -> Address EEPROM option 37: 1094
[... 252 lines cut ...]
14:30:43.599 -> Adding known Max ID to list: 00000000
14:30:43.599 -> Adding known Max ID to list: 00000000
14:30:43.599 -> Adding known Max ID to list: 00000000
14:30:43.632 -> Adding known Max ID to list: 00000000
14:30:43.632 -> Adding known Max ID to list: 00000000
14:30:43.632 -> Adding known Max ID to list: 00000000
14:30:43.665 -> Adding known Max ID to list: 00000000
14:30:43.665 -> Adding known Max ID to list: 00000000
14:30:43.665 -> Adding known Max ID to list: 00000000
14:30:43.665 -> Adding known Max ID to list: 00000000
14:30:43.698 -> Adding known Max ID to list: 00000000
14:30:43.698 -> Adding known Max ID to list: 00000000
14:30:43.698 -> Adding known Max ID to list: 00000000
14:30:43.732 -> Adding known Max ID to list: 00000000
14:30:43.732 -> Adding known Max ID to list: 00000000
14:30:43.732 -> Adding known Max ID to list: 00000000
14:30:43.765 -> Adding known Max ID to list: 00000000
14:30:43.765 -> Adding known Max ID to list: 00000000
14:30:43.765 -> Adding known Max ID to list: 00000000
14:30:43.798 -> Adding known Max ID to list: 00000000
14:30:48.044 -> Connection to max_cul: failed
14:30:49.735 -> READY
14:30:49.835 -> Reading EEPROM...
```
Before I do something like /NE or a reboot now or before I'll describe the other problems I had with my other system in the past, I just wanted to report this behaviour because it already doesn't seem to work as expected.

dukess, 2020-12-01:
> Then I changed two settings via the webinterface:I set 'Read config from

Thank you for detail description!
Can i ask you add manually eeprom dump function before reboot and do config 
one more time. Probably info from eeprom can help us in fix this issue

"U.

1coderookie, 2020-12-01:
Sure, no problem - you just need to tell me exactly what I have to do / how I have to do it: Do I have to add something in the code? In that case I would need more details, I'm a noob when it comes down to programming.. ;)

1coderookie, 2020-12-01:
Oh, and did you see this part:
```
14:30:03.686 -> /C
14:30:03.686 -> URL: C
14:30:03.786 -> EEPROM dump:
14:30:03.786 -> ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff [... cut ...] f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
14:30:06.704 -> dc 8a 00 0b 06 3d 0d 05 19 4f 8c 
14:30:06.704 -> GET /favicon.ico HTTP/1.1
```
I know this always appeared in the past, but I thought you changed something about this in one of the latest patches..

malansk, 2020-12-01:
Hello,

I have been using Version 1.1.29 for some weeks with my PPS Heating Controller RVA53.243 (Year 2007).

Use case is to log some Controller Values (Vorlauftemperatur, Mischervorlauftemperatur, Außentemperatur) as well as 3 One-Wire tempsensors via CCU3 / CCU-Historian in Homematic smart home system.
Other use case is to switch Heating Controller to "Reduziertemperatur (18°C)" for 3hours if the sun is shining in the morning.

With the V1.1, I had the problem that as soon as I enabled PPS Bus Write access several Controller values are set to 0 (e.g. Set temperatures and times of all heating programs) - probably because BSB-Lan is no able to read the values programed in the controller propperly, therefore defaulting to "0" value which is then sent to the Controller as new value after boot of BSB-LAN.

I have solved this with some entries in the BSB_lan_custom_setup.h 
e.g. "set(15014, "22.5", 0);         //Komfort-Sollwert [°C]"
With this workaround, the proper values are written to the heating controller at very boot of BSB-LAN.
But of course changing some values needs recompile of BSB-Lan, therefore I wanted to switch to V2. 

With Version 2, this no longer works.
After start of Version 2, there are some odd values like 8 or 4 degrees set instead of 22.5 and 18.0 for the temperatures, and the times for heating are all 0.
Also, manual change from the WEB-UI for these values does not work. 
In V1 there used to be a "SET" button to click to transmit the new value, this button is now missing.

Also, fixed IP is not working. The device is not reachable and not seen in AVM Fritz Box if DHCP is set to false.
Setting DHCP back to true will also not work, not even after reflash of V2. 
I had to use setting "useEEPROM=0" in order to have the device reachable again (looks like corrupt setting in EPROM ?).

Sorry to cause so much bugs with my old PPS heating system but it was working so well with V1 :-)

Please let me know if I can do some more logging / testing.

Greetings,
Martin

fredlcore, 2020-12-01:
@malansk: I can only answer your PPS-value issues: With PPS we cannot read from the heating system. Rather, the heating system is constantly sending all sorts of parameter information in no specific order and at the same time it also requests information from the QAA. Therefore, when you switch to write-enable, these values are zero until the parameter is received from the heating system. 

However, as for comfort temperature setpoint ("Komfort-Sollwert"), this is a value transferred from the QAA (i.e. BSB-LAN) to the heater. That's why there is a "Set" button on the webinterface, so you can store the desired temperature (e.g. 22.5 degrees) in the EEPROM, so this will be read upon restart.
Does this not work for you in Version 2?

Fixed IP works for me, have you checked the output of the Serial Monitor when the device boots up? It should state the assigned IP address (no matter if DHCP or fixed).

If you think the EEPROM is corrupted, please run /NE - this clears the EEPROM to zero. We have had issues that a fresh EEPROM (which is filled with all 0xFF) can cause problems and we're investiagting this.

dukess, 2020-12-01:
@1coderookie I add EEPROM dump output for /CI command (after saving in EEPROM)
when @fredlcore accept pull request take it

Thank you.

fredlcore, 2020-12-01:
Already done :).

1coderookie, 2020-12-01:
Hmm, now it's getting even more weird: after flashing with the Arduino IDE, the regular boot sequence should have started. But: This is the output of the serial monitor:
```
READY
19:58:44.806 -> Reading EEPROM...
```
after that nothing happened anymore..
Then I pressed the reset button, same behaviour:
```
READY
19:58:44.806 -> Reading EEPROM...READY
20:05:47.808 -> Reading EEPROM...
```
I have to go now, sorry, but I'll proceed testing tomorrow if you have any more suggestions. Can also test with another fresh EEPROM then - one more left in stock.. ;)

fredlcore, 2020-12-01:
I had a similar problem which could be fixed by running /NE which clears the EEPROM. But since @dukess is also investigating this problem, maybe he has some other tests he would like you to run in order to get rid of this error...

fredlcore, 2020-12-01:
As for @malansk: I just checked it again, and the values that I stored using "Set" button (such as desired comfort temperature etc.) are read after boot.
However, @dukess : The new parameter "Write parameters to system" seems not to be stored in EEPROM. At least it is not active anymore once I boot.

malansk, 2020-12-01:
I did some more testing after clearing the EEPROM.
I have overlooked the "Write parameters to system" setting since it was previously not required and the WebGUI has the separate setting "PPS write enable" which I had set.
-> Maybe if Bus-Mode = PPS and "PPS write enabled" = yes then the "Write parameters to system" could be ignored since changing from "Engineer" to "Engineer + OEM" does not seem to make a difference for PPS.

The "Write parameters to system" setting also reverted back after boot, but I was able to work around by editing BSB_lan_config.h:
// defines default flag for parameters (use "#define DEFAULT_FLAG 0" to make (almost) all parameters writeable)
// if #define DEFAULT_FLAG FL_RONLY is set, then the program will always be in read-only mode.
//#define DEFAULT_FLAG FL_SW_CTL_RONLY
#define DEFAULT_FLAG 0

So now everything is working for me (did not test the fixed IP anymore), thank you very much for making this marvelous piece of hardware and software !
I upgraded my 1986 Heating system controller to a newer Heating controller from 2007 (baught used for 70€) and now I can program it via WebGUI and integrate it into my Homematic smart home system for data logging and room temperature control according to the smart home sensors :-)

fredlcore, 2020-12-01:
PPS write enabled is something different than "Write parameters to system", although I get that it is confusing in a PPS context. With "PPS write enabled", BSB-LAN actively sends data on the bus, even if it is just responding to a query from the heating system. This means that it will collide with a QAA that is on the same bus. "Write parameters to system" in this context means that values can be changed. For example, it makes sense to set "PPS write enable" to off when you still have a QAA on the bus, listen on all the settings (like time programs) and then store these settings in BSB-LAN's EEPROM using the "Set" button. Once all parameters have been stored this way, you can take the QAA off the bus and switch "PPS write enable" to on, then BSB-LAN will send the same values as the QAA before without you having to re-enter all of them manually.

And yes, setting DEFAULT_FLAG to 0 also does the trick, but that's just a workaround and you won't be able to change it in the webinterface then.

In any case, I'm really glad to hear stories like yours and your 1986 heating system. That's one of the main motivations why I put work in this project so that people don't just trash their old heating systems which may still be good in terms of the heating technology, but are just too "dumb" for current standards and BSB-LAN can help to make them smart and more energy saving again :).

dukess, 2020-12-02:
Found hanging reason. It was simple.
More harder: why some options doesn't write to EEPROM.

1coderookie, 2020-12-02:
When you're already investigating that, let me just mention some other things:

1) The problem that settings weren't been saved was one of the main problems I had, already with an older version - it never worked for me (e.g.: Activating 'Read config from EEPROM', activating IPWE, grant write access and more). 
That's why I used a fresh EEPROM & Due yesterday when I wanted to test the new version with the OEM-write-access.

2) In config.h the QAA70 model is set as default (PPS) - but the webinterface shows the QAA50 instead (PPS weren't not used/activated though, I'm always talking about the default settings of config.h now). 

3) Another thing that doesn't really make sense to me: 
We want the webinterface/webconfig to become the regular way to change settings. Therefore the usage of the EEPROM must be the standard. In config.h we have the two definements 'WEBCONFIG' and 'CONFIG_IN_EEPROM' activated by default. Fine so far.
But: If I understood @fredlcore right in one of our private discussions, I have to activate 'Read config from EEPROM' via webinterface to achieve that saved settings will still be present after a reboot. In my opinion this setting should also be the default setting, otherwise it doesn't really make a lot of sense - if I use the webconfig and change settings with the mind of a regular user, I expect that these settings will still be present after a reboot. That I have to set something before the 'real' usage to achieve this, doesn't really make sense to me in this case.

1coderookie, 2020-12-02:
4. Also tested with a previous version (before writeable OEMs were integrated): I changed the setting within the config.h of the write access and inserted the flag "0" to grant general write access. After doing that, the option 'All parameters is read only' were missing in the webconfig. 
Sure, probably because one isn't able to change that setting via webconfig if it's set within the config.h - but in that case it maybe would be better to still show that option but to block the function. 
I hope you understand what I mean - can't describe it with better words right now.. ;)

dukess, 2020-12-02:
>However, @dukess : The new parameter "Write parameters to system" seems not to be stored in EEPROM. At least it is not active anymore once I boot.

"not active" you mean  "not visible"? If yes then you need set DEFAULT_FLAG FL_SW_CTL_RONLY.
it will stay invisible until FL_SW_CTL_RONLY not set.

fredlcore, 2020-12-02:
@dukess: No, I have set FL_SW_CTL_RONLY. What I mean is that for PPS, the value I set here is not stored in EEPROM. But on my other Due which runs on a BSB bus, it works.

1coderookie, 2020-12-02:
Should I test the bugfix #199 now or are you already working on another one right now @dukess?

dukess, 2020-12-02:
> maybe would be better to still show that option but to block the function.
It possible but need to think how do it beautyful.

> If I understood @fredlcore right in one of our private discussions, I have to activate 'Read config from EEPROM' via webinterface to achieve that saved settings will still be present after a reboot. In my opinion this setting should also be the default setting

This is fixed in last pull: on "new" EEPROM  "Using EEPROM" will enabled by default.
Behavior on erased EEPROM depend from erasing pattern: if EEPROM will be set with 0x00 then "Using EEPROM" will be disabled after erasing.  if EEPROM will be set with 0xff then "Using EEPROM" will be enabled after erasing. 

> In config.h the QAA70 model is set as default (PPS) - but the webinterface shows the QAA50 instead (PPS weren't not used/activated though, I'm always talking about the default settings of config.h now).

I think i fix it in https://github.com/fredlcore/bsb_lan/pull/199

fredlcore, 2020-12-02:
@1coderookie: Regarding point 4., the option "off" (which would be "read-only") is displayed here in my setup. The thing is that DEFAULT_FLAG has to be FL_SW_CTL_RONLY for the webinterface to take effect. If users change this to some other value, they do so willingly. I wouldn't want to waste memory space to add additional code to block specific settings then, because then we would have to do so in other places as well.

1coderookie, 2020-12-02:
@fredlcore: The errors I reported were all also present with the recent version from yesterday. I just mentioned the 'older' version to make clear, that it wasn't a 'new' problem which suddenly appeared within the recent version.

fredlcore, 2020-12-02:
Ah, ok, thanks!

dukess, 2020-12-02:
@fredlcore You change PPS with /K48 (15000-15042) ?

1coderookie, 2020-12-02:
Thanks for the quick fix, I'll start testing asap!

> This is fixed in last pull: on "new" EEPROM "Using EEPROM" will enabled by default.
> Behavior on erased EEPROM depend from erasing pattern: if EEPROM will be set with 0x00 then "Using EEPROM" will be [...]

Just for my understanding (because I'll need to describe the procedure of the usage within the manual in the future anyway): why did you deactivate it if the EEPROM already have been erased once with 0x00? Don't you think it would make more sense to activate it by default as soon as 'WEBCONFIG' and/or 'CONFIG_IN_EEPROM' is activated? So that any saved setting within the webinterface will still be the same after a reboot? 
Or did I understand that 'Read config from EEPROM' wrong..?

fredlcore, 2020-12-02:
@dukess: PPS is compiled in through bus_type in _config.h. I'm talking about the configuration page. If I activate write enabled, /K48 works until reboot. After that, no more, and I have to enable it again in configuration page to be able to cahnge parameters in /K48.

dukess, 2020-12-02:
>So that any saved setting within the webinterface will still be the same after a reboot?

I think there may be reasons why a user might want to load with the configuration from a _config.h without recompiling the program.

fredlcore, 2020-12-02:
Ah, that makes sense, so I can read switch back to reading the compiled configuration from _config.h if I set it to off? Then that's fine.

And yes, the heating parameters are updated in EEPROM, but when I go to configuration page (/C) and set "Write parameters to system" to "on", this is not stored in EEPROM. It stays only until reboot. So I can change parameters in /K48, but after reboot, I have to set "Write parameters to system" again to "on".

dukess, 2020-12-02:
> this is not stored in EEPROM. It stays only until reboot. 

Can you drop EEPROM dump here?
It is possible that the parameter is being written successfully. But back when loading it is no longer readable due to the bug that Ulf mentioned.

fredlcore, 2020-12-02:
Sure:
```
POST /4444/CI HTTP/1.1                                                          
/4444/CI                                                                        
URL: CI                                                                         
Option ID: 0                                                                    
Option value: 0                                                                 
Option ID: 6                                                                    
Option value: 2                                                                 
Option ID: 7                                                                    
Option value: 66                                                                
Option ID: 8                                                                    
Option value: 66                                                                
Option ID: 9                                                                    
Option value: 0                                                                 
Option ID: 10                                                                   
Option value: 1                                                                 
Option ID: 43                                                                   
Option value: 83                                                                
Option ID: 17                                                                   
Option value: 00:80:41:19:69:91                                                 
Option ID: 18                                                                   
Option value: 0                                                                 
Option ID: 19                                                                   
Option value: 192.168.1.150                                                     
Option ID: 20                                                                   
Option value: 255.255.255.0                                                     
Option ID: 21                                                                   
Option value: 192.168.1.1                                                       
Option ID: 22                                                                   
Option value: 192.168.1.1                                                       
Option ID: 23                                                                   
Option value: 80                                                                
Option ID: 24                                                                   
Option value: 0.0.0.0                                                           
Option ID: 25                                                                   
Option value: 0.0.0.0                                                           
Option ID: 11                                                                   
Option value: 0                                                                 
Option ID: 12                                                                   
Option value: 0                                                                 
Option ID: 13                                                                   
Option value: 8700,8323,8326                                                    
Option ID: 14                                                                   
Option value: 0                                                                 
Option ID: 15                                                                   
Option value: 3600                                                              
Option ID: 16                                                                   
Option value: 8700,8743,8314                                                    
Option ID: 26                                                                   
Option value: 4444                                                              
Option ID: 27                                                                   
Option value:                                                                   
Option ID: 29                                                                   
Option value: 7                                                                 
Option ID: 30                                                                   
Option value: 2,3                                                               
Option ID: 31                                                                   
Option value: 0                                                                 
Option ID: 32                                                                   
Option value: 8700,8743,8314,8750,8830,8740,8741,8326,8323,8337,8703,8704       
Option ID: 33                                                                   
Option value: 0                                                                 
Option ID: 34                                                                   
Option value: 192.168.1.5                                                       
Option ID: 35                                                                   
Option value: 1                                                                 
Option ID: 36                                                                   
Option value: 1                                                                 
[... cut ...]
00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 0
0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
0 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 
```

fredlcore, 2020-12-02:
And this is the dump after reboot:
```
READY                                                                           
Reading EEPROM                                                                  
Using settings from config file                                                 
Reading done.                                                                   
EEPROM schema v.3 Program schema v.3                                            
Address EEPROM option 0: 0                                                      
Address EEPROM option 1: 1                                                      
Address EEPROM option 2: 2                                                      
Address EEPROM option 3: 6                                                      
Address EEPROM option 4: 226                                                    
Address EEPROM option 5: 306                                                    
Address EEPROM option 6: 0                                                      
Address EEPROM option 7: 0                                                      
Address EEPROM option 8: 0                                                      
Address EEPROM option 9: 0                                                      
Address EEPROM option 10: 0                                                     
Address EEPROM option 11: 0                                                     
Address EEPROM option 12: 0                                                     
Address EEPROM option 13: 0                                                     
Address EEPROM option 14: 0                                                     
Address EEPROM option 15: 0                                                     
Address EEPROM option 16: 0                                                     
Address EEPROM option 17: 0                                                     
Address EEPROM option 18: 0                                                     
Address EEPROM option 19: 0                                                     
Address EEPROM option 20: 0                                                     
Address EEPROM option 21: 0                                                     
Address EEPROM option 22: 0                                                     
Address EEPROM option 23: 0                                                     
Address EEPROM option 24: 0                                                     
Address EEPROM option 25: 0                                                     
Address EEPROM option 26: 0                                                     
Address EEPROM option 27: 0                                                     
Address EEPROM option 28: 0                                                     
Address EEPROM option 29: 0                                                     
Address EEPROM option 30: 0                                                     
Address EEPROM option 31: 0                                                     
Address EEPROM option 32: 0                                                     
Address EEPROM option 33: 0                                                     
Address EEPROM option 34: 0                                                     
Address EEPROM option 35: 0                                                     
Address EEPROM option 36: 0                                                     
Address EEPROM option 37: 0                                                     
Address EEPROM option 38: 0                                                     
Address EEPROM option 39: 0                                                     
Address EEPROM option 40: 0                                                     
Address EEPROM option 41: 0                                                     
Address EEPROM option 42: 0                                                     
Address EEPROM option 43: 0                                                     
Address EEPROM option 44: 0                                                     
Address EEPROM option 45: 0                                                     
Address EEPROM option 46: 0                                                     
Size of cmdtbl1: 46032                                                          
Size of cmdtbl2: 44832                                                          
Size of cmdtbl3: 192                                                            
free RAM: 85799                                                                 
Init One Wire bus...                                                            
numSensors: 0                                                                   
PPS settings:                                                                   
Starting SD..ok                                                                 
192.168.1.150                                                                   
255.255.255.0                                                                   
192.168.1.1                                                                     
Waiting 3 seconds to give Ethernet shield time to get ready...                  
Calculating free space on SD...2097151 MB free                                  
Start network services                                                          
                                                                                
[... cut ...]
00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 00 00 00 0
0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
0 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 
```

dukess, 2020-12-02:
First byte is 0 so configuration will read from _config.h

>Address EEPROM option 6: 0                                                      
Address EEPROM option 7: 0                                                      
Address EEPROM option 8: 0                                                      
Address EEPROM option 9: 0                                                      
Address EEPROM option 10: 0                                                     
Address EEPROM option 11: 0                                                     
Address EEPROM option 12: 0                                                     
Address EEPROM option 13: 0                                                     
Address EEPROM option 14: 0                                                     
Address EEPROM option 15: 0                                                     

I catch same problem: if first byte is non-0x96 then "Address EEPROM option" is zero. But here must be addresses.

fredlcore, 2020-12-02:
Anything else I can do?

dukess, 2020-12-02:
No. I think i fix it.

fredlcore, 2020-12-02:
Hm, one more strange thing that may be related: I'm trying out a new QAA-Type for PPS (0x66 / 102). If I set QAA_TYPE in _config.h to 102 and erase the EEPROM with /NE, it still somehow uses 0x53 in the telegrams. Also explicitly setting "Read config from EEPROM" to "off" does not help. I wonder where the 0x53 still comes from...

dukess, 2020-12-02:
It's legacy. Line 5781    if (pps_values[PPS_QTP] == 0) pps_values[PPS_QTP] = QAA_TYPE;

fredlcore, 2020-12-02:
Exactly, that's why I was wondering. If I hard-code `pps_values[PPS_QTP] =102`, then it works, so that's why I wonder where the 0x53 comes from...

dukess, 2020-12-02:
We'll see. Fresh version may fix this problem.

fredlcore, 2020-12-02:
You mean #200 ? No, that didn't do it...

dukess, 2020-12-02:
What happened if we add new string to ENUM15062 with QAA name and id?

fredlcore, 2020-12-02:
Could do it, but I'm still wondering why the value in _config.h is not taken?

dukess, 2020-12-02:
It can give us little bit more info

fredlcore, 2020-12-03:
Ok, I ran some checks and it is this line here in (currently) line 8310:
`readFromEEPROM(CF_PPS_VALUES);`
that changes pps_value[PPS_QTP] from whatever value is configured in _config.h in varialbe QAA_TYPE. Even though I have repeatedly cleared the EEPROM with /NE, the value still changes to 0x83 after that line. But 0x83 is nowhere in code other than the dropdown menu in the configuration page which I don't access in that test.
It's not a big deal, but still strange why it is not possible to set a fixed value unless you set UseEEPROM in _config.h to 0.

dukess, 2020-12-03:
Thank you for your investigation!
I rewrote code responsible for initializing PPS:
Now it do this:
1. read PPS settings from EEPROM (in any case of UseEEPROM, as pre 2.x version)
2. if UseEEPROM is true then read QAA_TYPE from EEPROM.
3. if PPS settings is wrong (0xFFFF or 0x0000) then QAA_TYPE will read from _config.h

fredlcore, 2020-12-03:
Great, thanks, will try it out later!

1coderookie, 2020-12-03:
Thanks for the quick patches! :)

I just flashed the newest version with the default config.h and the system I mentioned at the very first post. 
This time a problem with starting of the SD appeared:
```
10:32:58.969 -> READY
10:32:59.035 -> Reading EEPROM
10:32:59.267 -> EEPROM schema CRC mismatch
10:32:59.267 -> Reading done.
10:32:59.267 -> EEPROM schema v.255 Program schema v.3
10:32:59.267 -> Update EEPROM schema
10:32:59.997 -> Address EEPROM option 0: 0
10:32:59.997 -> Address EEPROM option 1: 1
10:32:59.997 -> Address EEPROM option 2: 2
10:32:59.997 -> Address EEPROM option 3: 6
10:32:59.997 -> Address EEPROM option 4: 226
10:32:59.997 -> Address EEPROM option 5: 306
10:32:59.997 -> Address EEPROM option 6: 418
10:32:59.997 -> Address EEPROM option 7: 419
10:32:59.997 -> Address EEPROM option 8: 420
10:32:59.997 -> Address EEPROM option 9: 421
10:33:00.030 -> Address EEPROM option 10: 422
10:33:00.030 -> Address EEPROM option 11: 423
10:33:00.030 -> Address EEPROM option 12: 427
10:33:00.030 -> Address EEPROM option 13: 428
10:33:00.030 -> Address EEPROM option 14: 588
10:33:00.030 -> Address EEPROM option 15: 589
10:33:00.030 -> Address EEPROM option 16: 593
10:33:00.030 -> Address EEPROM option 17: 753
10:33:00.030 -> Address EEPROM option 18: 759
10:33:00.030 -> Address EEPROM option 19: 760
10:33:00.030 -> Address EEPROM option 20: 764
10:33:00.030 -> Address EEPROM option 21: 768
10:33:00.030 -> Address EEPROM option 22: 772
10:33:00.030 -> Address EEPROM option 23: 776
10:33:00.063 -> Address EEPROM option 24: 778
10:33:00.063 -> Address EEPROM option 25: 782
10:33:00.063 -> Address EEPROM option 26: 786
10:33:00.063 -> Address EEPROM option 27: 850
10:33:00.063 -> Address EEPROM option 28: 914
10:33:00.063 -> Address EEPROM option 29: 915
10:33:00.063 -> Address EEPROM option 30: 916
10:33:00.063 -> Address EEPROM option 31: 926
10:33:00.063 -> Address EEPROM option 32: 927
10:33:00.063 -> Address EEPROM option 33: 1087
10:33:00.063 -> Address EEPROM option 34: 1088
10:33:00.063 -> Address EEPROM option 35: 1092
10:33:00.096 -> Address EEPROM option 36: 1093
10:33:00.096 -> Address EEPROM option 37: 1094
10:33:00.096 -> Address EEPROM option 38: 1095
10:33:00.096 -> Address EEPROM option 39: 1099
10:33:00.096 -> Address EEPROM option 40: 1131
10:33:00.096 -> Address EEPROM option 41: 1163
10:33:00.096 -> Address EEPROM option 42: 1195
10:33:00.096 -> Address EEPROM option 43: 1227
10:33:00.096 -> Address EEPROM option 44: 1229
10:33:00.096 -> Address EEPROM option 45: 1230
10:33:00.096 -> Address EEPROM option 46: 1231
10:33:00.096 -> Size of cmdtbl1: 46032
10:33:00.129 -> Size of cmdtbl2: 44832
10:33:00.129 -> Size of cmdtbl3: 192
10:33:00.129 -> free RAM: 85807
10:33:00.129 -> Init One Wire bus...
10:33:00.129 -> numSensors: 0
10:33:00.129 -> PPS settings:
10:33:00.129 -> Starting SD..failed
10:33:02.816 -> 192.168.178.37
10:33:02.816 -> 255.255.255.0
10:33:02.816 -> 192.168.178.1
10:33:02.816 -> Waiting 3 seconds to give Ethernet shield time to get ready...
[... 253 lines cut ...]
11:03:06.036 -> HEIZ->LAN ANS 6226 Konfiguration - Gerätevariante: 100
11:03:06.036 -> dc 80 42 0e 07 05 3d 00 03 00 00 64 68 0e 
11:03:06.036 -> #6226: 100
11:03:06.036 -> Device family: 96
11:03:06.069 -> Device variant: 100
11:03:06.069 -> 
11:03:06.069 -> Setup complete
```

Then I flashed the same version again onto the Due, just to see if everything would be as the default config.h or if something else would be set:
1) The problem with the 'Starting SD....failed' appeared again, also after a two reboots. 
(Maybe the card got compromised somehow..? I have to check with a card reader.. :( )  
2) The settings I changed before (IPWE=on and writeaccess=Engineer as mentioned above) are still the same - so they are NOT as the default config.h which has been flashed now! 
Up to me, this shouldn't be - flashing BSB-LAN completely new to the Due should bring the result, that any setting I did before would be erased and the (default) settings of the config.h should appear again. 
I don't know why this doesn't happen - is there no routine to erase the EEPROM before flashing the new version to the Due or something like that?  

Here is the EEPROM-dump of /C after the second reboot, maybe it helps somehow:
```
11:23:35.269 -> /C
11:23:35.269 -> URL: C
11:23:35.368 -> EEPROM dump:
11:23:35.368 -> 96 03 41 f0 ff fa 4b 45 51 30 35 30 32 33 32 36 00 4b 45 51 30 35 30 35 30 38 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  [... cut ...] ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 
```

I'll do some more tests soon, also another procedure with a complete fresh system (Due&EEPROM).

1coderookie, 2020-12-03:
> 2. The settings I changed before (IPWE=on and writeaccess=Engineer as mentioned above) are still the same - so they are NOT as the default config.h which has been flashed now!
>    Up to me, this [...]

Hmm, just thought about it - maybe "it's not a bug - it's feature"?!? ;)
It can be a nice feature if one doesn't have to set all parameters new after upgrading/flashing a new version of BSB-LAN.
But on the other hand it's maybe causing problems or irritations if the settings of the config.h seem to be 'ignored' after flashing a new or clean version of BSB-LAN.
What do you guys think?
(Btw: It wasn't due to a browser-cache problem or something like that, I also checked with a different browser which I haven't used before..)

dukess, 2020-12-03:
> Hmm, just thought about it - maybe "it's not a bug - it's feature"?!? ;)

Yes. This is a feature. :) 
When config options will get their values from _config.h:
1. This is a new board.
2. UseEEPROM in_config.h is set as 0;
3. "Using EEPROM" is set as "Off" in web-config.
4. When we delete any unneeded option from config or add new config option with version less or equal current (current version is 3). In this case we will get error "Config schema CRC mismatch", then loading values from _config.h, rebuilding schema and writing default values to EEPROM.

in other cases program will load their settings from EEPROM. (Yes, if we add new options with version +1 to current (Ex. 4) then program will load all available options (versions 0 - 3) from EEPROM, get other options (version 4) from _config.h, rebuild schema CRC and store new options to EEPROM. After next start it will be schema version 4. And so on.)

Now in your EEPROM stored QAA50: 52 00 00 00 00 42 42
Can you do /NE (Reboot with EEPROM erasing) and show EEPROM dump after that?
(It is possible in EEPROM was stored values from old "wrong" program)

fredlcore, 2020-12-03:
@dukess: Wow, this is a really sophisticated mechanism, amazing how much thought you put into it :)...
@1coderookie: The EEPROM should not be erased after flashing a new version, that would really be a hassle and has also not been the case with the Mega (where the EEPROM was just used to store the PPS values). If you want to erase the EEPROM, you have the /NE command, afterwards it should really look like a fresh board. 
As for the SD card, I also have this sometimes, maybe the card is getting worn out?

dukess, 2020-12-03:
With SD card you can try to change

`  if(!SD.begin(4, SPI_DIV3_SPEED)) printToDebug(PSTR("failed\r\n")); // change SPI_DIV3_SPEED to SPI_HALF_SPEED if you are still having problems getting your SD card detected`

to 

`  if(!SD.begin(4, SPI_HALF_SPEED)) printToDebug(PSTR("failed\r\n")); // change SPI_DIV3_SPEED to SPI_HALF_SPEED if you are still having problems getting your SD card detected`

1coderookie, 2020-12-03:
> > Hmm, just thought about it - maybe "it's not a bug - it's feature"?!? ;)
> 
> Yes. This is a feature. :)

Ah ok, that' nice! :)

> Now in your EEPROM stored QAA50: 52 00 00 00 00 42 42
> Can you do /NE (Reboot with EEPROM erasing) and show EEPROM dump after that?
> (It is possible in EEPROM was stored values from old "wrong" program)

Sure, I just did, but it still shows QAA50 in webconfig-interface.
SerMo-output:
```
13:32:35.919 -> GET /NE HTTP/1.1

13:32:35.919 -> /NE
13:32:35.919 -> URL: NE
13:32:36.814 -> Cleared EEPROM
13:32:36.814 -> READY
13:32:36.914 -> Reading EEPROM
13:32:36.914 -> Using settings from config file
13:32:36.914 -> Reading done.
13:32:36.914 -> EEPROM schema v.0 Program schema v.3
13:32:36.914 -> Update EEPROM schema
13:32:37.908 -> Address EEPROM option 0: 0
13:32:37.908 -> Address EEPROM option 1: 1
13:32:37.908 -> Address EEPROM option 2: 2
13:32:37.908 -> Address EEPROM option 3: 6
13:32:37.908 -> Address EEPROM option 4: 226
13:32:37.908 -> Address EEPROM option 5: 306
13:32:37.941 -> Address EEPROM option 6: 418
13:32:37.941 -> Address EEPROM option 7: 419
13:32:37.941 -> Address EEPROM option 8: 420
13:32:37.941 -> Address EEPROM option 9: 421
13:32:37.941 -> Address EEPROM option 10: 422
13:32:37.941 -> Address EEPROM option 11: 423
13:32:37.941 -> Address EEPROM option 12: 427
13:32:37.941 -> Address EEPROM option 13: 428
13:32:37.941 -> Address EEPROM option 14: 588
13:32:37.941 -> Address EEPROM option 15: 589
13:32:37.941 -> Address EEPROM option 16: 593
13:32:37.941 -> Address EEPROM option 17: 753
13:32:37.974 -> Address EEPROM option 18: 759
13:32:37.974 -> Address EEPROM option 19: 760
13:32:37.974 -> Address EEPROM option 20: 764
13:32:37.974 -> Address EEPROM option 21: 768
13:32:37.974 -> Address EEPROM option 22: 772
13:32:37.974 -> Address EEPROM option 23: 776
13:32:37.974 -> Address EEPROM option 24: 778
13:32:37.974 -> Address EEPROM option 25: 782
13:32:37.974 -> Address EEPROM option 26: 786
13:32:37.974 -> Address EEPROM option 27: 850
13:32:37.974 -> Address EEPROM option 28: 914
13:32:37.974 -> Address EEPROM option 29: 915
13:32:37.974 -> Address EEPROM option 30: 916
13:32:38.008 -> Address EEPROM option 31: 926
13:32:38.008 -> Address EEPROM option 32: 927
13:32:38.008 -> Address EEPROM option 33: 1087
13:32:38.008 -> Address EEPROM option 34: 1088
13:32:38.008 -> Address EEPROM option 35: 1092
13:32:38.008 -> Address EEPROM option 36: 1093
13:32:38.008 -> Address EEPROM option 37: 1094
13:32:38.008 -> Address EEPROM option 38: 1095
13:32:38.008 -> Address EEPROM option 39: 1099
13:32:38.008 -> Address EEPROM option 40: 1131
13:32:38.008 -> Address EEPROM option 41: 1163
13:32:38.008 -> Address EEPROM option 42: 1195
13:32:38.041 -> Address EEPROM option 43: 1227
13:32:38.041 -> Address EEPROM option 44: 1229
13:32:38.041 -> Address EEPROM option 45: 1230
[... 22 lines cut ...]
13:32:43.881 -> HEIZ->LAN ANS 6226 Konfiguration - Gerätevariante: 100
13:32:43.881 -> dc 80 42 0e 07 05 3d 00 03 00 00 64 68 0e 
13:32:43.881 -> #6226: 100
13:32:43.881 -> Device family: 96
13:32:43.881 -> Device variant: 100
13:32:43.881 -> 
13:32:43.881 -> Setup complete
13:32:46.304 -> DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
13:32:46.304 -> dc 8a 00 0b 06 3d 0d 05 19 4f 8c 
13:32:46.371 -> HEIZ->DISP ANS 8310 Diagnose Erzeuger - Kesseltemperatur: 41.1 &deg;C
13:32:46.404 -> dc 80 0a 0e 07 0d 3d 05 19 00 0a 48 99 8c 
13:32:47.266 -> GET /favicon.ico HTTP/1.1

13:32:47.266 -> /favicon.ico
13:32:48.128 -> GET /C HTTP/1.1

13:32:48.161 -> /C
13:32:48.161 -> URL: C
13:32:48.260 -> EEPROM dump:
13:32:48.260 -> 96 03 41 f0 ff fa 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 52 00 00 00 00 42 42 00 00 00 00 00 00 00 fc 21 00 00 86 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 0e 00 00 fc 21 00 00 27 22 00 00 7a 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 [... cut ...] ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
13:32:56.484 -> dc 8a 00 0b 06 3d 0d 05 19 4f 8c 
13:32:56.551 -> HEIZ->DISP ANS 8310 Diagnose Erzeuger - Kesseltemperatur: 41.1 &deg;C
13:32:56.584 -> dc 80 0a 0e 07 0d 3d 05 19 00 0a 48 99 8c 
```
(Btw: I didn't do anything with the SD card yet, right now it seems to work again.. So never mind - as long as it's no bug within BSB-LAN, then everything is fine ;) )

1coderookie, 2020-12-03:
And here a little screenshot of that QAA-line:
[image]

1coderookie, 2020-12-03:
> Now in your EEPROM stored QAA50: 52 00 00 00 00 42 42

Just if it could be a helpful information: I _never_ set the QAA50, I dind't change anything within the config.h, always used the default settings.

fredlcore, 2020-12-03:
You set it the moment you save the configuration. Then all the config options are written to EEPROM exactly as they are displayed in the config page. So even if you just set Read config from EEPROM to yes, the QAA50 value gets written. That's why you have to erase the EEPROM first.

dukess, 2020-12-03:
Ok. I will try tomorrow (or may be today) erase eeprom and look on this one more time.

1coderookie, 2020-12-03:
@fredlcore: Sorry, but I don't get it - I flashed the default config.h where the QAA70 is preset by default. To a brand new EEPROM. I also did /NE to erase the EEPROM. I _never_ set the QAA50 anywhere - neither in the config.h nor via webinterface. I just didn't touch it. Still the QAA50 shows up.

What you said would only make sense to me, if the QAA50 would be set by default if PPS wasn't used and therefore the default value of the config.h would be ignored. If it's like that - it's fine with me. I just tested it with the default config.h, expected the same output within the webinterface as the config.h-values are and saw, that it's not the same. I think/thought that it's a bug and reported it here.

fredlcore, 2020-12-03:
I'm only saying that the moment you change _anything_ in the webinterface _all_ options are written. So if at any point you were writing the config (as you did in some of your testing), the "wrong" QAA50 value was written to EEPROM, too. In the meantime there have been a few bugfixes that should prevent that behaviour, but nevertheless your logs indicate that the QAA50 value is in the EEPROM and will stay there until erased.

fredlcore, 2020-12-03:
EDIT: I see that the 0x52 is still in the EEPROM dump after you issued /NE, so that's likely a problem. I don't exactly understand how the EEPROM is re-initialized after /NE because the "magic bytes" in the beginning are apparently written there, too, but the 0x52 should not be there if the value is initialized otherwise in _config.h.

1coderookie, 2020-12-03:
> So if at any point you were writing the config (as you did in some of your testing), the "wrong" QAA50 value was written to EEPROM, too. In the meantime there have been a few bugfixes that should prevent that behaviour, 

Understood that - that's why I did the tests today with the latest version (after you merged the last pullrequest this morning, I think it was #201). So I expected this to be fixed. Because it wasn't and even showed up after /NE, I reported it again ;)

fredlcore, 2020-12-03:
Correct :) - see my EDIT above.

1coderookie, 2020-12-03:
Yep, just wanted to mention that I used the latest version which already included the latest QAA-bugfixes from this morning ;)

[33 more replies]

closed 2021-01-04 by fredlcore