fredlcore/BSB-LAN #197: LMS14 - 8327 - Water Pressure Sensor

anonymous-one, 2020-12-01:
Hello Fred / all others who put the hard work in bringing this project to life.

I have an LMS14 based boiler, working great with BSB-LAN. I apologize if I butcher the terminology a bit here, but I will do my best to be as clear as possible.

A) I already sent the new parameters to Fred a few months ago when I was installing it.

B) My LMS14 has a water pressure sensor for heating circuit.

C) I know this sensor works FOR SURE as we recently had a leak in our heating line and pressure dropped to 0 and the panel on the boiler showed the correct error and stopped heating the water etc. Also the correct error message was recorded in BSBLAN (6801 Error - Historie 1 Fehlercode: 119 - Fehler Druckschalter).

D) On the panel (LCD on boiler), parameter 8327 DOES NOT show a value.

E) "Check for new parameters" does find it though, here is a paste:

```
Start test...

8327 - Diagnosis producer - Water pressure
0x053D3063
DC C2 00 0B 06 3D 05 30 63 2B 9A
DC 80 42 0E 07 05 3D 30 63 01 00 00 26 20
```

F) 8327 under BSBLAN shows "not supported".

Any way to have BSBLAN show me the psi / bar value of this sensor? Would be very useful for me to plot this over time and record and changes.

Here is a complete dump from my "check for new parameters". I don't see anything sensitive here so I am posting this publically:

```
Version: 1.1.19-20200903101000
Scanning for devices......
Device address found: 0
Device address found: 10

Teste Geräteadresse 0...
Device family: 162
Device variant: 14
Geräte-Identifikation: LMS14.002A100
Device SW version: 3.5
Entwicklungs-Index: 00005B - decoding error
Device OV version: 0.8
Bootloader-Version: (parameter not supported)
EEPROM-Version: 22.0
Konfiguration - Info 2 OEM: 0
Zugangscode Inbetriebnahme?: 0
Zugangscode Fachmannebene ?: 0
Zugangscode OEM?: 0
Zugangscode OEM2?: 0
Bisher unbekannte Geräteabfrage: 20
Hersteller-ID (letzten vier Bytes): 132210069
Bisher unbekannte Geräteabfrage: 00010000DC - unknown type
Outside temp sensor local (10003): 0.0 °C
Outside temp sensor local (10004): 0.0 °C

6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
162;14;LMS14.002A100 ;3.5;;0.8;22.0;0;0;0;0;0;20;132210069;00010000DC;

Start test...

8327 - Diagnosis producer - Water pressure
0x053D3063
DC C2 00 0B 06 3D 05 30 63 2B 9A
DC 80 42 0E 07 05 3D 30 63 01 00 00 26 20

Test finished.

Teste Geräteadresse 10...
Device family: 92
Device variant: 100
Geräte-Identifikation: AVS37.294/100
Device SW version: 6.5
Entwicklungs-Index: (parameter not supported)
Device OV version: 102.0
Bootloader-Version: (parameter not supported)
EEPROM-Version: (parameter not supported)
Konfiguration - Info 2 OEM: (parameter not supported)
Zugangscode Inbetriebnahme?: (parameter not supported)
Zugangscode Fachmannebene ?: (parameter not supported)
Zugangscode OEM?: (parameter not supported)
Zugangscode OEM2?: (parameter not supported)
Bisher unbekannte Geräteabfrage: ---
Hersteller-ID (letzten vier Bytes): 1028352
Bisher unbekannte Geräteabfrage: (parameter not supported)
Outside temp sensor local (10003): °C (parameter not supported)
Outside temp sensor local (10004): °C (parameter not supported)

6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
92;100;AVS37.294/100;6.5;;102.0;;;;;;;---;1028352;;

Start test...

Test finished.

Done.
```

anonymous-one, 2020-12-01:
I see this is mentioned here:

https://github.com/fredlcore/bsb_lan/issues/45#issuecomment-571999102

But google is doing a horrible job of translating DE to EN, so I am unsure if this was resolved / is resolvable?

fredlcore, 2020-12-01:
Hi Tomasz,

I'm sorry if I forgot to notify you, but I have included all the parameters that were found for your system. You just have to install the most recent version, your's is from 3rd of September, but since you sent me your parameter check on 2nd of September, I'm quite sure that the parameters I added were only in a later version. So please check the most current version and let us know if the problem still exists or if we can close this problem.

1coderookie, 2020-12-01:
> But google is doing a horrible job of translating DE to EN, so I am unsure if this was resolved / is resolvable?

Just a little hint for future purposes: try https://www.deepl.com/translator instead of Google translator, it seems to be doing a much better job ;)

anonymous-one, 2020-12-01:
@1coderookie Thanks!

@fredlcore Thanks! Will update and re-open if it persist.

anonymous-one, 2020-12-01:
I had a moment to flash the latest version to my arduino.

It still lists 8327 as not supported.

a) As mentioned previously, on the LCD panel on the boiler when a few months ago I checked what 8327 was being shown as, it was not showing an actual bar value. From what I recall it was just "-".

b) We had this sensor replaced about 3 years ago during our yearly maint as it was leaking slightly. From what I recall the serviceman asked "XX for a original... or XX (lower) for an aftermarket..."... Knowing how things go, I most likely went for the aftermarket one. Although I am unsure if it matters as its probably a simple sensor that varies resistance as pressure changes.

c) On a related note to B, again, not sure how this would impact BSB-LAN and reading of that value... but... Could it be that this sensor does not send over a range but rather just trips on / off if the water pressure drops below a certain value in my particular unit?

d) If it helps at all, the "needle" value as shown by the mechanical pressure gauge was showing about 1.1bar when I pulled the dump below. Although it looks like the needle gauge is connected to the return pipe and the electrical sensor sits on the exit pipe (or vice versa, unsure which is which to be honest....).

Here is a paste of the latest dump:

```
Version: 2.0.14-20201117142051
Scanning for devices......
Device address found: 0
Device address found: 10

Teste Geräteadresse 0...
Device family: 162
Device variant: 14
Geräte-Identifikation: LMS14.002A100
Device SW version: 3.5
Entwicklungs-Index: 00005B - decoding error
Device OV version: 0.8
Bootloader-Version: (parameter not supported)
EEPROM-Version: ---
Konfiguration - Info 2 OEM: 0
Zugangscode Inbetriebnahme?: 0
Zugangscode Fachmannebene ?: 0
Zugangscode OEM?: 0
Zugangscode OEM2?: 0
Bisher unbekannte Geräteabfrage: 20
Hersteller-ID (letzten vier Bytes): 132210069
Bisher unbekannte Geräteabfrage: 04010100DC - unknown type
Outside temp sensor local (10003): -3.6 °C
Outside temp sensor local (10004): -3.6 °C

6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
162;14;LMS14.002A100 ;3.5;;0.8;---;0;0;0;0;0;20;132210069;04010100DC;

Start test...

8327 - Diagnosis producer - Water pressure
0x053D3063
DC C2 00 0B 06 3D 05 30 63 2B 9A
DC 80 42 0E 07 05 3D 30 63 01 00 00 26 20

Test finished.

Teste Geräteadresse 10...
Device family: 92
Device variant: 100
Geräte-Identifikation: AVS37.294/100
Device SW version: 6.5
Entwicklungs-Index: (parameter not supported)
Device OV version: 102.0
Bootloader-Version: (parameter not supported)
EEPROM-Version: (parameter not supported)
Konfiguration - Info 2 OEM: (parameter not supported)
Zugangscode Inbetriebnahme?: (parameter not supported)
Zugangscode Fachmannebene ?: (parameter not supported)
Zugangscode OEM?: (parameter not supported)
Zugangscode OEM2?: (parameter not supported)
Bisher unbekannte Geräteabfrage: ---
Hersteller-ID (letzten vier Bytes): 1028352
Bisher unbekannte Geräteabfrage: (parameter not supported)
Outside temp sensor local (10003): °C (parameter not supported)
Outside temp sensor local (10004): °C (parameter not supported)

6225;6226;6224;6220;6221;6227;6229;6231;6232;6233;6234;6235;6223;6236;6237;
92;100;AVS37.294/100;6.5;;102.0;;;;;;;---;1028352;;

Start test...

Test finished.

Done.
```

fredlcore, 2020-12-01:
Thanks, this really was an error on my part, and thanks for finding and linking the old issue #45 - I have now added the 05... command ID for your device family 162 / variant 14 and hope that it was really just an error on my part and not two machines with identical device families/variants that have different command IDs for the same parameter. But I'll find out about this once the next person does a /Q run and wants to revert back to the old one ;)...
Please test this and close the issue in case it runs fine now (as it should).

closed 2020-12-01 by fredlcore

anonymous-one, 2020-12-02:
Thanks! Its showing up now.

Shows ---, but I am sure that's due to the LMS not BSBLAN as that's the same thing thats showing on the LCD panel.

Have a great day!

fredlcore, 2020-12-02:
Yes, that's correct because the data payload of the telegram is 01 00 00 and the first byte 01 stands for "disabled", the actual value would be after that, i.e. 00 00, i.e. zero.