fredlcore/BSB-LAN #207: Add text identifier for parameter type and unit type to JSON export

fredlcore, 2020-12-06:
@konne got me to add clear text identifiers for each parameter type into the optbl[] struct which can now be accessed via optbl[x].type_text. The same goes for the unit type which so far has been just an ENUM id which (though at the current stage rather unlikely) could be bound to change. This one can be accessed via the new dt_types_text[x].type_text.
@dukess, do you think it would be good to add this to those JSON responses where now some kind of parameter or data type is sent? External modules would then have it easier to process the values returned.

dukess, 2020-12-06:
good idea but with worst consequences: program no longer fits in RAM of Mega (all identifiers placed in RAM , ~830 bytes). Yes, I remember that we are ending support of Mega but this is too harsh.
Can we do two structures and tables: For Mega and for Due?

fredlcore, 2020-12-06:
Oops, that was not the intention - I thought it would work as it does with the unit texts (like "rpm", "°C" etc.), these are all stored in flash. Doesn't the compiler store all const chars in flash anyway? Or Is there a way to use PSTR or F() here or would I have to work with char pointers which are then referring to const chars?

dukess, 2020-12-06:
Tried F and PSTR - compiler return errors " statement-expressions are not allowed outside functions nor in template-argument lists". So here it can be do through references as other strings but i'm not sure that will be enough.

fredlcore, 2020-12-06:
Hm, but all the char arrays in cmdtbl are in flash memory, so it must be somehow possible to move them from RAM to Flash?

dukess, 2020-12-06:
Yep. So me must do same thing: define variables in far memory and set 
references to variables in table.

fredlcore, 2020-12-06:
Ok, will do...

fredlcore, 2020-12-06:
Done. Hope it works again on the Mega now.

dukess, 2020-12-06:
i will look on this tomorrow

dukess, 2020-12-07:
New version has no RAM consumption. It's good.

fredlcore, 2020-12-07:
Ok, as for adding both data and unit types in clear text into the JSON responses, do you want to do it or should I do it?

dukess, 2020-12-07:
as you wish but i have pain when need to made two or three step conversion from far address to char *.

konne, 2020-12-07:
@fredlcore thanks for the changes, they look great and I will try to use it for the structure generation

dukess, 2020-12-07:
An idea occurred to me: adding names for each data type (VT_* ) is still useless because in JSON used DT_* type only. 
@konne Or here is reason for show low-level data type?

fredlcore, 2020-12-07:
Yes, parameter type (VT_*) may be not be as useful for external modules because we send the data type (DT_*) together with the unit as string. But with the parameter type, external modules would be able to handle other aspects more specifically. For example, it may be helpful for the module to know that the value received is of the "pressure" type which could then be displayed differently in comparison to a "temperature" type. Currently, the data type for both will be DT_VALS (i.e. a numerical value). The external module could parse the unit string, but that again is not the same across all languages. So transmitting the parameter type would make it easier for external modules if they choose to interpret it.

As for data type (DT_*), currently we send an ENUM id number. While this is unlikely to change in the future, it is still safer to send a unique (text) identifier that modules could parse then. Then it wouldn't matter if the ENUM table with the DT_* entries gets resorted.

fredlcore, 2020-12-07:
> as you wish but i have pain when need to made two or three step conversion from far address to char *.

Ok, then I'll look into it, no problem...

dukess, 2020-12-07:
> For example, it may be helpful for the module to know that the value received is of the "pressure" type which could then be displayed differently in comparison to a "temperature" type. 

Clear!

dukess, 2020-12-08:
@fredlcore  I look on last diffs: it looks cool!

fredlcore, 2020-12-08:
But I haven't implemented the sending of data/parameter types in text identifier form. If it is as difficult as you say, I may just add this information in JSON-Output for Due users, as it also saves memory for Mega users who may not need this feature.

dukess, 2020-12-08:
Pull https://github.com/fredlcore/bsb_lan/pull/215

fredlcore, 2020-12-08:
That's great to hear!
I would put them in all JSON/MQTT output functions where so far we have referred to optbl[].type.

fredlcore, 2020-12-20:
Can we consider this closed?

closed 2021-01-04 by fredlcore