fredlcore/BSB-LAN #208: Improve MQTT

Luposoft63, 2020-12-06:
Hi fredlcore,

here now my changes in the BSB-Lan.ino. 
I commented my changes with '//Luposoft'.
With it BSB-LAN is able to receive publishes from a broker to control the heater.

syntax from FHEM - 'set <mqtt2Server> publish fromFHEM <parameter>=<value>'
e.g. : set mqtt2Server publish fromFHEM 700=1

[BSB_lan.zip](https://github.com/fredlcore/bsb_lan/files/5649050/BSB_lan.zip)

Known issue: actually it's important to have at least 1 value in log_parameters, because the connection with mqtt runs in the loop
The better way could be, a separate function for the mqtt-connect-instructions to call this single time while start.
But this is something for later...

I hope you want to include it
Thanks a lot

fredlcore, 2020-12-06:
Thank you for your contribution! I have only had a short look at it, but there are some things that won't work (well enough) to include it into the repository: For one, all commands are set using the SET telegram type. Room temperature, for example, requires an INF telegram. Furthermore, please do not write directly to Serial. Use the printToDebug() function, otherwise people who use the telnet output won't be able to read these messages.
Another thing that need improvement is proper indention.
Once this is done, please try to familiarize yourself with "pull requests" here at GitHub, it makes it much easier for us to merge the code and we can write comment and questions to you right in the code layout itself, that makes fixing these things much easier for both sides.

But these things aside, it's great to see that BSB-LAN can work as a MQTT receiver and that would certainly be a functionality that people would be looking for.

Luposoft63, 2020-12-07:
Ok, I understand the needing changes.
I close this issue and will call a pull-request later. (when I'm able to do...)

closed 2020-12-07 by Luposoft63

rroblik, 2021-01-08:
@fredlcore is that backported or not ? Thanks

fredlcore, 2021-01-08:
Backported to what?

1coderookie, 2022-02-01:
@Luposoft63 
A question from a user came up and I realized that I don't know the answer and also don't have anything about it in the manual. When using the new syntax to query parameters from the broker side, is it possible to query more than one parameter at the same time?
So right now the example in the manual is
```set mqtt2Server publish BSB-LAN /700```
which queries 700. But could one also do sth like
 ```set mqtt2Server publish BSB-LAN /700/701/702/1000```
to query all four parameters at the same time (don't mind the parameter numbers, it's just an example ;) )?