fredlcore/BSB-LAN #214: Determine enable flag bytes of various parameter types

fredlcore, 2020-12-08:
A number of parameters still cannot be set yet because we never received a serial log of the SET process of one of these parameters. This has to be done at the heating system itself while BSB-LAN is running and the serial monitor captures the relevant telegrams.

I'll post the parameters in question here in a format that you can copy-paste to your browser where just your BSB-LAN IP needs to be put in front. Then you can see which parameters are supported by your heating system and check if it is available on the heater itself.

I will update this list (as long as GitHub lets me) so that only the still undecoded parameters are displayed here. Please note that only **ONE** decoded parameter for each parameter type is enough, so you don't need to check all X other parameters of the same parameter type.

```
VT_BYTE10
/2904
VT_GRADIENT_SHORT
/3834
VT_LPBADDR
/6600/6650
VT_PRESSURE50
/2491/2492
VT_SECONDS_SHORT
/2222/2238/2446/2459/2511/2560/2655/2656/2803/2852/2853/2873/2966/3017/3831/5105/5142/5145/5151/5486/5487/5488/5575/5585/5732/5733/5772/6127
VT_VOLTAGE
/5953/5955/6049/6051/7399/7401/7710/7711/7725/7840/7845/7849/7854/7870/7881
VT_VOLTAGEONOFF
/7884/7911/7912/7913/7914/7915/7916/7917/7919/7945/7946/7950/7951/7952
VT_SPF
/8398
VT_CURRENT
/2700/2730/2742/2743/2745/8329/9618/9619
VT_CURRENT1000
/8329
VT_DAYS_WORD
/7011
VT_ERRORCODE
/6705/6801/6803/6805/6807/6809/6811/6813/6815/6817/6819/6821/6823/6825/6831/6833/6835/6841/6843/6845/6851/6855/6861/6865/6871/6875/6881/6885/6891/6895/6901/6911/6921/6931/6941/6951/6961/6971/6981/6991
VT_FP1
/70/6220/6227/6228/6229/7073
VT_GRADIENT
/794/1094/1394/6112
VT_INTEGRAL
/2863/2864/2883
VT_MONTHS_WORD
/7045/7046
VT_MINUTES_WORD10
/5480
VT_POWER_WORD
/2330/2331/2441/2442/2443/2444/2445/2452/3108/9504/9512/9524/9529
VT_POWER_WORD100
/3195/3196
VT_ENERGY_WORD
/5740/5811/5813/8526
VT_ENERGY_CONTENT
/3106
VT_PRESSURE_WORD
/6140/6141/6143/6144/8327/8424/8433/9005/9006/9009
VT_PRESSURE_1000
/8327
VT_PROPVAL
/2540/2543/9630
VT_SPEED
/2441/7050/9504/9512/9524/9527/9563
VT_SPEED2
/2441/2442/2443/2444/2452/7050/8323/8324/9504/9505/9506/9512/9513/9514/9523/9524/9526/9529/9530/9531/9551/9616/9651
VT_TEMP_WORD60
/2510
VT_VOLTAGE_WORD
/5953/5955/5963/5965/6049/6051/6057/6059/6065/6067
VT_LITERPERHOUR
/2898/2899/3097/3098/3257
VT_LITERPERMIN
/8460/8461/8521/8860
VT_PPS_TIME
/15061
VT_HOURS
/6222/7044/8330/8332/8336/8338/8339/8340/8444/8448/8449/8450/8452/8454/8456/8481/8530/8531/8532/8570/8840/8842/8990
VT_SECONDS_DWORD
/7231/7232/7233
VT_POWER
/2445
VT_POWER100
/8395/8396/8397
VT_ENERGY10
/8527
```

dukess, 2020-12-08:
You need results of SET or query enough?

2491 - 0x05/0x06
LAN->HEIZ QUR 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc c2 00 0b 06 3d 05 30 68 9a f1
HEIZ->LAN ANS 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: 0.10 UNIT_BAR_TEXT
dc 80 42 0d 07 05 3d 30 68 00 05 79 32
#2491: 0.10 UNIT_BAR_TEXT

2700 - ? readonly
URL: JQ=2700
LAN->HEIZ QUR 2700 ENUM_CAT_19_TEXT - STR2700_2_TEXT:
dc c2 00 0b 06 3d 09 0f 31 31 e7
HEIZ->LAN ANS 2700 ENUM_CAT_19_TEXT - STR2700_2_TEXT: 46.42 UNIT_CURR_TEXT
dc 80 42 0e 07 09 3d 0f 31 00 12 22 d3 0e
#2700: 46.42 UNIT_CURR_TEXT

8329
URL: JQ=8329
LAN->HEIZ QUR 8329 ENUM_CAT_2c_TEXT - STR2730_TEXT:
dc c2 00 0b 06 3d 09 0e 16 56 53
HEIZ->LAN ANS 8329 ENUM_CAT_2c_TEXT - STR2730_TEXT: 46.14 UNIT_CURR_TEXT
dc 80 42 0e 07 09 3d 0e 16 00 12 06 7b da
#8329: 46.14 UNIT_CURR_TEXT

fredlcore, 2020-12-08:
No, I need (only) SET telegrams because the 10th byte in a SET telegram is either 06 or 01, and we can only figure out when logging a 'real' set procedure at the heating system, so you have to have one of these parameters on your heater in order to do this test. For that, I have listed all the possible parameters in each parameter type category. So e.g. for VT_HOURS, you only need to check 6222 or 7044 or ... Once you have found one that you can set on your boiler, you don't need to check the others.

dukess, 2020-12-10:
6140

POST /LKJBHksdgfrleneaslkrwklh/JS HTTP/1.1
/LKJBHksdgfrleneaslkrwklh/JS
URL: JS
setting line: 6140 val: 01 1d
LAN->HEIZ SET 6140 ENUM_CAT_24_TEXT - STR6140_TEXT: --- UNIT_BAR_TEXT
dc c2 00 0d 03 3d 05 05 d8 01 1d 6f 8c
HEIZ->LAN ACK 6140 ENUM_CAT_24_TEXT - STR6140_TEXT:
dc 80 42 0b 04 05 3d 05 d8 82 9e
Setting parameter 6140 to 2.9 with type 1
setting line: 6140 val: 01 1d
LAN->HEIZ SET 6140 ENUM_CAT_24_TEXT - STR6140_TEXT: --- UNIT_BAR_TEXT
dc c2 00 0d 03 3d 05 05 d8 01 1d 6f 8c
HEIZ->LAN ACK 6140 ENUM_CAT_24_TEXT - STR6140_TEXT:
dc 80 42 0b 04 05 3d 05 d8 82 9e
Setting parameter 6140 to 2.9 with type 1

8327 - read only

fredlcore, 2020-12-10:
But this is again from BSB-LAN, not on from the heater, right? Or did you test it with 0x01 and 0x06 and only 0x01 was working?

dukess, 2020-12-10:
Yep, it is from BSB-LAN. little bit dumb :-(

1coderookie, 2020-12-11:
Here are the rest of the parameters which worked at my system (probably with a little bit more info then you asked for ;) ):

```
8526 -> readonly
---
6140: OEM!
10:35:40.566 -> RGT1->HEIZ SET 6140 Konfiguration - Wasserdruck Maximum: --- bar
10:35:40.566 -> DC 86 00 0D 03 3D 05 05 D8 06 00 EA A8
---
6141: OEM!
> RGT1->HEIZ SET 6141 Konfiguration - Wasserdruck Minimum: --- bar
10:44:12.906 -> DC 86 00 0D 03 3D 05 05 D9 05 00 88 CB 
---
9005,9006,9009: readonly
---
6222: readonly

7044:
10:54:55.586 -> RGT1->HEIZ SET 7044 Wartung/Sonderbetrieb - Wartungsintervall: --- Monate
10:54:55.586 -> DC 86 00 0D 03 3D 05 05 E1 05 01 F4 EE 

8330, 8332, 8530, 8531, 8570: values only resettable
e.g. 8570: 
10:52:03.965 -> RGT1->HEIZ SET 8570 Diagnose Erzeuger - Betr'std Feststoffkessel: --- h
10:52:03.965 -> DC 86 00 10 03 3D 51 08 92 01 00 00 00 00 28 5F 
---
8527: only resettable
10:56:51.138 -> RGT1->HEIZ SET 8527 Diagnose Erzeuger - Gesamtertrag Solarenergie: --- kWh
10:56:51.138 -> DC 86 00 10 03 3D 49 05 98 01 00 00 00 00 67 39
```

fredlcore, 2020-12-11:
Great, thanks!

dukess, 2020-12-11:
Now all parameters from heater/RGT

2441 - set is 0x01

RGT1->HEIZ QUR 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 86 00 0b 06 3d 09 0f 80 7f a0
HEIZ->RGT1 ANS 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: 6800 UNIT_RPM_TEXT
dc 80 06 0e 07 09 3d 0f 80 00 1a 90 85 74
RGT1->HEIZ SET 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: --- UNIT_RPM_TEXT
dc 86 00 0e 03 3d 09 0f 80 01 1a 9a 2f 4e
HEIZ->RGT1 ACK 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 80 06 0b 04 09 3d 0f 80 e5 b0
RGT1->HEIZ QUR 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 86 00 0b 06 3d 09 0f 80 7f a0
HEIZ->RGT1 ANS 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: 6810 UNIT_RPM_TEXT
dc 80 06 0e 07 09 3d 0f 80 00 1a 9a 24 3e
RGT1->HEIZ QUR 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 86 00 0b 06 3d 09 0f 80 7f a0
HEIZ->RGT1 ANS 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: 6810 UNIT_RPM_TEXT
dc 80 06 0e 07 09 3d 0f 80 00 1a 9a 24 3e
RGT1->HEIZ QUR 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 86 00 0b 06 3d 09 0f 80 7f a0
RGT1->HEIZ SET 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: --- UNIT_RPM_TEXT
dc 86 00 0e 03 3d 09 0f 80 01 1a 90 8e 04
HEIZ->RGT1 ACK 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 80 06 0b 04 09 3d 0f 80 e5 b0
RGT1->HEIZ QUR 2441 ENUM_CAT_18_TEXT - STR2441_TEXT:
dc 86 00 0b 06 3d 09 0f 80 7f a0
HEIZ->RGT1 ANS 2441 ENUM_CAT_18_TEXT - STR2441_TEXT: 6800 UNIT_RPM_TEXT
dc 80 06 0e 07 09 3d 0f 80 00 1a 90 85 74

2491 - set is 0x01

RGT1->HEIZ QUR 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc 86 00 0b 06 3d 05 30 68 63 4c
HEIZ->RGT1 ANS 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: 0.10 UNIT_BAR_TEXT
dc 80 06 0d 07 05 3d 30 68 00 05 ba 04
HEIZ->EM1 INF      0504024E 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00
dc 80 03 1b 02 05 04 02 4e 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 29 73
HEIZ->EM1 INF      05040227 00 00 00 20
dc 80 03 0f 02 05 04 02 27 00 00 00 20 fc 84
RGT1->HEIZ SET 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: --- UNIT_BAR_TEXT
dc 86 00 0d 03 3d 05 30 68 01 06 7d aa
HEIZ->RGT1 ACK 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc 80 06 0b 04 05 3d 30 68 c3 0f
RGT1->HEIZ QUR 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc 86 00 0b 06 3d 05 30 68 63 4c
HEIZ->RGT1 ANS 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: 0.12 UNIT_BAR_TEXT
dc 80 06 0d 07 05 3d 30 68 00 06 8a 67
RGT1->HEIZ SET 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: --- UNIT_BAR_TEXT
dc 86 00 0d 03 3d 05 30 68 01 05 4d c9
HEIZ->RGT1 ACK 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc 80 06 0b 04 05 3d 30 68 c3 0f
RGT1->HEIZ QUR 2491 ENUM_CAT_18_TEXT - STR2491_TEXT:
dc 86 00 0b 06 3d 05 30 68 63 4c
HEIZ->RGT1 ANS 2491 ENUM_CAT_18_TEXT - STR2491_TEXT: 0.10 UNIT_BAR_TEXT
dc 80 06 0d 07 05 3d 30 68 00 05 ba 04

2510 - set is 0x06, set --- is 0x05

RGT1->HEIZ QUR 2510 ENUM_CAT_18_TEXT - STR2510_TEXT:
dc 86 00 0b 06 3d 0d 11 21 26 d7
HEIZ->RGT1 ANS 2510 ENUM_CAT_18_TEXT - STR2510_TEXT: 0.0 UNIT_DEG_TEXT
dc 80 06 0e 07 0d 3d 11 21 00 00 00 7c 83
RGT1->HEIZ SET 2510 ENUM_CAT_18_TEXT - STR2510_TEXT: --- UNIT_DEG_TEXT
dc 86 00 0e 03 3d 0d 11 21 05 00 00 6c 54
HEIZ->RGT1 ACK 2510 ENUM_CAT_18_TEXT - STR2510_TEXT:
dc 80 06 0b 04 0d 3d 11 21 aa f6
RGT1->HEIZ QUR 2510 ENUM_CAT_18_TEXT - STR2510_TEXT:
dc 86 00 0b 06 3d 0d 11 21 26 d7
[... 46 lines cut ...]
RGT1->HEIZ QUR 5953 ENUM_CAT_24_TEXT - STR5953_TEXT:
dc 86 00 0b 06 3d 05 0b 7b 98 71

6140 - set is 0x06

RGT1->HEIZ QUR 6140 ENUM_CAT_24_TEXT - STR6140_TEXT:
dc 86 00 0b 06 3d 05 05 d8 3e f7
HEIZ->RGT1 ANS 6140 ENUM_CAT_24_TEXT - STR6140_TEXT: 2.8 UNIT_BAR_TEXT
dc 80 06 0d 07 05 3d 05 d8 00 1c 57 4f
RGT1->HEIZ SET 6140 ENUM_CAT_24_TEXT - STR6140_TEXT: --- UNIT_BAR_TEXT
dc 86 00 0d 03 3d 05 05 d8 06 1b 49 f2
HEIZ->RGT1 ACK 6140 ENUM_CAT_24_TEXT - STR6140_TEXT:
dc 80 06 0b 04 05 3d 05 d8 9e b4
RGT1->HEIZ QUR 6140 ENUM_CAT_24_TEXT - STR6140_TEXT:
dc 86 00 0b 06 3d 05 05 d8 3e f7
HEIZ->RGT1 ANS 6140 ENUM_CAT_24_TEXT - STR6140_TEXT: 2.7 UNIT_BAR_TEXT
dc 80 06 0d 07 05 3d 05 d8 00 1b 27 a8

/7231 - time, sec /7232 - time, sec /7233 - times - all read only
8327 - read only

8329 - read only
8526 - read only
8860 -read only
9618/9619 - read only

fredlcore, 2020-12-11:
Thanks a lot!
I had to leave VT_SPEED in there because 2441 on your heater is VT_SPEED2 (which we already have identified).

fredlcore, 2020-12-11:
Ok, I figured out there is also a way to test the enable bytes just from BSB-LAN, but it takes a bit more effort:
You can edit the third value in the optbl[] table (currently line 510 onwards in _defs.h) which should be zero in the parameters above we are looking for.
You can then set it to either 1 or 6, flash it, and try to set one of these parameters via BSB-LAN. In the serial monitor, you will see that after the SET telegram, there will either bei a ACK (correct enable byte) or a NACK (wrong enable byte) telegram. You can then post the SET and ACK telegrams here.
Of course this only works if the parameter in question works within BSB-LAN, so for "parameter not supported" parameters, you don't need to try.

fredlcore, 2022-02-01:
@loehnertj has given me the helpful information that in his observations the 1 stands for non-nullable/disableable parameters and 6 for parameters that you can disable / set to null. 5 is then used to actually disable / set to null these parameters. Thanks for that!

dukess, 2022-02-02:
wow that's cool!

fredlcore, 2022-02-02:
No, we can derive from `enable_flag` whether the parameter is nullable or not. If it's 6, then it's nullable, if it's 1, it's not. Not sure if it requires any further changes in the code because users (should) already know that setting a parameter to an empty value is the same as disabling it. We could merely add a check for parameters with `enable_byte == 1` and inform them that this parameter cannot be disabled. Otherwise they will get the generic error that writing to the parameter is not possible which might be misleading...

fredlcore, 2022-02-02:
And I just realized why we never came across this: For some parameters (such as temperatures, of which we have quite a lot) it still works if you set them with enable byte 1 even though they use a 6 when changing it on the heater itself, as I just tried with parameter 730. But an error does in fact occur if you want to disable / null that setting! This will only work if the enable byte for that command is set to 6 (and subsequently a 5 is used when disabling that parameter). However, it does not seem to work the other way round, so using a 6 for (at least some of the) parameters that use a 1 did not work. 

So what be basically would have to do is to test all parameters with a data type that has an `enable_byte` of 1 and check if that parameter also works with a 6. If that's the case, then it's very likely that this parameter is in fact nullable.  This may lead to new data types such as now with VT_TEMP which currently is set to 1, but some parameters (such as 730) require it to be running with 6, so a new VT_TEMP6 (or so) would be required. I'll see if I can come up with a test script and then it would be good if you guys could run that against your machines...

dukess, 2022-02-02:
>  VT_TEMP6 (or so) would be required
or we can add #define FL_NULLABLE       64 for "flags" field in command tables.

fredlcore, 2022-02-02:
That's actually a great idea, the problem is that we'd have to change _all_ parameters then and we would 'waste' one of the bits in `dev_flags`. Once we need one more bit for flags, we'd have to go from uint8_t to uint16_t which would cost us one byte per line in cmd_tbl, so approximately 3kB.

But while we're at it, we could move information about the parameter such as signed or unsigned in that bitfield as well. It would make the cmdtbl a bit more cluttered (because you could end up with flags such as DEFAULT_FLAG+FL_NULLABLE+FL_SIGNED, but it would reduce the number of value types that we have. If we have to sacrifice one more byte anyways, we could also use four of those bits to encode the length of the payload, thus doing away with VT_SECONDS_SHORT, VT_SECONDS_WORD and VT_SECONDS_DWORD.

I think I'll have to have a look at how many parameters would actually be affected from this and then see what's the easier, less time consuming way...

loehnertj, 2022-02-06:
Some particular discrepancies I found while comparing my own (i.e. ``bsbgateway``'s) parameter set with the BSB-LAN one. My data is based solely on my own device (Broetje ISR-Plus Ecotherm Plus WGB-S 17/20 E -- built 2012). Would be good if this could be cross-checked on other devices.

* 1646 Legionellenfunktion Verweildauer is nullable but enable_byte=1
* 3825 Ladezeit relativer Vorrang is nullable but enable_byte=1
* 3827 Wartezeit Parallelbetrieb is nullable but enable_byte=1
* 3830 Kollektorstartfunktion is nullable but enable_byte=1
* 3832 Kollektorstartfunktion ein is NOT nullable but enable_byte=6
* 3833 Kollektorstartfunktion aus is NOT nullable but enable_byte=6
* 3834 Kollektorstartfkt. Gradient is nullable but enable_byte=1
* 3881 Frost'mittel Konzentration is NOT nullable but enable_byte=6
* 3884 Pumpendurchfluss is NOT nullable and has enable_byte=0

fredlcore, 2022-02-06:
Hm, these are my findings here on Elco Thision S 19 with LMU75 (device family 97):
1646 has enable_byte = 6
The other ones are solar parameters which I don't have.
Could you please change these parameters and log the telegrams on the bus while you are changing them? It would be interesting to have a look. And please send us your device family and variant for your heater as well. Thanks!

loehnertj, 2022-02-06:
Device family 162 variant 10.
You can find some traced telegrams in my notes file: https://github.com/loehnertj/bsbgateway/blob/bdafba143fb23b5fd3f88a4d6dc70ba6c693ff39/doc/notes.txt#L175
(addresses: "10" = operations panel, "0" = central device)
I'll see if I can collect logs for all of the above fields, if it is any help.

fredlcore, 2022-02-06:
Thanks, but those logs are not sufficient (or the link didn't point to the right line). What we need is something like this:
```
DSP1->HEIZ SET 1646 Trinkwasser - Legionellenfunktion Verweildauer: --- min     
DC 8A 00 0E 03 3D 31 04 96 05 00 0A 9D 49                                       
HEIZ->DSP1 ACK 1646 Trinkwasser - Legionellenfunktion Verweildauer:             
DC 80 0A 0B 04 31 3D 04 96 B0 FC                                                
DSP1->HEIZ QUR 1646 Trinkwasser - Legionellenfunktion Verweildauer:             
DC 8A 00 0B 06 3D 31 04 96 AC 1E                                                
HEIZ->DSP1 ANS 1646 Trinkwasser - Legionellenfunktion Verweildauer: --- min     
DC 80 0A 0E 07 31 3D 04 96 01 00 0A 81 90                                       
```
The clear text lines are of course not necessary, but the raw data is.

loehnertj, 2022-02-06:
Raw data is in the ``[raw: XX XX XX]`` snippet. It contains just the payload bytes, i.e. offset 0x09 ... (length-2) of the message. Admittedly the four-byte command ID is missing. Will generate better traces soon :-)

fredlcore, 2022-02-06:
That would theoretically be sufficient, but then your conclusions above seem not to be correct because for 3830 you say that enable_byte is 1 and it is nullable, but this is what I find in your logs:
```
<BsbTelegram 10 -> 0: set 3830 Solar - Kollektorstartfunktion = 5 min [raw:06 05 ] @1449653838.984828>
<BsbTelegram 0 -> 10: ack 3830 Solar - Kollektorstartfunktion = None min [raw:] @1449653839.080084>
```
If I interpret your statement correctly, then 06 would be the enable byte, not 1 as you said above, correct? 
The same goes for 3832 which you say is not nullable, but enable byte is 6. Your logs, however, say otherwise.

So maybe you could check your data again, because it would really make me wonder if this difference really exists.
As I said, it's possible to set a nullable parameter with 1 as long as you don't want to disable it. So maybe that's where this comes from?

loehnertj, 2022-02-07:
Hi, see edit above: "is nullable / not nullable" = bsbgateway's command table (hopefully matching the notes ;-)); "enable_byte=x" = BSB-Lan's table. Please excuse my being unprecise.

For example: 3830 can be set to NULL (according to the log), thus it is marked "nullable" in bsbgateways params. The compare script found enable_byte=1 instead of the expected 6 in the BSB-Lan table. Thus the line says "parameter is nullable, but [...BSB-Lan..] enable_byte=1". The others are to be read analogously.

fredlcore, 2022-02-07:
Ah, now I get it - yes, there are quite a few errors still in our table. As mentioned above, I just assumed that enable_byte 1 works in most cases, and if there were no errors, I just kept it. I didn't realize that it was connected to the nullable feature and hardly every tested a parameter against being nullable. So if you have concrete findings about certain parameters where the enable_byte needs to be changed, please let us know. BTW: enable_byte 0 means we have no knowledge about the enable_byte so far, so these are the parameters that still need to be looked at.

fredlcore, 2022-10-12:
No longer relevant with upcoming version 3.0

closed 2022-10-12 by fredlcore