fredlcore/BSB-LAN #217: write parameter via json do not work

cn72, 2020-12-09:
Hi,
I'am working with IOBroker and the BSB-Lan Adapter. The Adapter set the Parameter via json, but with Version 2.0 it will not work.
I tyr to set the Parameter wirh value (trying 1-3) using curl from console. I get the status = 1 but the value is not setting.

with version 1.1.53 it works.

I have the Log from serielle Monitor:
with Version 2.0
`GET /JA HTTP/1.1

/JA
URL: JA
POST /JS HTTP/1.1

/JS
URL: JS
setting line: 700 val: 01 00 
LAN->HEIZ INF      3D2D0574 01 00 
DC C2 00 0D 02 3D 2D 05 74 01 00 D6 15 
Setting parameter 700 to Type with type 0
GET /JQ=700 HTTP/1.1

/JQ=700
URL: JQ=700
LAN->HEIZ QUR  700 Heizkreis 1 - Betriebsart: 
DC C2 00 0B 06 3D 2D 05 74 9C 4B 
HEIZ->LAN ANS  700 Heizkreis 1 - Betriebsart: 3 - Komfort
DC 80 42 0D 07 2D 3D 05 74 00 03 87 10 
#700: 3 - Komfort
DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C `

and hre from Version 1.1.53

`POST /JS HTTP/1.1

/JS
setting line: 700 val: 01 02 
LAN->HEIZ SET  700 Heizkreis 1 - Betriebsart: ---
DC C2 00 0D 03 3D 2D 05 74 01 02 4E 36 
HEIZ->LAN ACK  700 Heizkreis 1 - Betriebsart: 
DC 80 42 0B 04 2D 3D 05 74 44 75 
Setting parameter 700 to 2 with type 1
GET /JQ=700 HTTP/1.1

/JQ=700
LAN->HEIZ QUR  700 Heizkreis 1 - Betriebsart: 
DC C2 00 0B 06 3D 2D 05 74 9C 4B 
HEIZ->LAN ANS  700 Heizkreis 1 - Betriebsart: 2 - Reduziert
DC 80 42 0D 07 2D 3D 05 74 00 02 97 31 
#700:  2 - Reduziert
DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C 
HEIZ->DISP ANS 8310 Diagnose Erzeuger - Kesseltemperatur: 55.9 &deg;C
DC 80 0A 0E 07 0D 3D 05 19 00 0D FC E7 44 `

fredlcore, 2020-12-09:
Ok, that's interesting. First of all, thanks for the detailed logs. The reason why setting the parameter does not work is that in version 2.0, the SET type is interpreted as type 0 (i.e. INF telegram). This, however, is only used for very few parmeters.
In version 1.1.53, it is correctly identified as type 1 (i.e. SET telegram).
Can you also post the JSON string that is sent to BSB-LAN? Is it really the same or is the SET type different in both requests?

dukess, 2020-12-09:
Until last weekend, the parameters were applied successfully

fredlcore, 2020-12-09:
@dukess: Does it mean they no longer work fine with you, too, now? @cn72 was reporting the error already before last week to me and in another forum, so it seems to be like this a while - at least under certain circumstances...

dukess, 2020-12-09:
i can test fresh version at friday

dukess, 2020-12-09:
https://github.com/fredlcore/bsb_lan/commit/de056280629c63646d061ca3f0e98d71c0f81603

This line like me not:
If  we set search_val > 255 then this condition will be true and function 
will build value from two byte in any case (enums with values < 255 and > 255)

if((byte)(pgm_read_byte_far(enumstr+c+1))!=' ' || search_val > 255) {

Or i'm wrong?

cn72, 2020-12-09:
I just load the last version from github.
With curl from linus it works.

`
curl -v -H "Content-Type: application/json" -X POST -d '{"Parameter":"700", "Value":"1", "Type":"1"}' http://<IP-Adresse>/JS

setting line: 700 val: 01 02 `
now there is the value right (here 02), prior Version i tried writes always 00. Also with curl. And with curl the Type parameter was not set to 1 with the prior Version of 2.0.

But via IObroker adapter it will not work. I'am not able to fetch the JSON String from BSBLAN Adpter.
The setting line is okay, but the Type semms to be set to INF.

I only try via curl without the Type specification. Then it is setting type to 00 and it dosen't work (expected).
Perhaps it's possible the IObroker adapter write not "Type" rather "dataType" getting via /JQ=700 ? But V1.1.53 do write "dataType" via /JQ=700....?

fredlcore, 2020-12-09:
The problem is that printENUM function looks for the first space in the string. If the search_val is \x01\x20 (for example like in ENUM_SWCODE_01_20_TEXT), then \x20 will be interpreted as space character.
If search_val is greater than 255, the ENUM string in question will always have two bytes in the beginning, so it is no problem that the condition matches. Under what circumstance would the condition match if search_val is < 255?
Or do you mean something else?

hacki11, 2020-12-09:
https://github.com/hacki11/ioBroker.bsblan/blob/65fb24d6444a85a18cf12501e3de0d27e8a7236d/lib/bsb.js#L52
That's how I'm calling the endpoint in iobroker. 

Writetype for 700 is 1:

https://github.com/hacki11/ioBroker.bsblan/blob/65fb24d6444a85a18cf12501e3de0d27e8a7236d/lib/config.js#L133

cn72, 2020-12-09:
@hacki11 are there the quotation marks missing in the body ?("Parameter": "id", ...) I tried with curl without quotation marks and the setting are not accepted.

fredlcore, 2020-12-09:
The code is looking for a colon (":") or quotation marks (") as delimiters, so it should be fine, but I'm uploading a version with debug output right now, please test it with activated serial monitor with both the curl and the ioBroker call, then we'll know more...

fredlcore, 2020-12-09:
Ok, no need to check, it tried it myself via cURL, and yes, it is the missing quotation marks.
@dukess: I only had a quick glance on the code in line 6900++, and it seems that you identify the value behind "Parameter", "Value" and "Type" by counting the quotation marks and colon. These increment the `stage` variable, and after the third occurrence, you read the value. If there are no quotation marks, then this solution does not work. 
My previous implementation seemed to have worked without the quotation marks, and since I don't know how many of the 13 external modules have the same problem as ioBroker has now: Do you see a chance to change the code so that it does not rely on quotation marks?

hacki11, 2020-12-10:
I think quotation marks on property keys in JSON are required by RFC. I can not believe javascript is doing something wrong here, but we need to check. Does the curl call with quotation work?

fredlcore, 2020-12-10:
Yes, the curl call with quotation mark works:
`curl -v -H "Content-Type: application/json" -X POST -d '{"Parameter":"700", "Value":"2", "Type":"1"}' http://192.168.1.50/4444/JS`
This one does not:
`curl -v -H "Content-Type: application/json" -X POST -d '{Parameter: 700, Value: 0, Type: 1}' http://192.168.1.50/4444/JS`

In the older version, I did the check differntly by just scanning for a numerical value after finding the words "Parameter", "Value" or "Type". That's why it worked with or without quotation marks.

dukess, 2020-12-10:
I make bugfix for this issue (Yep it is my mistake) but it need to test: i have no board right now.

Desc: "More complexity parsing of structure of incoming JSON message: values can be strings or numbers.
Multiple JSON messages in a single request (/JS) allowed"

fredlcore, 2020-12-10:
I have merged @dukess' bugfix, can you please try it out, @cn72 and maybe also @hacki11?

cn72, 2020-12-10:
Now it wont work from ioBroker,
POST /JS HTTP/1.1

/JS
URL: JS
setting line: 700 val: 01 02 
LAN->HEIZ INF      3D2D0574 01 02 
DC C2 00 0D 02 3D 2D 05 74 01 02 F6 57 
Setting parameter 700 to 2 with type 0
GET /JQ=700 HTTP/1.1

/JQ=700
URL: JQ=700
LAN->HEIZ QUR  700 Heizkreis 1 - Betriebsart: 
DC C2 00 0B 06 3D 2D 05 74 9C 4B 
HEIZ->LAN ANS  700 Heizkreis 1 - Betriebsart: 3 - Komfort
DC 80 42 0D 07 2D 3D 05 74 00 03 87 10 

and from cURL without quotation marks
GET /JA HTTP/1.1

/JA
URL: JA
POST /JS HTTP/1.1

/JS
URL: JS
Setting parameter -1 to H/  with type 0

with cURL an quotation marks it works
/JA
URL: JA
POST /JS HTTP/1.1

/JS
URL: JS
setting line: 700 val: 01 03 
LAN->HEIZ SET  700 Heizkreis 1 - Betriebsart: ---
DC C2 00 0D 03 3D 2D 05 74 01 03 5E 17 
HEIZ->LAN ACK  700 Heizkreis 1 - Betriebsart: 
DC 80 42 0B 04 2D 3D 05 74 44 75 
Setting parameter 700 to 3 with type 1
DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
DC 8A 00 0B 06 3D 0D 05 19 4F 8C 
HEIZ->DISP ANS 8310 Diagnose Erzeuger - Kesseltemperatur: 56.6 &deg;C
DC 80 0A 0E 07 0D 3D 05 19 00 0E 26 D8 20

dukess, 2020-12-10:
Thank you for testing!

> Setting parameter -1 to H/� � with type 0

Will be fixed in next pull.

@hacki11 FYI: /JA not supported anymore, because averages can be requested as usual programs 20000-20699

dukess, 2020-12-10:
Tried to test last changes.

Now possible multiple queries in one JSON:
curl -v -H "Content-Type: application/json" -X POST -d '{{"Parameter": 700 , "Value": 1, "Type":1},{"Parameter": 1603 , "Value": "testvalue", "Type":1},{}}' http://192.168.66.108/JS

Single query allowed too:
curl -v -H "Content-Type: application/json" -X POST -d '{"Parameter":"700", "Value":2, "Type":1}' http://192.168.66.108/JS

Parameters must be placed inside "". Numerical values can be placed or not inside "". String values must be wrapped by quotes.

/JQ, /JC working too.
curl -v -H "Content-Type: application/json" -X POST -d '{{"Parameter": 700 },{"Parameter": 1603}}' http://192.168.66.108/JC

fredlcore, 2020-12-10:
Great, thanks a lot, I have merged your PR already. @cn72, can you please test again? And @hacki11, in case your requests do not comply with the RFC as you stated above, could you adjust your module? That would be great, thanks everyone!

cn72, 2020-12-10:
with curl it works, but from ioBroker not.
@hacki11 : I tried to change it and this works, but I set Type to 1 fix:
    async write(id, value, type) {
        return queue.add(() => rp({
                method: 'POST',
                uri: "http://" + this.host + "/JS",
                headers: this.auth,
                json: true,
                body: {
                    "Parameter": id,
                    "Value": this.convert(value, type),
                    "Type": 1 ,
                },
                timeout: 15000,
                retry: 2,
I'am not able to use '${config.objects[id].writeType}', then everytime BSB-LAN set Type to 0. But I am not fit with javascript....
Temporarily it will work for me.

hacki11, 2020-12-10:
I think the issue is, the type was double quotation marked:
[image]

that explains why your fix works. you removed the "" around the Type value

cn72, 2020-12-10:
@hacki11 
it work...
 async write(id, value, type) {
        return queue.add(() => rp({
                method: 'POST',
                uri: "http://" + this.host + "/JS",
                headers: this.auth,
                json: true,
                body: {
                    "Parameter": id,
                    "Value": this.convert(value, type),
                    "Type": `${config.objects[id].writeType}` ,
                },
many thanks....  at all

hacki11, 2020-12-10:
i just pushed the fix to master (v0.2.1)

hacki11, 2020-12-10:
Yes, Backticks are the "code" for `code-formatting`. if you enclose your complete code between 3x ` backticks inside the codeblock will show fine
```
async write(id, value, type) {
return queue.add(() => rp({
method: 'POST',
uri: "http://" + this.host + "/JS",
headers: this.auth,
json: true,
body: {
"Parameter": id,
"Value": this.convert(value, type),
"Type": `${config.objects[id].writeType}` ,
},
```

cn72, 2020-12-12:
@dukess @fredlcore 
is there a posibility to get the information using INF (0) or SET (1) while use /JQ=700 ?
For example:
```
700 |  
name | "Betriebsart"
dataType_name | "ENUM"
dataType_family | "ENUM"
error | 0
value | "2"
desc | "Reduziert"
dataType | 1
readonly | 0
unit | ""
writeType | 1
```
the laste row....

fredlcore, 2020-12-12:
No, because we don't know that information for every parameter.

fredlcore, 2020-12-20:
Can we consider this matter closed for now or is there still anything on anyone's ToDo list, @cn72, @hacki11 and @dukess?

dukess, 2020-12-20:
I finished if no problem here.

cn72, 2020-12-20:
it looks all ok now.

fredlcore, 2020-12-20:
Ok, thanks!

closed 2020-12-20 by fredlcore