fredlcore/BSB-LAN #221: Disabled ENUM parameters show list of categories in dropdown

fredlcore, 2020-12-14:
@konne reported this to me, unfortunately, I don't have parameters that can replicate this behaviour, but here is the description as well as logs and screenshots:

Parameters of type VT_ENUM which are disabled ("---") do show the "---" in the text next to the parameter when queried, but the dropdown is filled with the list of categories. I assume that this is because of two reasons:
1. We have not handled the case of disabled ENUMs (at least not in the display of dropdowns) yet.
2. Because enumstr is not written when the heater replies "disabled" to a parameter query, enumstr contains the last list which in that case is the list of categories (because the corresponding category is listed in the same line).

Therefore, we have to add "---" to every ENUM dropdown list with an empty option value. This option must get the SELECTED attribute if the first byte of the payload is 0x01 (normally, it's 0x00).

@dukess, I'm not sure if this is handled best with another attribute of decodedTelegram? Such as decodedTelegram.enabled? Or what would you think?

Bus log:
DC C2 00 0B 06 3D 05 09 A2 5D FA
QUR 0x053D09A2 Kesselpumpe Q1 (8304)
DC 80 42 0D 07 05 3D 09 A2 **01 00** 24 0E
ANS 0x053D09A2 Kesselpumpe Q1 (8304)

[image]
[image]

konne, 2020-12-14:
@fredlcore I already plan to implement into my bsbJS a Server Test mode. So that you have a chance to test issues from the field.
The server mode just returns answers for 0x00 for a given request, so that you can test it.

**Question**: Does this 0x01 means that this enum is enabled? Because the enable_byte = 1 so this can not be the root cause of this issue.

dukess, 2020-12-14:
First idea: when heater return ---, then we can set something to 
decodedTelegram.error, ex. 263.
Second idea like your: add to printENUM() condition for searching and 
printing "---" value.
PS. Need to improve build_pvalstr().

fredlcore, 2020-12-14:
@konne: The payload is 01 00. The first byte of payload data is either 00 (enabled) or 01 (disabled). If disabled, the remaining payload data is ignored. Disabled does not mean the parameter itself is unavailable, but that the functionality is disabled (like in a time program that is not used).

So far, we never had "disabled" ENUMs because usually that would still mean some kind of function (even if it is a negated functionality) that would be addressed with a specific option value, like "off".
In your case, you probably don't have a QX1 pump sensor (and likewise no wireless equipment, given by the result of parameter 130 you sent me), and instead of other heating systems which just report "parameter not supported", yours replies with "disabled". That we would have to address somehow.

konne, 2020-12-14:
@fredlcore but than I don't understand your SET code:
If ->  enable_byte = 1 that means that 1 is enabled and 0 is disabled and not the opposite in the described case above.

```c
    param[0]=enable_byte;  //enable
...
  } else {
    param[0]=enable_byte-1;  // disable
```

fredlcore, 2020-12-14:
There is no contradiction. You are talking about the SET telegram, here we are talking about the ANS telegram. Same numbers, different meaning.

dukess, 2020-12-14:
> In your case, you probably don't have a QX1 pump sensor (and likewise no wireless equipment, given by the result of parameter 130 you sent me), and instead of other heating systems which just report "parameter not supported", yours replies with "disabled". That we would have to address somehow.

@fredlcore thank you for explanation. In this case we can set decodedTelegram.error = 7 (parameter not supported) for any disabled parameters. Or here can be problems on other systems?

fredlcore, 2020-12-14:
@dukess: Yes, that would probably be the easiest way.
@konne: Could you please try and set the ENUM via the URL (e.g. /S8304=1) and post the data telegrams here?
If the SET is not accepted, then we can go the way you said, @dukess. If it's possible to set the parameter (and thus "enable" ist), we have to add the "---" as an option because then it may somehow make sense to disable and ENUM (which so far does not make sense to me, but who am I ;) ).

1coderookie, 2020-12-14:
> QUR 0x053D09A2 Kesselpumpe Q1 (8304)

> In your case, you probably don't have a QX1 pump sensor

Just because it seems that Q1 & QX1 have been mixed up: we have to differentiate between Q1 and QX1, that's not the same. 

QX1 (and  QX-connections) are multifunctional relais connections which can be used for different meanings. They are _outputs_. One of the optional usecases is to connect the boilerpump labeled internally as 'Q1' (Kesselpumpe Q1) to it - if you have it. 

One can't connect a _sensor_ to the QX-connectors though, for that you have to use e.g. BX(1 or so).

As an example here's a screenshot of the config of one of my testsystems regarding QX1, QX2 and the Q1 @konne was talking about initially. 
[image]

You can see that I don't have a Q1 configured (also not at other multifunctional connectors). 
If I change the config now and set Q1@QX1, then the output of 8304 is correct though:
[image]

I'm wondering if it would be a good idea if that 8304 (=Q1) would be displayed as 'error7 parameter not supported', because that wouldn't really be the case. In my understanding the parameter _is_ supported - it's just not configured in the first example.

..of course, showing the categories within the dropdown menu for that special ENUM type is not nice, same with e.g. 5700..

Or am I missing something..?

1coderookie, 2020-12-14:
> If the SET is not accepted, 

It shouldn't be possible to SET 8304, because (in my understanding) it just shows the state of Q1 if connected. Note that it's located within the category  _diagnosis_..

konne, 2020-12-14:
> @konne: Could you please try and set the ENUM via the URL (e.g. /S8304=1) 
...  FEHLER: Setzen fehlgeschlagen! - Parameter ist nur lesbar.

this is the answer of the BSB-LAN

But if I follow up on 1coderookie you have to add "---" to make disable possible

dukess, 2020-12-14:
Partial fix: root category displayed instead ENUM when heater return "---"

https://github.com/fredlcore/bsb_lan/pull/223

dukess, 2020-12-14:
Question: what value for an enum would be safe to use as "---"? 65535?

fredlcore, 2020-12-14:
That's even more strange, because it wouldn't make sense to be able to write a read only parameter. It's probably set to read only in _defs.h. Could you try the sane with 130?

If this only affects read only parameters, it doesn't make sense to provide access to disable the parameter.

1coderookie, 2020-12-14:
> Partial fix: root category displayed instead ENUM when heater return "---". Test it, please.
> 
> #223

Just did a quick test with 5700 (because there also the category list appeared) and with 8304:
[image]
It seems, that the option that imho 'fits' best in the pulldown-menu is located at the bottom of the list, but the menu shows the first setting:
- For 8304 it is shown 'Aus', the last option in the pulldown is 'Keine Funktion' (='no function'), which imho fits better. Btw: 'Ein' is shown two times there.
- For 5700 the last entry is 'Nicht verwendet' (='not used') which imho fits better then the value which is displayed (note that 5700 isn't accessable through my QAA75 though).

I can search for other parameters of that ENUM-type and test again if you want..?

dukess, 2020-12-14:
> Btw: 'Ein' is shown two times there.

@1coderookie here is "crutch" because heater can return 1 or 255 for "On".

1coderookie, 2020-12-14:
Ah ok, got it :)

fredlcore, 2020-12-14:
Ok, thanks @1coderookie for the explanation and the detailed example, now I think I understand it better. 
In that case, "disabled" is a correct answer to be displayed in the dropdown, or "---" (which would not have to be translated and is known to users from other disabled parameters). The BSB-LAN logic so far is (for setting parameters as "disabled") is to send an empty string to the Arduino. 

I haven't thought this through completely, so I'm not sure if this is compatible / easy to integrate in the current ENUM logic which relies on values, but maybe we could just test if the first byte of payload is 01 and then ignore the rest of the payload?

fredlcore, 2020-12-14:
@dukess' PR merged.

dukess, 2020-12-14:
When first byte is 0x01 - situation is clear.
Is there a way to know if the value **_can_** be "---" when requested(some flag value, special request, etc.) so that the user knows that it is possible to disable the parameter even if some value is currently assigned? 
Or, if we need it, we need add one more column in cmdtbl[] and test all programs with sending empty strings?

fredlcore, 2020-12-14:
We don't know which parameters can be disabled. For the webinterface, this has so far been implemented that with non-Dropdown menus, the user can send an empty value which will then (try to) disable that parameter. As I said, I don't see an ENUM parameter that could/should be disabled. For a read-only parameter, it now makes sense to me because it depends on another parameter. But for a general ENUM parameter it doesn't make sense (at least by the concept of ENUMs).

dukess, 2020-12-14:
I make new patch https://github.com/fredlcore/bsb_lan/pull/224/ 
It append ENUMs with line value 65535 and description "---".
VT_ENUM will return 65535 when query return "disabled"

Possible it will be working.
Possible side effects.
For testing and further discussion.

1coderookie, 2020-12-17:
@dukess: Just did a quick check with the recent version, unfortunately now the "65535" is shown within the output of webinterface right before the ---, see screenshot:
[image]

dukess, 2020-12-17:
Here is two way : 
1. We can add condition for hiding 65535 in HTML. (In JSON it will still)
2. or leaving it as is if it not hinder to work

fredlcore, 2020-12-17:
I think we can leave the 65535, because every ENUM entry has a value, so if we hide or disable it, it would possibly break compatibility with external modules which expect a value here. 
I think this matter is solved then, so I'll close the issue here, but anybody feel free to reopen it if the need arises.

closed 2020-12-17 by fredlcore