fredlcore/BSB-LAN #227: Question: TYPE_QRV behavior.

dukess, 2020-12-15:
For more deeper understanding:
I tried do reset program value to their default by calling reset().
reset() send TYPE_QRV telegram and recieve TYPE_ARV with default value as payload. But this value not applied in heater in same time.
So there question - what is it: reset() function contain error(s) or after recieving TYPE_ARV user must send TYPE_SET with value that was received in TYPE_ARV?

Heater: Baxi Luna Platinum.

fredlcore, 2020-12-15:
Your observation is correct. If you take a look at the logs when calling the "Reset" function of one of the parameters, the display does the following:
1. Send a QRV telegram
2. Receive an ARV telegram and take the reset value from the payload
3. Send a SET telegram with the received reset value
4. Wait for an ACK telegram to confirm the new value

So maybe "Reset" is a misnomer, it's more like querying the default value. What the user (or the heater) does with it, is up to the user (or the heater).

dukess, 2020-12-15:
Thank you! Propose to rename from reset() to queryDefaultValue() or queryResetValue()

fredlcore, 2020-12-15:
Sure!

closed 2020-12-15 by fredlcore