fredlcore/BSB-LAN #234: bsbWeb

konne, 2020-12-20:
hi,

I like to start a discussion about the idea to have a full new web interface that only communicates through the JSON API.
I'm starting at the moment to develop a bsbJS Lib that can run perfectly run on rapsi and any system that can run node.
The library uses the full C definition file of bsb_lan so that it 
For that, I have to develop a full web interface that delivers the same results to the end-user.

My Idea now, why not have in the future only ONE web frontend that is used by everybody with Arduino and node user.
Here my Pro/Contra list. Please add missing points so that in the end you as a team can make your decisions.

Pro:

* All user have the same frontend
* less Arduino code because only JSON API is necessary, easier to maintain,...
* possibility to reduce code footprint in Flash
* easier to change web functionality
* "maybe" a possibility to get the AVR user back on board, because the code can be reduced

Contra:

* breaking changes for users that use the HTML as API and not JSON
* bsb_lan rely on bsbWeb
* bsb_lan developer need to learn if they want to change/extend bsbWeb
* if "compiled" bsbWeb is added to flash (50kb) is needed, if delivered via sd card, everybody needs an sd card and has a separate copy step

konne, 2020-12-20:
here a short demo of the current version:

[image]

konne, 2020-12-20:
please @fredlcore  @1coderookie  @LucekDev  @dukess add your feedback

fredlcore, 2020-12-20:
Thank you for your efforts and the demo, it really looks nice!
If I understand this correctly, it could already be done through what @dukess has provided with the WEBSERVER definement and the possibility to run BSB-LAN via webpages to display the user interface. Maybe the two of you should discuss how your approach could be implemented in that way?

>     * if "compiled" bsbWeb is added to flash (50kb) is needed, if delivered via sd card, everybody needs an sd card and has a separate copy step

As for a general replacement of the current user interface, we have discussed the use of SD cards in the past already (somewhere down in the 5000+ post thread in the FHEM forum). There has been a design decision relatively early on to choose the Arduino in favor of a Raspi or other platform because of the fact it runs both without a operating system that would need to be maintained as well as any parts that are exposed to wear and tear (such as SD cards). The idea was to have a "fire and forget" kind of system that could run for years and decades without further interaction. Faulty SD cards can be a pain in the back to identify as origins of errors, as I myself have witnessed several times, so for all our functionality add-ons, we have always chosen the use of an SD card as optional.

If your new web interface can save space for those users who are willing to use a SD-card, and we can make it optional, I'm all in, provided that we can clearly separate both approaches (the same way that Sergey has done it with the WEBSERVER definement), so that we can also split the workload.

konne, 2020-12-20:
@fredlcore  thanks for your feedback. 
I think at the moment during your big changes towards V2.0 is a good time to discuss it, because after your
big changes, with some breaking, changes everybody expects a long stable period.

I understand your ideas around sd-card.

What I see at the moment is, that if I compile it, that there are still 200kb of flash free, so if the person would not like to
have it on the sd-card, it could be compiled into the flash?

Would this a possible solution, or not?

fredlcore, 2020-12-20:
It's true that for Latin (i.e. 8-Bit ASCI)I languages, we still have 200k left. The situation looks different for UTF-8 encoded languages like Russian or Greek where the language files are basically double the size - that was the reason why @dukess came up with the possibility to run the whole webinterface via SD-card. To add another 50k to that would bring us relatively soon to a point where we will be forced to make decisions which features to compile in. If we have a solution for this through the means that Sergey has provided, I would rather want us to make use of that.

konne, 2020-12-21:
Hi, have you ever thought to use something like https://github.com/siara-cc/Unishox_Arduino_Progmem_lib
I think that can solve your issue for non-Latin languages and then there would be also enough space for bsbweb

fredlcore, 2020-12-21:
I can have a look at it, but that won't be before beginning of January. I would prefer to check whether Sergey's approach is feasible because it would also compensate for all the other cons you have listed.

1coderookie, 2020-12-21:
It's probably not the same, but didn't @dukess already wrote something like that..? https://github.com/dukess/bsb_lan_ajax

1coderookie, 2020-12-21:
But in general, also in my opinion the usage should be possible without the (general) need of an SD card.
Looks good though @konne! :)

konne, 2020-12-21:
@1coderookie thx for the hint to dukess bsb_lan_ajax, my variant is completely different, I will also talk to @dukess to allign here. It doesn't make sense to develop thing multiple times.

dukess, 2020-12-21:
As told Mao Zedong "lets blooms thousand flowers"!
@konne Now i understand why you need "Access-Control-Allow" directives in HTTP headers. When i just start develop AJAX i was forced to make a redirector based on PHP file on HTTP server. 

Yes, we can cut most part of native web interface for saving flash (with new definition like I_DO_NOT_NEED_NATIVE_WEB_INTERFACE). Yes, WEBSERVER module can deliver /index.html when user query / URL so internal interface can be replaced by HTML from SD card. 

> Hi, have you ever thought to use something like https://github.com/siara-cc/Unishox_Arduino_Progmem_lib

It is interesting idea. Possible it will be next stage of optimization. But I'm not ready to take it on, as at the moment for my purposes there is no need for it because all language resources placed on SD.

fredlcore, 2020-12-21:
> Yes, WEBSERVER module can deliver /index.html when user query / URL so internal interface can be replaced by HTML from SD card.

Great, so that sounds like an approach that combines best of both worlds.

konne, 2020-12-21:
@dukess #define I_DO_NOT_NEED_NATIVE_WEB_INTERFACE would be great if you can add this to the code and also the direct read for index.html from the sd card.

dukess, 2020-12-21:
First we need to agree on what to cut. If we leave configure through URL, we can free ~12Kb (incl. D3 library). Not too much.

konne, 2020-12-21:
@dukess I can compile down my Vue App at the moment to around 48k but without d3. From my standpoint the real flash save has to come from the Unishox to reduce the strings. I think from the code the biggest advantage of a pure URL / JSON codebase would be the reduction of code and no mix between parsing and generation of Html.
My suggestion is at the moment to have the two-way mode with bsbWeb on SD or generated Html from the code. To have a full ajax / json webapp, there are some more JSON APIs needed, like fetch for unknown commands,..
And if the web app that I develop is really great and everybody likes it, the decision can maybe be made in 2-3 months.

konne, 2021-01-03:
@dukes @fredlcore @1coderookie
I just like to give you an update. Currently, I'm working on the bsbJS API and this is now nearly 100% compatible with the bsb-lan JSON API. (/JQ works 100%) now I implement (JS) after this is done I will take invest more time into bsbWEB

1. I like to return Units correct and not URL Encoded
2. VT_SUMMERPERIOD should be for me formatted with a dot after the month

# bsbWEB
```json
{
  "0": {
    "name": "Datum/Zeit",
    "error": 0,
    "value": "02.01.2021 21:46:39",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "6": {
    "name": "Sommerzeitende Tag/Monat",
    "error": 0,
    "value": "25.10.",
    "desc": "",
    "dataType": 6,
    "readonly": 0,
    "unit": ""
  },
  "500": {
    "name": "Mo",
    "error": 0,
    "value": "1. 04:00 - 00:00 2. --:-- - --:-- 3. --:-- - --:--",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "600": {
    "name": "Mo",
    "error": 0,
    "value": "1. 05:30 - 07:30 2. 10:30 - 12:30 3. 16:30 - 18:30",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "700": {
    "name": "Betriebsart",
    "error": 0,
    "value": 1,
    "desc": "Automatik",
    "dataType": 1,
    "readonly": 0,
    "unit": ""
  },
  "701": {
    "name": "Präsenztaste (temporäre Abwesenheit)",
    "error": 7,
    "value": "",
    "desc": "",
    "dataType": 1,
    "readonly": 0,
    "unit": ""
  },
  "710": {
    "name": "Komfortsollwert",
    "error": 0,
    "value": "20.5",
    "desc": "",
    "dataType": 0,
    "readonly": 0,
    "unit": "°C"
  },
  "832": {
    "name": "Antrieb Typ",
    "error": 0,
    "value": 255,
    "desc": "3-Punkt",
    "dataType": 1,
    "readonly": 1,
    "unit": ""
  },
  "1602": {
    "name": "TWW Status",
    "error": 0,
    "value": "01000101",
    "desc": "",
    "dataType": 2,
    "readonly": 0,
    "unit": ""
  },
  "5011": {
    "name": "Trinkwasser-Speicher Ladevorlegungszeit",
    "error": 0,
    "value": "01:00",
    "desc": "",
    "dataType": 4,
    "readonly": 0,
    "unit": ""
  },
  "6224": {
    "name": "Geräte-Identifikation",
    "error": 0,
    "value": "LMS15.003A100",
    "desc": "",
    "dataType": 7,
    "readonly": 0,
    "unit": ""
  }
}
```

# vs bsbLAN
```
json
{
  "0": {
    "name": "Datum/Zeit",
    "error": 0,
    "value": "02.01.2021 21:36:03",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "6": {
    "name": "Sommerzeitende Tag/Monat",
    "error": 0,
    "value": "25.10",
    "desc": "",
    "dataType": 6,
    "readonly": 0,
    "unit": ""
  },
  "500": {
    "name": "Mo",
    "error": 0,
    "value": "1. 04:00 - 24:00 2. --:-- - --:-- 3. --:-- - --:--",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "600": {
    "name": "Mo",
    "error": 0,
    "value": "1. 05:30 - 07:30 2. 10:30 - 12:30 3. 16:30 - 18:30",
    "desc": "",
    "dataType": 5,
    "readonly": 0,
    "unit": ""
  },
  "700": {
    "name": "Betriebsart",
    "error": 0,
    "value": "1",
    "desc": "Automatik",
    "dataType": 1,
    "readonly": 0,
    "unit": ""
  },
  "701": {
    "name": "Präsenztaste (temporäre Abwesenheit)",
    "error": 7,
    "value": "",
    "desc": "",
    "dataType": 1,
    "readonly": 0,
    "unit": ""
  },
  "710": {
    "name": "Komfortsollwert",
    "error": 0,
    "value": "20.5",
    "desc": "",
    "dataType": 0,
    "readonly": 0,
    "unit": "&deg;C"
  },
  "832": {
    "name": "Antrieb Typ",
    "error": 0,
    "value": "255",
    "desc": "3-Punkt",
    "dataType": 1,
    "readonly": 1,
    "unit": ""
  },
  "1602": {
    "name": "TWW Status",
    "error": 0,
    "value": "01000101",
    "desc": "",
    "dataType": 2,
    "readonly": 0,
    "unit": ""
  },
  "5011": {
    "name": "Trinkwasser-Speicher Ladevorlegungszeit",
    "error": 0,
    "value": "01:00",
    "desc": "",
    "dataType": 4,
    "readonly": 0,
    "unit": ""
  },
  "6224": {
    "name": "Geräte-Identifikation",
    "error": 0,
    "value": "LMS15.003A100",
    "desc": "",
    "dataType": 7,
    "readonly": 0,
    "unit": ""
  }
}
```

rroblik, 2021-01-05:
Not sure to understand all ... but I'm new user of bsb_lan hard/software for now in testing mode.

My target is to integrate it into my automation software so I think be closer to a "standard" JSON api is the way to go for this great and active project !

I subscribe to this post, maybe I can help you somehow ...

@konne can I test your work ?

konne, 2021-01-05:
@rroblik I'm working at the moment on two main libraries:

1. **bsbJS**: full implementation of the current bsb-lan in JS so that you can run it for example on an Rapsi, NAS,... you only need the adapter (Raspi) or a USB / RS232 / aLAN/RS232 adapter to run it on other platforms. This will be  very soon documented. The current status is (marked means ready)

- [X] JQ=id1,id2,...
- [X] JS
- [X] JK=ALL
- [ ] WIP (70%) JK=1
- [ ] WIP JI
- [ ] Documentation
- [ ] config / logging ... later

2. **bsbWEB**: a web interface that runs on VUE and has the same look and feels like the current bsb-lan but only uses the JSON API. That bsbWEB can than used together with bsbJS or bsbLAN
The projects are already online on github, but I wait to make the links public until the software and the documentation is more complete. If you just search on my profile you will find it.

- [X] read settings
- [ ] WIP write settings
- [ ] logging (real time with websocket)
- [ ] search for new parameters
- [ ] history (but only for bsbLAN)

bye

rroblik, 2021-01-07:
@konne  : I'm going to test it once available

fredlcore, 2021-02-08:
I'm moving this to the discussion area as this is a better environment where comments can be made directly to specific matters/questions.

closed 2021-02-08 by fredlcore