fredlcore/BSB-LAN #235: BME280 integration (was: Discussions #220)

1coderookie, 2020-12-21:
Here we can continue thinking about a possible BME280 integration, I initially started it in 'Discussions #220' cuz I thought it would be better there as it isn't an 'issue' yet.. ;)

dukess, 2020-12-21:
> I'm not really well informed about both I2C as well as SPI. I just know that SPI is not really an option if we want to keep open the option for WiFi (which certainly more people will use compared to BME sensors). If I2C can be used with both the EEPROM as well as the sensor, then this would be a feasible option, though.

If WiFi will have different I2C address (not 0x76 nor 0x77 nor EEPROM) then it would be more real to do. Probably.

fredlcore, 2020-12-21:
WiFi will run via SPI, not I2C. As there is no ready shield available, users will have to use wire jumpers, so you can either connect the WiFi board or the sensors...

dukess, 2021-01-23:
I want to use https://github.com/BlueDot-Arduino/BlueDot_BME280 for BME connection. It supports two mode: I2C and SPI. 
Which pins can i use (library supports many sensors and need unique pin per sensor) in SPI mode? 
Can we share MOSI, MISO, SCK pins between SD, ETH/WIFI and BME?

Or it better to use I2C? 

How many sensors we must support simultaneous? (1-2 can be handled with I2C, SPI can handle at least 4 sensors).

fredlcore, 2021-01-23:
The Ethernet shield does not provide a SPI header, so it won't be able to use it with that one. Let's go for I2C if that is possible.

1coderookie, 2021-01-24:
I agree. SDA1/SCL1 isn't available at the LAN shield either though (only directly at the Due), but at least it's free ;)

fredlcore, 2021-01-24:
Hm, not sure I understand I2C well enough, but we have SDA/SCL on pins 20/21 which are currently used by the BSB-LAN board for accessing the EEPROM. At least on previous boards, these pins are not accessible for additional periphery (and I don't know if you can access more than one device at the same time).
Then there is SDA1/SCL1 which are on the top left of the board (looking at the board with the USB ports on the top). Not sure if these are the same pins or a second I2C bus, but these pins should be accessible also with an Ethernet shield installed.

1coderookie, 2021-01-24:
Due offers two I2C 'connectors': SDA/SCL at 20/21 and SDA1/SCL1 near to the AREF-Pin. LAN shield doesn't plug in SDA1/SCL1 (so they're not 'carried through' to the top) but they are free to access 'underneath' the LAN shield.

fredlcore, 2021-01-24:
Ok, good enough still :)...

dukess, 2021-01-25:
https://github.com/fredlcore/bsb_lan/pull/311

closed 2021-01-31 by 1coderookie