fredlcore/BSB-LAN #236: Configuration through JSON

dukess, 2020-12-22:
During the discussion https://github.com/fredlcore/bsb_lan/issues/234 , I had the idea to import and export the configuration of BSB-Lan in JSON format.
Here is first version: https://github.com/dukess/bsb_lan/commit/9ed4d5d9ec77ac9d09656fa01d6ed338d34acafe

It can export config (/JL command, see below) and import (/JW, not tested yet) config options.

Near target: fast save/restore config when it needed.
Far target: possibility of disabling module of configuration through URL commands. Possibility of disabling internal web interface (excluding /Q command).
I read on FHEM forum that someone was thinking about BSB-Lan without HTML interface.

What you thinking?

Example:
```
{
  "0": {
    "id": "0",
    "type": "1",
    "format": "2",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_USEEEPROM_TEXT",
    "value": "150",
    "possibleValues": [
      { "enumValue": "105", "desc": "MENU_TEXT_OFF" },
      { "enumValue": "150", "desc": "MENU_TEXT_ON" }
    ]
    },
  "3": {
    "id": "6",
    "type": "1",
    "format": "3",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_BUSTYPE_TEXT",
    "value": "0",
    "possibleValues": [
      { "enumValue": "0", "desc": "BSB" },
      { "enumValue": "1", "desc": "LPB" },
      { "enumValue": "2", "desc": "PPS" }
    ]
    },
  "4": {
    "id": "7",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_OWN_BSBADDR_TEXT",
    "value": "66"
    },
  "5": {
    "id": "8",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_OWN_LPBADDR_TEXT",
    "value": "66"
    },
  "6": {
    "id": "9",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_DEST_LPBADDR_TEXT",
    "value": "0"
    },
  "7": {
    "id": "10",
    "type": "1",
    "format": "2",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_PPS_WRITE_TEXT",
    "value": "0",
[... 312 lines cut ...]
    "value": "BSB-LAN"
    },
  "45": {
    "id": "42",
    "type": "4",
    "format": "1",
    "category": "CAT_MQTT_TEXT",
    "name": "CF_MQTT_TOPIC_TEXT",
    "value": "MyHeater"
    },
  "46": {
    "id": "46",
    "type": "1",
    "format": "2",
    "category": "CAT_MQTT_TEXT",
    "name": "MENU_TEXT_NVA",
    "value": "1",
    "possibleValues": [
      { "enumValue": "0", "desc": "MENU_TEXT_OFF" },
      { "enumValue": "1", "desc": "MENU_TEXT_ON" }
    ]
    }

}
```

1coderookie, 2020-12-22:
I think that's a good idea (like a backup always is ;) ) - especially for users who have complex installations with eg a dozen of parameters configured for logging/MQTT and so on.

Where will that exported config be saved and from where will it be imported? Didn't get that yet..

> I read on FHEM forum that someone was thinking about BSB-Lan without HTML interface.

Where did you read that, do you have a link maybe?

fredlcore, 2020-12-22:
I think that was me maybe, when I was introducing the feature to control BSB-LAN via serial port. All URL (HTTP-GET) commands can be issued via the serial interface and the response is again printed on serial port. But this is not going to be a 100% replacement of the webinterface. Also, this needs the URL commands because it shares the same basis.

1coderookie, 2020-12-22:
Ah ok, thx for explanation!

dukess, 2020-12-24:
Added as pluggable module. Let's see how popular it will be.

closed 2020-12-24 by dukess

rroblik, 2021-01-08:
@dukess where is it ?

Thanks