fredlcore/BSB-LAN #239: New web configuration options for WiFi settings?

fredlcore, 2020-12-27:
@dukess: The current repo has the new WiFi library included which so far still requires a definement (WIFI) in _config.h as well as two variabled (wifi_ssid and wifi_pass). Would it be possible to make this configurable via the webinterface? Or if you are busy at the moment, could you give me a few pointers where I need to add these variables so they are added to the web-based config as well as EEPROM? Thanks!

dukess, 2020-12-27:
@fredlcore Done!

fredlcore, 2020-12-27:
Wow, that was fast, thanks! But the GitHub actions say that checks have failed, could you have a look at these, please, before I merge?

dukess, 2020-12-27:
Yep. Config files for testing will be replaced with new version.

fredlcore, 2020-12-27:
Thanks a lot!

closed 2020-12-27 by fredlcore