fredlcore/BSB-LAN #241: Can't set clock

dukess, 2020-12-28:
Frederik, can you look on this?

POST /JS HTTP/1.0
/JS
URL: /JS
date time: 28.12.2020 11:56:55
setting line: 0 val: 06 78 0C 1C 01 0B 38 37 00
LAN->HEIZ SET    0 ENUM_CAT_00_TEXT - STR0_TEXT: ---
DC C2 00 14 03 3D 05 00 0B 06 78 0C 1C 01 0B 38 37 00 68 93
HEIZ->LAN NACK    0 ENUM_CAT_00_TEXT - STR0_TEXT:  VT_DATETIME len !=9: 
06 DC 80 42 0C 05 05 3D 00 0B 06 B8 6C
set failed NACK

POST /JS HTTP/1.0
/JS
URL: /JS
date time: 28.12.2020 11:57:47
setting line: 0 val: 06 78 0C 1C 01 0B 39 2F 00
LAN->HEIZ SET    0 ENUM_CAT_00_TEXT - STR0_TEXT: ---
DC C2 00 14 03 3D 05 00 0B 06 78 0C 1C 01 0B 39 2F 00 D5 79
HEIZ->LAN NACK    0 ENUM_CAT_00_TEXT - STR0_TEXT:  VT_DATETIME len !=9: 
06 DC 80 42 0C 05 05 3D 00 0B 06 B8 6C
set failed NACK
Setting parameter 0 to "28.12.2020_11:57:47" with type 1
Renewed success
My IP address: 192.168.67.5

URL: /WS0=28.12.2020_13:52:08
Setting temporary destination to 0
set ProgNr 0 = 28.12.2020_13:52:08date time: 28.12.2020 13:52:8
setting line: 0 val: 06 78 0C 1C 01 0D 34 08 00
LAN->HEIZ SET    0 ENUM_CAT_00_TEXT - STR0_TEXT: ---
DC C2 00 14 03 3D 05 00 0B 06 78 0C 1C 01 0D 34 08 00 2F C0
HEIZ->LAN NACK    0 ENUM_CAT_00_TEXT - STR0_TEXT:  VT_DATETIME len !=9: 
06 DC 80 42 0C 05 05 3D 00 0B 06 B8 6C
set failed NACK

Other programs looks fine:
POST /JS HTTP/1.0
/JS
URL: /JS
setting line: 700 val: 01 02
LAN->HEIZ SET  700 ENUM_CAT_0b_TEXT - STR700_TEXT: 65535
DC C2 00 0D 03 3D 2D 05 74 01 02 4E 36
HEIZ->LAN ACK  700 ENUM_CAT_0b_TEXT - STR700_TEXT:
DC 80 42 0B 04 2D 3D 05 74 44 75
Setting parameter 700 to "2" with type 1

URL: /JS
setting line: 710 val: 01 05 D3
LAN->HEIZ SET  710 ENUM_CAT_0b_TEXT - STR710_TEXT: --- UNIT_DEG_TEXT
DC C2 00 0E 03 3D 2D 05 8E 01 05 D3 27 21
HEIZ->LAN ACK  710 ENUM_CAT_0b_TEXT - STR710_TEXT:
DC 80 42 0B 04 2D 3D 05 8E 0A 20
Setting parameter 710 to "23.3" with type 1

fredlcore, 2020-12-28:
Is this the first time you tried setting the clock via BSB-LAN? If yes, then maybe your heating system uses a different command ID. Could you set the clock on the heating system and log the telegram and post it here?
The error messages come from the NACK error telegram that printDateTime cannot parse.

dukess, 2020-12-28:
No. Until begin december clock was set successfully.

dukess, 2020-12-28:
The question is: your clock can be set correctly?

fredlcore, 2020-12-28:
Fixed. The enable_byte was set to 0x06, but 0x01 is correct.

closed 2020-12-28 by fredlcore