fredlcore/BSB-LAN #242: Consolidate used apostrophes in translation files

hacki11, 2020-12-28:
Currently in german translation file, three different apostrophes are used.
Examples:
Betr'std Feststoffkessel 0x27 APOSTROPHE (U+0027)
Betr`stunden Solarertrag 0x60 GRAVE ACCENT (U+0060)
Betr’stunden Elektro TWW 0xE28099 Unicode Character 'RIGHT SINGLE QUOTATION MARK' (U+2019)

Please consolidate every usage of to e.g. 0x27 APOPSTROPHE.

fredlcore, 2020-12-28:
That would certainly be nice, and if you (or someone else) has the time to do so, I would suggest to use 0x27 for that. If I add it to my ToDo list, it will certainly not be coming up anytime soon.

Even better would be to spell out the word in full, because these result from users submitting what they saw on their heater. On a small display, there is a need to shorten these words, but not in BSB-LAN. So instead of "Betr'stunden", it should say "Betriebsstunden".

hacki11, 2020-12-28:
Which file is source of truth for localization?
lang_xx.js
LANG_XX.h
lang_xx.json

dukess, 2020-12-28:
.js and .json builded from LANG_XX.h

fredlcore, 2020-12-28:
LANG_DE.h
(edit: @dukess was faster :) )

hacki11, 2020-12-28:
I have started to change the names, but now the names can not be matched to the user manuals anymore. Imho this is not that good, because with the new names you will not find any technical documentation. 
Example:
[image]

I think we should just fix the namings, where we got it wrong (like the apostrophes).

fredlcore, 2020-12-28:
The naming scheme varies from heating system to heating system and we will never be able to have a consistent naming scheme across all heating systems (at least not without significant flash consumption if we were to add each and every different kind of spelling). Most users will go for the parameter numbers anyway, as it is the only way to access a function directly. If we had a "parameter name search function", I would agree with you, but given that this is not the case, I'd plead for easily readable/understandable parameter names.

dukess, 2020-12-28:
It is best to give names without abbreviations, IMHO.

1coderookie, 2021-01-05:
What's the current state of this topic? Has anybody (@hacki11 ?) already searched&replaced the different types? Or is the final decision to write the names w/o abbreviations?

hacki11, 2021-01-05:
I have started to eliminate the abbreviations, but the more i do this the more i hesitate at all. Currently i'm optimizing my heating and i have to google all the abbreviated names and find everything. As fredlcore said, i could also search for parameter numbers but they do not exist for so much things like enum values or other stuff. It will getting so hard to use google to find relevant stuff. 

That is why i hesitate here and still favor the consolidation of apostrophes. I'm still fine with not changing anything here in favor of elimination of the original names.

fredlcore, 2021-01-05:
I would really vote for changing the descriptions to the full names because that is what translators will be using, too. Imagine how hard it is for them to find out what "Betr`std." means if we're already having trouble with it...

1coderookie, 2021-01-05:
Ok, so we'll try to go with the complete names.. I'll look up some manuals and will get back to you..

hacki11, 2021-01-05:
I can share my current work, I already have the most found out

1coderookie, 2021-01-05:
Yes, that would be nice of course! Maybe you wanna send me the file via Email? Or can you attach it here? But then it won't be a PR at the end @fredlcore - would/could that be a problem..?

fredlcore, 2021-01-05:
Someone will be doing the changes in the end, right? That person could then do the PR.

hacki11, 2021-01-05:
https://github.com/fredlcore/bsb_lan/pull/261

1coderookie, 2021-01-05:
I just left a comment there, but I'll post it here again also, maybe it could be useful somehow for someone else who stumbles across this issue:

Just really quick: in general, pretty much all names should be available within the detailed description of the specific parameter. If you don't find that specific parameter within that specific manual where you are looking for it in that moment, try another manual or just search for the name of that abbreviation alternatively.
E.g. from an elco manual: 693 not available in the specific manual I was looking at right now, but the same is mentioned within 3010 = Drehz max V'lator/Q'Pump = Begrenzt den Regelbereich der Ventilator- / Quellenpumpen-Drehzahl gegen oben. -> now we know what V'lator/Q'pump is.

I'll try to complete the list mentioned within that PR-draft tomorrow and post the names in the comment, then we'll see what else comes up. Ok?

fredlcore, 2021-01-05:
It says, it's a draft request, so I cannot merge it (yet), but thanks a lot!
"V'lator/Q'Pump" is really a great example why these abbreviations should be done away with, especially if the heating system manual is so lousy as Elco's (I have one of these, too :( )...

hacki11, 2021-01-05:
> It says, it's a draft request, so I cannot merge it (yet), but thanks a lot!
> 
That was intentionally, because i have. not finished it.

fredlcore, 2021-01-05:
Ok, then I'll wait for it...

fredlcore, 2021-02-08:
Any progress here, @hacki11?

1coderookie, 2021-02-08:
I _think_ I removed the different types of apostrophes and most of the abbreviations, but I'll have a look at it again when I'll have the time. Closing this for now.

closed 2021-02-08 by 1coderookie