fredlcore/BSB-LAN #243: "delete" when using WiFiSpiClient?

fredlcore, 2020-12-28:
@dukess: When using the WiFiSpi library, I get the following warning when activating MAX_CUL:

```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino: In function 'void connectToMaxCul()':
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:5946:12: warning: deleting object of polymorphic class type 'WiFiSpiClient' which has non-virtual destructor might cause undefined behavior [-Wdelete-non-virtual-dtor]
     delete max_cul;
```

Why are you using "delete" here and can it safely be omitted by using "#ifndef WIFI"? Or should we add a max_cul.stop() or something else here or somewhere else?

dukess, 2020-12-28:
Same warning i have for EthernetClient. I found not destructor (and examples of usage) in manual for EthernetClient so used standart call.

fredlcore, 2020-12-28:
Hm, for EthernetClient I don't get this warning. So should we just ignore it or close the client in some other way? I don't see a max_cul.stop() anywhere, and WiFiSpi requires clients to be stopped after use because only a maximum of four sockets can be used at the same time.

dukess, 2020-12-28:
At least add max_cul.stop() before delete.

fredlcore, 2020-12-28:
Doesn't work:
```
/Users/frederik/Documents/PlatformIO/Projects/BSB-LAN/src/BSB_lan.ino:5947:13: error: request for member 'stop' in 'max_cul', which is of pointer type 'WiFiSpiClient*' (maybe you meant to use '->' ?)
     max_cul.stop();
```
Is there an advantage to use "new" and "delete" compared to just do a ".begin()" and ".stop()" as we do for the overall "client" object? And is "delete" something else than "max_cul = NULL"?

At least here I read that deleting such an object is bad:
https://stackoverflow.com/questions/12994920/how-to-delete-an-object-of-a-polymorphic-class-type-that-has-no-virtual-destruct

fredlcore, 2020-12-29:
max_cul->stop() did work, so I added it.
As for the warning: I added this line to WiFiSpiClient.h:
    virtual ~WiFiSpiClient();
And the warning disappeared. Inside WiFiSpiClient.cpp, I added an empty function of the same name. Not sure if this is just a dirty hack or if this really prevents undefined behaviour. It would probably be really good to remove the no longer used socket in any case...

dukess, 2020-12-29:
> Doesn't work:

My mistake: when we create object "by value" we can get access to metods with ".". When we create object "by refenence" - with "->".

> Not sure if this is just a dirty hack or if this really prevents undefined behaviour. 

Need to do test: Add addidional debug code like 

```
void connectToMaxCul() {
  if(max_cul){
    printFmtToDebug(PSTR("Free RAM: %d\r\n"), freeRam());
    max_cul->stop();
    printFmtToDebug(PSTR("Free RAM: %d\r\n"), freeRam());
    delete max_cul;
    max_cul = NULL;
    printFmtToDebug(PSTR("Free RAM: %d\r\n"), freeRam());
    if(!enable_max_cul) return;
  }
```
enable and disable several times MAX sensors in Web Config and looking on behavior. If memory leaks then we do something wrong.

But if it work then need to add note about WiFi library modification because we forget about it after some time.

fredlcore, 2020-12-29:
Thanks a lot, I think I now finally understood the difference between dot and arrow notation :). As for the missing destructor, @JiriBilek is looking into it here:
https://github.com/JiriBilek/WiFiSpi/issues/21

dukess, 2020-12-29:
Tried to test on Mega with debug output. I see no problem/memory leaks.

dukess, 2020-12-29:
>Thanks a lot, I think I now finally understood the difference between dot and arrow notation :). As for the missing destructor, @JiriBilek is looking into it here:
JiriBilek/WiFiSpi#21

Cool!

closed 2021-01-04 by fredlcore