fredlcore/BSB-LAN #246: not use VT_WEEKDAY

konne, 2020-12-30:
Hi,

why you have the VT_WEEKDAY defined. I found no command that uses this.
see https://github.com/fredlcore/bsb_lan/blob/7d35a6745a6b58f80054d425beb6a50e10299f56/BSB_lan_defs.h#L547

Suggestions to remove this definition

bye
Konrad

dukess, 2021-01-01:
This is a legacy, but it will not be possible to remove it painlessly: the identifiers of data types will shift and compatibility will be broken.

konne, 2021-01-01:
@dukess thx, so I can just ignore it. Maybe it would make sense just to add a comment so that it is only for nonbreaking changes there.

Can you please explain to me, why this breaks compatiblity? I really like to understand the issue behind

dukess, 2021-01-01:
If we remove 
`{VT_WEEKDAY,        1.0,    1, 1, DT_WDAY, 0,  U_NONE, sizeof(U_NONE), STR_WEEKDAY},`
here will be no problem but if we delete DT_WDAY in dt_types_t enum then all constants below DT_WDAY will change their values. Because  DT_* is used in JSON delevopers and end users must to rewriting code and configs.

fredlcore, 2021-01-01:
VT_WEEKDAY can safely be ignored. It was used in one of the first versions for parameter 1642 (weekday for Legionella function), but that was later on changed to a VT_ENUM type. We should mark it accordingly, though.

fredlcore, 2021-02-08:
VT_WEEKDAY is now needed for PPS since that bus only stores day of week and not the real date, so we need a differentiation here.